espressif/arduino-esp32 #6557: ESP32-A1S support

serkansatuk, 2022-04-10:
### Related area

ESP32-A1S support

### Hardware specification

ESP32-A1S

### Is your feature request related to a problem?

Hello.
Please add ESP32-A1S board.
Thanks...

### Describe the solution you'd like

Hello.
Please add ESP32-A1S board.
Thanks...

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-04-11:
Hello, adding third-party boards is more likely up to the manufacturers of these boards. We are not able to be aware of all types and their specifications of non-Espressif Boards.

You can take a look on some closed PRs, how in general are boards added into Arduino IDE.

VojtechBartoska, 2022-05-04:
Closing as won't fix. We are not able to help as mentioned above.

closed 2022-05-04 by VojtechBartoska