espressif/arduino-esp32 #6558: esp32s3-devkitc 4mb max on 32mb chip!

bm16ton, 2022-04-10:
### Board

esp32s3_devkitc

### Device Description

Stock esp32s3_devkitc

### Hardware Configuration

Nothing attached needed to prove this

### Version

latest master

### IDE Name

arduino

### Operating System

linux

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Not sure if this should have gone under s3 support, or somewhere else. I know 32mb flash isnt officially supported, and s3 is still new so i aint complaining, but any partition scheme equal to 4mb (so only first 2 listed in menu) work all the rest cause a non-stop bootloop without error. Ive updated esptool and s3 branch when it existed and now master every couple of days. Im just super curious where the 4mb limit is coming from? Is it in the esptool? Was it hardcoded as the max for the s3 somewhere to get around some sorta issue? Ive tried many custom partitin schemes as well. Def dont like anything other then the first two partition options starting with 4mb in the label. Doubt id be useful but if i had a place to focus on id try and mess with fixing it. Thank you! Love your product.

### Sketch

```cpp
Any sketch will do just select any partition scheme above 4mb and it will boot loop
```

### Debug Message

```plain
No errors, no debug, it never complains before/during bootloop.
```

### Other Steps to Reproduce

Just select a > 4mb partition scheme

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-10:
I have a 8MB / 8MB (Flash / PSRAM) S3 devkit. After providing a custom partition scheme it does work without an issue. Both can be used fully.

bm16ton, 2022-04-10:
Yeah from what i can gather the issue stems from it being 32MB i havent
seen anyone with the smaller flash sizes having any issues this way, but i
have found other 32MB people with same issue. I know 32mb support is/has
being added im just not sure where the hiccup with the 4mb limit is coming
from. Esptool has had a lot of 32mb updates and a few a long the way caused
the exact same bootloop problem that we see in arduino with bigger then 4mb
partition scheme, so that seems like a possible culprit.

Jason2866, 2022-04-10:
There is no 4MB limit since it works with 8MB

modi12jin, 2022-04-11:
@bm16ton #6453 

Maybe you have this problem

bm16ton, 2022-04-11:
Oooh! Awesome! Im gonna checkout that commit as soon as i get a chance!
Thank you that may just do it!

bm16ton, 2022-04-11:
Unfortunatly already have that patch installed. And for sum reason the
newest round of patches for esptool cause the bootloop with any and all
partition schemes. Going back to a previous commit and all is back as was.
But i appreciate the effort thank you!

Jason2866, 2022-04-11:
When not using esptool v3.3 you will encounter problems with the S3.

VojtechBartoska, 2022-05-03:
@bm16ton the issue still persist? I'm not sure if I get it from the last comment :)

bm16ton, 2022-05-14:
yes it does sorry for super late reply i never noticed the email. I simply
cannot use anything other then the 2 optoons for 4mb, otherwise it
bootloops. I keep updating everything in hope of a fix. I also do complete
fresh installs on 2nd pc just to rule out a bad install etc. I havent in
about a week so i should probou look if any new commits may resolve my
issue. I appreciate your help! Havent had the best luck with online help,
so truely thank you.

DustinWatts, 2022-05-15:
> I have a 8MB / 8MB (Flash / PSRAM) S3 devkit. After providing a custom partition scheme it does work without an issue. Both can be used fully.

@Jason2866 Can you provide your partition scheme / share your csv? I for the life of me can't get mine to work.

Jason2866, 2022-05-15:
@DustinWatts  https://github.com/Jason2866/Tasmota/blob/development/boards/esp32s3_8M.json
and https://github.com/arendst/Tasmota/blob/development/partitions/esp32_partition_app2944k_fs2M.csv

DustinWatts, 2022-05-16:
@Jason2866 Thanks!

VojtechBartoska, 2022-05-18:
can I consider this as solved?

VojtechBartoska, 2022-06-14:
Closing as expired. If needed, reopen the issue.

closed 2022-06-14 by VojtechBartoska

bm16ton, 2022-10-11:
You said your running a 16mb chip right? And its a 32mb chip problem. 32mb
isnt even officialy supported yet. But your correct on the officially
supported sizes their is no 4mb limit