espressif/arduino-esp32 #6559: ESP32 compile error with arduino-esp32 v2.0.3-rc1 (error: 'esp32_gpioMux' was not declared in this scope)

bghavami, 2022-04-10:
### Board

Esp32S2 Dev Module and Esp32S3 Dev Module

### Device Description

NodeMcu-32S or Esp32-S3-DevKitC-1 v1.6
Testing with Dallas OneWire Temperature sensor

### Hardware Configuration

GPIO 2 connected to Dallas Temperature sensor

### Version

latest master

### IDE Name

Arduino IDE 1.8.20

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am getting error compiling sample/example code for Dallas Temperature sensor. 
Steps to generate the error:
1- Load the code: File -> Examples -> Dallas Temperature -> Simple
2- verify / Compile

Compiler generates the following error:
OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope  ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;

It appears that "esp32_gpioMux" is no longer defined!!!

### Sketch

```cpp
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

/*
 * The setup function. We only start the sensors here
 */
void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // Start up the library
  sensors.begin();
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  // call sensors.requestTemperatures() to issue a global temperature 
  // request to all devices on the bus
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  // After we got the temperatures, we can print them here.
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  float tempC = sensors.getTempCByIndex(0);

  // Check if reading was successful
  if(tempC != DEVICE_DISCONNECTED_C) 
  {
    Serial.print("Temperature for the device 1 (index 0) is: ");
    Serial.println(tempC);
  } 
  else
  {
    Serial.println("Error: Could not read temperature data");
  }
}
```

### Debug Message

```plain
In file included from C:\Users\eghav\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:36,
                 from C:\Users\eghav\Documents\Arduino\libraries\OneWire\OneWire.cpp:142:
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
C:\Users\eghav\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:232:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
C:\Users\eghav\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
exit status 1
Error compiling for board ESP32S3 Dev Module.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-10:
The OneWire lib you use is not compatible to the changes made in latest Aruino core.
You have to ask the maintainer for an update.

AlainF-ing, 2022-04-30:
I have the same error. 
Program compile without error  on ESP32 core 2.0.2

bghavami, 2022-04-30:
I replaced the old OneWire library with OneWireNG library. It is not a one-to-one replacement, therefore, I had to totally rewrite the old code to make it work.  Make sure to remove all old include library references to <OneWire.h>, otherwise you'll continue getting the same error.

Jason2866, 2022-04-30:
We modified the Onewire lib to work with all esp and core 2.0.x. You can find it [here](https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreakerl)

AlainF-ing, 2022-05-01:
> We modified the Onewire lib to work with all esp and core 2.0.x. You can find it [here](https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreaker)

With this library my program compile without error. I am now waiting for my ESP32-S2 delivery to test all functions.

Jason2866, 2022-05-10:
@brentru no, since we do not use latest version. The change has to be adopted for the upstream version. Feel free to do.

VojtechBartoska, 2022-05-16:
It's an external library which is not allocated in this repository.

zekoli123, 2022-05-17:
ESP32S doesn't work in 2.0.2

kinematic, 2022-05-22:
on 2.0.2 it`s compiling, but on 2.0.3 doesn`t want

Jason2866, 2022-05-22:
I prepared a PR for OneWire lib. Since i am away for weeks i cant test it.
Should work. You can try. You find it [here](https://github.com/Jason2866/OneWire)

brentru, 2022-05-23:
@Jason2866  Could you check out the PR on this repository? I can test it on hardware for you since you're away. Please ping me in the PR so I can get notified.

Jason2866, 2022-05-23:
@brentru  What do you mean with checkout? I can do the PR no problem. I just do not have ESP32 devices around...

brentru, 2022-05-23:
I meant "create a new PR". I have ESP32's to test the PR for you while you're out.

petervis, 2022-05-23:
tested on esp32 it did not crash, but could not get address is 000000 from temp-device

petervis, 2022-05-24:
works perfect thanks

VojtechBartoska, 2022-05-25:
Can I consider this as solved?

Jason2866, 2022-05-25:
@VojtechBartoska  Yes, i did a PR in OneWire github which is merged solving this issue.
The OneWire code owner will release a new version if he gets positive feedbacks regarding the PR.
Anyways it is NOT a Arduino issue and can be closed.

petervis, 2022-05-25:
Yes it is working perfect now for the last days, so yes I think it is
solved😁😎👍

brentru, 2022-05-25:
@Jason2866 @VojtechBartoska  Tested on ESP32 and ESP32-S2/3.

closed 2022-05-25 by VojtechBartoska