espressif/arduino-esp32 #6564: Investigate possible optimizations of startup time

igrr, 2022-04-11:
### Related area

Default configuration (sdkconfig), initialization code

### Hardware specification

All ESP32 series chips

### Is your feature request related to a problem?

When building low-power devices which wake up, do some work, and go to sleep, it's important that wakeup time is small.

ESP-IDF docs provide some recommendations for reducing startup time: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html#improving-startup-time.

### Describe the solution you'd like

- [ ] investigate if these or other methods can be applied in arduino-esp32 project to reduce startup time
- [ ] add an example showing how to achieve low startup time and measure it
- [ ] add an automated test based on the above example, to detect significant regressions to startup time

### Describe alternatives you've considered

Refer users to ESP-IDF docs and ask them to use lib-builder or arduino-as-idf-component.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

jakabo27, 2025-02-20:
I see this is marked as complete but I can't find anything online about how to implement improved boot times, is there an example or documentation somewhere?