espressif/arduino-esp32 #6571: EthernetClientSecure

simogaspa84, 2022-04-12:
### Related area

Creating a class for Ethernet support of TLS based on WiFiClientSecure

### Hardware specification

ESP32

### Is your feature request related to a problem?

Yes the problem is raised when you want to add connectivity to cloud providers (for example google iot core)
but the libraries you find and example requires Wifi and TLS.
At the moment there are not working libs for ethernet with TLS for ESP32 under the Arduino framework.
Some machines inside production plants have no wifi connectivity and the only option is the cable.
We must provide this functionality for these cases

### Describe the solution you'd like

Creating a class for Ethernet support of TLS based on WiFiClientSecure

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

zekageri, 2022-04-27:
yes yes yes

felmue, 2022-07-23:
Hello @simogaspa84 

maybe helpful in your situation?

https://github.com/OPEnSLab-OSU/SSLClient#sslclient-with-ethernet
https://github.com/khoih-prog/EthernetWebServer_SSL

Thanks
Felix

ArminPP, 2022-08-19:
I'm eagerly awaiting this feature. :-) 👍 
It appears also in 2020:  https://github.com/espressif/arduino-esp32/issues/3902

BR, Armin

closed 2025-01-16 by Parsaabasi