espressif/arduino-esp32 #6573: RainMaker support for ESP32-S2, S3 and C3.

pedrominatel, 2022-04-13:
### Board

ESP32-S2, ESP32-S3 and ESP32-C3

### Device Description

ESP32-S2-DevKitC-1
ESP32-S3-DevKitC-1
ESP32-C3 Rust Board (custom board)

### Hardware Configuration

Nothing connected.

### Version

other

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

- Add RainMaker partition table scheme on each generic board.
- Change the provisioning method for ESP32-S2 to WiFi and for S3 and C3 BLE.

### Sketch

```cpp
RMakerSwitch.ino
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-13:
[rainmaker_partitions.patch.zip](https://github.com/espressif/arduino-esp32/files/8480956/rainmaker_partitions.patch.zip)
here is the patch for the partition scheme support on all boards

VojtechBartoska, 2022-04-13:
Probably related issue which can be closed by this one: https://github.com/espressif/arduino-esp32/issues/6435

pedrominatel, 2022-04-13:
Not sure @VojtechBartoska. I'll test on ESP32 and try to reproduce the #6435.

commit d8a99ed4 (2022-04-26): Added RainMaker support on Arduino IDE for ESP32-C3/S2/S3 (#6598)

* Added RainMaker support on Arduino IDE for ESP32-C3/S2/S3

Closes #6573
Note related to the issue #6435
closed 2022-04-26 by me-no-dev