espressif/arduino-esp32 #6578: `temp_sensor_set_config' missing

xinkiknix, 2022-04-13:
### Board

ESP32s3 Dev Mod

### Device Description

Nothing

### Hardware Configuration

Nothing

### Version

latest master

### IDE Name

Arduino IDE 2.0.0-rc5

### Operating System

Windows 11

### Flash frequency

stdandard

### PSRAM enabled

yes

### Upload speed

115200

### Description

Trying to compile wifimanager (“WiFiManager by tzapu”) basic example I get following error:

Linking everything together...
"C:\\Users\\Johan\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32s3-elf-g++" "-Wl,--Map=C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57/Basic.ino.map" "-LC:\\Users\\Johan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3-RC1/tools/sdk/esp32s3/lib" "-LC:\\Users\\Johan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3-RC1/tools/sdk/esp32s3/ld" "-LC:\\Users\\Johan\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3-RC1/tools/sdk/esp32s3/qspi_qspi" -T memory.ld -T sections.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib-time.ld -T esp32s3.peripherals.ld -mlongcalls -Wl,--cref -Wl,--gc-sections -fno-rtti -fno-lto -u _Z5setupv -u _Z4loopv -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u pthread_include_pthread_rwlock_impl -u ld_include_highint_hdl -u start_app -u start_app_other_cores -u __ubsan_include -Wl,--wrap=longjmp -u __assert_func -u vfs_include_syscalls_impl -Wl,--undefined=uxTopUsedPriority -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u newlib_include_assert_impl -u __cxa_guard_dummy -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 -Wl,--start-group "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\sketch\\Basic.ino.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFiManager\\WiFiManager.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFi.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiAP.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiClient.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiGeneric.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiMulti.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiSTA.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiScan.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiServer.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WiFi\\WiFiUdp.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\Update\\HttpsOTAUpdate.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\Update\\Updater.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WebServer\\Parsing.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WebServer\\WebServer.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\WebServer\\detail\\mimetable.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\DNSServer\\DNSServer.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\FS\\FS.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-sketch-89016A4D08A82565486C8B6FD84E8D57\\libraries\\FS\\vfs_api.cpp.o" "C:\\Users\\Johan\\AppData\\Local\\Temp\\arduino-core-cache\\core_ff83ea70cbb23f488e60ad73488eafc1.a" -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lasio -lbt -lcbor -lunity -lcmock -lcoap -lnghttp -lesp-tls -lesp_adc_cal -lesp_hid -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lesp_https_server -lesp_lcd -lprotobuf-c -lprotocomm -lmdns -lesp_local_ctrl -lsdmmc -lesp_serial_slave_link -lesp_websocket_client -lexpat -lwear_levelling -lfatfs -lfreemodbus -ljsmn -ljson -llibsodium -lmqtt -lopenssl -lperfmon -lspiffs -lusb -lulp -lwifi_provisioning -lbutton -lrmaker_common -ljson_parser -ljson_generator -lesp_schedule -lesp_rainmaker -lqrcode -lws2812_led -lesp-dsp -lesp-sr -lesp32-camera -lesp_littlefs -lfb_gfx -lasio -lcbor -lcmock -lunity -lcoap -lesp_lcd -lesp_websocket_client -lexpat -lfreemodbus -ljsmn -llibsodium -lperfmon -lusb -lesp_adc_cal -lesp_hid -lfatfs -lwear_levelling -lopenssl -lesp_rainmaker -lesp_local_ctrl -lesp_https_server -lwifi_provisioning -lprotocomm -lbt -lbtdm_app -lprotobuf-c -lmdns -lrmaker_common -lmqtt -ljson_parser -ljson_generator -lesp_schedule -lqrcode -larduino_tinyusb -lcat_face_detect -lhuman_face_detect -lcolor_detect -lmfn -ldl -lwakenet -lhufzip -lesp_audio_front_end -lesp_audio_processor -lmultinet -lesp-sr -lwakenet -lhufzip -lesp_audio_front_end -lesp_audio_processor -lmultinet -ljson -lspiffs -ldl_lib -lc_speech_features -lesp_tts_chinese -lvoice_set_xiaole -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_t [... cut ...]
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:   
}
```

### Debug Message

```plain
not relevant
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-13:
1) You need to run tools/get.py after pulling new versions of the dev branch.
1) We don't fix third party libraries here.
2) temp_sensor_* should only apply to esp32, and is not part of WiFiManager.

Jason2866, 2022-04-14:
S3 temp sensor not yet supported.

VojtechBartoska, 2022-04-14:
We are aware of this, will let you know when it's added.

xinkiknix, 2022-04-14:
Is there a way I can work around this? I do not currently require the temp support.

lbernstone, 2022-05-30:
Edit WiFiManager.cpp.  Remark out lines 2175 & 2176 (search for temperature).