espressif/arduino-esp32 #6579: LittleFS Mount Failure on ESP32-C3

brentru, 2022-04-14:
### Board

ESP32-C3

### Device Description

Adafruit QT Py ESP32-C3

### Hardware Configuration

Nothing attached.

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Attempts to mount a LittleFS filesystem fails on ESP32-C3...

I am using the FS read PR (https://github.com/espressif/arduino-esp32/pull/6569) as well.

LittleFS configuration settings:
```
[LittleFS] data   : /Users/brent/Downloads/esp32_esp8266_test_littlefs/data
[LittleFS] offset : 0
[LittleFS] start  : 2686976
[LittleFS] size   : 1472
[LittleFS] page   : 256
[LittleFS] block  : 4096
->/secrets.json
[LittleFS] upload : /var/folders/k6/g066g1tn567fn8mj2p42x4240000gn/T/arduino_build_941002/esp32_esp8266_test_littlefs.littlefs.bin
[LittleFS] address: 2686976
[LittleFS] port   : /dev/cu.usbmodem1433301
[LittleFS] speed  : 115200
[LittleFS] mode   : dout
[LittleFS] freq   : 80m
```

### Sketch

```cpp
/* Checks and writes out the files on a LittleFS filesystem on ESP8266 and ESP32 platforms
   This sketch only performs READ operations on the LittleFS filesystem and should not modify the filesystem's contents.

   NOTE: The LittleFS image must have already been uploaded prior to using this sketch.
*/

#include <Arduino.h>
#include <FS.h>
#include <LittleFS.h>
#include <ArduinoJson.h>

StaticJsonDocument<512> jsonDoc; ///< JSON document

// Via https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/LITTLEFS_test.ino
void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    #ifdef ARDUINO_ARCH_ESP8266
    File root = fs.open(dirname, "r");
    #else
    File root = fs.open(dirname);
    #endif

    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                #ifdef ARDUINO_ARCH_ESP8266
                listDir(fs, file.fullName(), levels -1);
                #else
                listDir(fs, file.path(), levels -1);
                #endif
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

// via https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/LITTLEFS_test.ino
void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\r\n", path);

    #ifdef ARDUINO_ARCH_ESP8266
    File file = fs.open(path, "r");
    #else
    File file = fs.open(path);
    #endif

    if(!file || file.isDirectory()){
        Serial.println("- failed to open file for reading");
        return;
    }

    Serial.println("- read from file:");
    while(file.available()){
        Serial.write(file.read());
    }
    Serial.println("");
    file.close();
}

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }

  // attempt to mount LittleFS filesystem
  Serial.print("Mounting LittleFS filesystem...");
  if(!LittleFS.begin()){
    Serial.println("ERROR: LittleFS Mount Failed!");
    return;
  }
  Serial.println("Mounted!");

  // list everything on the filesystem '/' directory where secrets is expected
  listDir(LittleFS, "/", 3);

  // check if WipperSnapper's `secrets.json` file exists on filesystem
  if (!LittleFS.exists("/secrets.json")) {
    Serial.println("ERROR: Unable to find secrets.json file on LittleFS filesystem!");
    while (1); // wait forever
  }

  // read out the secrets.json file in plain-text
  readFile(LittleFS, "/secrets.json");

  // Test we can open the secrets file using ArduinoJSON
  File secretsFile = LittleFS.open("/secrets.json", "r");
  if (!secretsFile) {
    Serial.println("ERROR: Unable to open secrets.json file from the LittleFS filesystem!");
    while (1); // wait forever
  }

  // Test we can deserialize the secrets file using ArduinoJSON
  DeserializationError err = deserializeJson(jsonDoc, secretsFile);
  if (err) {
    Serial.print("ERROR: Failed to deserialize secrets.json file with code: ");
    Serial.println(err.c_str());
    while (1); // wait forever
  }

  // Test parsing username from secrets.json file
  const char *io_username = jsonDoc["io_username"];
  // error check against default values [ArduinoJSON, 3.3.3]
  if (io_username == nullptr) {
    Serial.println("ERROR: Failed to parse io_username!");
    while(1); // wait forever
  }

  // Get IO key from JSON
  const char *io_key = jsonDoc["io_key"];
  // error check against default values [ArduinoJSON, 3.3.3]
  if (io_key == nullptr) {
    Serial.println("ERROR: Failed to parse io_key!");
    while(1); // wait forever
  }

  // Parse SSID
  const char *network_type_wifi_native_network_ssid = jsonDoc["network_type_wifi_native"]["network_ssid"];
  if (network_type_wifi_native_network_ssid == nullptr) {
    Serial.println("ERROR: Failed to parse network_type_wifi_native_network_ssid!");
    while(1); // wait forever
  }

  // Parse SSID password
  const char *network_type_wifi_native_network_password = jsonDoc["network_type_wifi_native"]["network_password"];
  // error check against default values [ArduinoJSON, 3.3.3]
  if (network_type_wifi_native_network_password == nullptr) {
    Serial.println("ERROR: Failed to parse network_type_wifi_native_network_password!");
    while(1); // wait forever
  }

  Serial.println("Parsed `secrets.json` values...");
  Serial.print("\tio_username: "); Serial.println(io_username);
  Serial.print("\tio_key: "); Serial.println(io_key);
  Serial.print("\tWiFi SSID: "); Serial.println(network_type_wifi_native_network_ssid);
  Serial.print("\tWiFi Password: "); Serial.println(network_type_wifi_native_network_password);

  // close the file
  secretsFile.close();

  // clear the JSON document and release all memory from the memory pool
  jsonDoc.clear();

  // close fs
  LittleFS.end();

  Serial.println("DONE!");
}

void loop() {
  // no-op
}
```

### Debug Message

```plain
14:33:38.869 -> Mounting LittleFS filesystem..../components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
14:33:38.869 -> E (5935) esp_littlefs: mount failed,  (-84)
14:33:38.869 -> E (5936) esp_littlefs: Failed to initialize LittleFS
14:33:38.869 -> [  3023][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
14:33:38.869 -> ERROR: LittleFS Mount Failed!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brentru, 2022-04-14:
@me-no-dev  I noticed a `.skip` file within https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/.skip.esp32c3? Does LittleFS not work on ESP32-C3?

brentru, 2022-04-14:
Getting an identical error on a ESP32-S3 board.

brentru, 2022-04-15:
Tried this with the ESP32-C3-DevKitC-02 and did not get the same error, perhaps an issue with Native USB?

lbernstone, 2022-04-15:
That's an interesting question.  Do you have CDC enabled on boot when using the USB?  Try toggling it and see if that makes a difference.  There is another issue about something similar, and I am unable to reproduce the problem.  If we can narrow it down, that will help troubleshooting a lot.

brentru, 2022-04-18:
@lbernstone I've disabled USB CDC on Boot and switched to programming/output over serial, same issue occurs.

Jason2866, 2022-04-23:
You do not have a partition scheme with SPIFFS if you do not provide one. So it fails.
I added a partition scheme with SPIFFS and your example does work.
```
Mounting LittleFS filesystem...Mounted!
Listing directory: /
  FILE: .settings.lkg.SIZE: 4096
  FILE: .settings.SIZE: 4096
  FILE: _persist.json.SIZE: 2
ERROR: Unable to find secrets.json file on LittleFS filesystem!
```

brentru, 2022-05-02:
@Jason2866  I was using the default partition scheme with SPIFFS (4mb with SPIFFS). I switched to your partition.csv, it returns the same error.

Jason2866, 2022-05-02:
Have you used hardware serial with a USB serial converter or the inbuilt USBmodem?
I tried with my devboard and used the USBmodem and the Hardware Serial with the onboard USB-serial adapter. Both worked. I can not explain why it works for me and it is failing for you.

DmytroKorniienko, 2022-05-08:
@Jason2866
Same issue for me, if I using 2.0.3

Working config is:
```
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
```

Non-working:
```
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
```

> 09:49:32.790 > ./components/esp_littlefs/src/littlefs/lfs.c:1790:debug: Bad block at 0x0
> 09:49:32.790 > ./components/esp_littlefs/src/littlefs/lfs.c:1796:warn: Superblock 0x0 has become unwritable
> 09:49:32.790 > E (2417) esp_littlefs: Failed to format filesystem
> 09:49:32.790 > [  2370][E][LittleFS.cpp:119] format(): Formatting LittleFS failed! Error: -1
> 09:49:32.790 > [  2376][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
> 09:49:32.790 > E (2535) esp_littlefs: Partition already used
> 09:49:32.790 > E (2535) esp_littlefs: Failed to initialize LittleFS
> 09:49:32.790 > [  2483][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: 259
> 09:49:32.790 > ./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
> 09:49:32.790 > E (2640) esp_littlefs: mount failed,  (-84)
> 09:49:32.790 > E (2640) esp_littlefs: Failed to initialize LittleFS

Any suggestions?

Jason2866, 2022-05-08:
@DmytroKorniienko 
Try
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
```

DmytroKorniienko, 2022-05-08:
> @DmytroKorniienko Try
> 
> ```
> platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
> ```

This is first thing what I tried, but without success.

```
[env:esp32c3] ; 2mb without OTA
framework = arduino
board_build.filesystem = littlefs

platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
;platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip

; platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
; platform_packages =
;     framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
;     ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

; platform = espressif32
; platform_packages =
;     toolchain-riscv-esp@file://d:\toolchains\riscv32-esp-elf ; https://github.com/espressif/crosstool-NG
;     framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
;     platformio/tool-esptoolpy @ ~1.30100

board = esp32-c3-devkitm-1
board_build.flash_mode = dio

; board = esp32dev ; board hack
; board_build.mcu = esp32c3
board_build.partitions = esp32c3_2MiB.csv
board_build.f_cpu = 160000000L
build_flags =
    ${extra32.build_flags}
    ; -DCONFIG_ESP_SYSTEM_MEMPROT_FEATURE=""
    ; -DBOARD_HAS_PSRAM
    -DLED_BUILTIN=3 ; 3,4,5
    ; -mfix-esp32-psram-cache-issue
    ; Logging level: 1 = error, 2 = warning, 3 = info, 4 = debug, 5 = verbose
    -DCORE_DEBUG_LEVEL=4
```

```
#https://github.com/espressif/arduino-esp32/issues/4551
#https://docs.espressif.com/projects/esp-idf/en/v3.3.4/api-reference/system/ota.html#ota-data-partition
# Name         Type    SubType  Offset           Size    Flags
nvs,            data,  nvs,     0x9000,          0x4000,
otadata,        data,  ota,     0xd000,          0x2000,
phy_init,       data,  phy,     0xf000,          0x1000,
app0,           app,   ota_0,   0x10000,         0x170000,
spiffs,         data,  spiffs,  0x180000,        0x080000, 
```

or

```
#https://github.com/espressif/arduino-esp32/issues/4551
#https://docs.espressif.com/projects/esp-idf/en/v3.3.4/api-reference/system/ota.html#ota-data-partition
# Name         Type    SubType  Offset           Size    Flags
nvs,            data,  nvs,     0x9000,          0x5000,
otadata,        data,  ota,     0xe000,          0x2000,
app0,           app,   ota_0,   0x10000,         0x170000,
spiffs,         data,  spiffs,  0x180000,        0x080000, 
```

Same result:

> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
> PLATFORM: Espressif 32 (2.0.3) > Espressif ESP32-C3-DevKitM-1
> HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (custom) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 2.0.3
>  - tool-esptoolpy 1.30300.0 (3.3.0)
>  - tool-mkfatfs 2.0.1
>  - tool-mklittlefs 1.203.210628 (2.3)
>  - tool-mkspiffs 2.230.0 (2.30)
>  - toolchain-esp32ulp 1.22851.191205 (2.28.51)
>  - toolchain-riscv32-esp 8.4.0+2021r2-patch3
> 
> 11:19:08.162 > [ 21458][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
> 11:19:08.162 > [ 21559][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: 259
> 11:19:08.162 > [ 21659][E][LittleFS.cpp:119] format(): Formatting LittleFS failed! Error: -1
> 11:19:08.162 > [ 21660][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
> 11:19:08.162 > [ 21761][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: 259

[image]

DmytroKorniienko, 2022-05-08:
Update:
I tried module C3FN4 instead C3-2M (both from AI-thinker) and it work with same config without issues.... Hmm... Something hardware related?

Jason2866, 2022-05-08:
You did not wrote you have a module with 2MB flash. ;-)

DmytroKorniienko, 2022-05-08:
> You did not wrote you have a module with 2MB flash. ;-)

I checked flash size only by incident, actually I have different dev-modules :), but this 2Mb module work with 2.0.2 and previous FW versions with an 4Mb board or a board-hack, but at 2.0.3 - not, seems something changed in the LittleFS code and related to real flash size.

Jason2866, 2022-05-08:
Have you made a custom boards.json? There is none predefined for a 2MB Esp32c3
AND you have to define a valid partitions.csv too.
Afaik checks are more restrictive with 2.0.3.

DmytroKorniienko, 2022-05-10:
> Have you made a custom boards.json? There is none predefined for a 2MB Esp32c3 AND you have to define a valid partitions.csv too. Afaik checks are more restrictive with 2.0.3.

Yes, I made custom board.json for 2mb version and yes - partition table custom and valid as well. Currently all work again with your version of 2.0.3 but still not working with original. Why it stop working unexpected twi days ago - I don't know, but seems this wasn't related with littlefs issue. I will investigate deeper. Anyway - thank you.

lonerzzz, 2022-06-08:
I also have this same issue with the C3 module C3FN4. The issue appeared when I upgraded from version 2.0.2 to 2.0.3. It disappears when I revert version to 2.0.2.

I am using the Minimal SPIFFs configuration but have tried other configurations always getting the same issue when upgraded to 2.0.3.

Jason2866, 2022-06-09:
@lonerzzz Please post your partitions.csv

lonerzzz, 2022-06-09:
Thanks for the quick response.

I have attached the preferences.txt and two of the partition files that I have tried without success. The partition files seem to be the same in both versions 2.0.2 and 2.0.3 so I am guessing the issue is in software changes between the two versions. Let me know if you were looking for different files than what I provided. Just in case, I also upgraded from Arduino 1.18.12 to 1.18.19 and the issue remains.

[default.csv](https://github.com/espressif/arduino-esp32/files/8871824/default.csv)
[min_spiffs.csv](https://github.com/espressif/arduino-esp32/files/8871825/min_spiffs.csv)
[preferences.txt](https://github.com/espressif/arduino-esp32/files/8871826/preferences.txt)
.

P-R-O-C-H-Y, 2022-06-13:
Tested SPIFFS with ESP32C3-DevkitC_02 (4Mb flash) with default partition scheme on 2.0.3. 
Module is XXN4 and no issues for me.

lbernstone, 2022-06-14:
See #6572

P-R-O-C-H-Y, 2022-06-15:
@lonerzzz Can you try to set flash mode = QIO ?

lonerzzz, 2022-06-24:
Sorry for the delayed response. Changing the flash mode to QIO resolved the issue for me.

brentru, 2022-06-24:
@P-R-O-C-H-Y Also working for me when using flash mode QIO.

P-R-O-C-H-Y, 2022-06-24:
Thanks @lonerzzz and @brentru,
for now you can use QIO Flash mode for C3/S3. We are curently working on a fix for DIO Flash mode for mentioned chips.

brentru, 2022-06-24:
Thanks, @P-R-O-C-H-Y, pls keep me in the loop for when the fix is released (I'll need to update the arduino-cli script we use).

P-R-O-C-H-Y, 2022-06-24:
@brentru will add all Related issues to the PR, so it should notify you once there is a PR with fix.

Jason2866, 2022-07-01:
@P-R-O-C-H-Y Currently it depends how the Arduino libs are compiled. With [this](https://github.com/Jason2866/esp32-arduino-lib-builder/blob/main_idf441/configs/builds.json) the C3 works in mode dout.
We changed to upstream version and had bootlooping devices when doing a OTA upgrade from earlier version we used mode dout too to compile the libs.
Will the fix solve this issue too?

VojtechBartoska, 2022-07-11:
Same here, can this be closed @P-R-O-C-H-Y as fully solved?

P-R-O-C-H-Y, 2022-07-11:
Yes, fixed by PR #6910. Closing

closed 2022-07-11 by P-R-O-C-H-Y

aeonSolutions, 2023-05-04:
LittleFS does not allow a `"/"` basepath. when mounting 
 
Try changing to something else. For instance "/storage"  

Here's a code snippet that is working:

```
    if(true==fs.begin( false, "/storage", maxFiles , "storage")){ // !! Only allow one file to be open at a time instead of 10, saving 9x4 - 36KB of RAM          
      this->mserial->printStrln( "Drive init...done.");
    //this->onBoardLED->led[0] = this->onBoardLED->LED_GREEN;
    //this->onBoardLED->statusLED(100, 1);
    }else if(fs.format()){
        this->mserial->printStrln("Drive formated sucessfully!");
      //  this->onBoardLED->led[0] = this->onBoardLED->LED_RED;
      //  this->onBoardLED->led[0] = this->onBoardLED->LED_GREEN;
      //  this->onBoardLED->statusLED(100, 2);
        if(true==fs.begin( false, "/storage", maxFiles , "storage")){
          this->mserial->printStrln( "Drive init...done.");
       // this->onBoardLED->led[0] = this->onBoardLED->LED_GREEN;
       // this->onBoardLED->statusLED(100, 1);
        }else{
          this->mserial->printStrln("Drive Mount Failed");
          return false;
        }   

    }else{
        this->mserial->printStrln("Drive init Failed");
     //   this->onBoardLED->led[0] = this->onBoardLED->LED_RED;
     //   this->mserial->printStrln("here 1");
     //   this->onBoardLED->statusLED(100, 5);
     //   this->mserial->printStrln("here 2");
        return false;
    }
```

The code above can be found here:
https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries