espressif/arduino-esp32 #6580: [E][WiFiClient.cpp:258] connect(): socket error on fd 55, errno: 113, "Software caused connection abort"

sglvladi, 2022-04-14:
### Board

ESP32 LolinD32 Pro

### Device Description

ESP32 LolinD32 Pro on breadboard, connected to [LAN8720 Ethernet module](https://www.waveshare.com/lan8720-eth-board.htm)

### Hardware Configuration

GPIO 0, 14, 18, 19, 21, 22, 23, 25, 26, 27 attached to Ethernet board.

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

460800

### Description

I am trying to change between ethernet and WiFi while making POST requests to a device registration API. 

Initially, the ethernet cable is connected and the device posts successfully to the registration API though the ethernet interface. Then, the cable is plugged out and the device connects to a WiFi network. The issue shows up on consecutive POST requests after connecting to WiFi.

A check is added after a failed POST, to see if a GET to google.co.uk succeeds and indeed it does, irrespective of the interface change. 

### Sketch

```cpp
while(1){

	....

	Serial.println(payload);
	HTTPClient httpClient;
	httpClient.begin(client, url);
	httpClient.addHeader("Content-Type", "application/json");
	httpClient.addHeader("Content-Length", String(payload.length()));
	int httpCode = httpClient.POST(payload);
	String response = httpClient.getString(); // Get the response payload
	httpClient.end();	 // Close connection after receiving response form server

	...

	if (httpCode < 0)
	{
		// Check connection with google
		String google_url = "http://www.google.co.uk";
		Serial.print("Checking connection with Google...");
		httpClient.begin(client, google_url);
		int httpCode2 = httpClient.GET();
		httpClient.end();
		if (httpCode2 > 0)
		{
			Serial.println("OK!");
		}
	}
}
```

### Debug Message

```plain
...

{"mac":"98-F4-AB-78-79-4C","hw_version":"2.0.1","fw_version":"2.4.0","model":"MyModel"}
Sending registration request to http://api.xxxxxxxxxx.io:80/v1/hw/regauth..[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: api.xxxxxxxxxx.io port: 80 url: /v1/hw/regauth
[D][HTTPClient.cpp:579] sendRequest(): request type: 'POST' redirCount: 0

[D][HTTPClient.cpp:1125] connect():  connected to api.xxxxxxxxxx.io:80
[D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 400
[D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 38
[D][HTTPClient.cpp:603] sendRequest(): sendRequest code=400

[D][WiFiClient.cpp:514] connected(): Disconnected: RES: 0, ERR: 128
[D][HTTPClient.cpp:1394] writeToStreamDataBlock(): connection closed or file end (written: 38).
[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 24 - ETH_DISCONNECTED
[WiFi-event] event: 24
ETH Disconnected
Connection to network has been lost. Attempting to reconnect....
Waiting for ethernet connection...
Ethernet didn't get IP! Starting WiFi Manager..

...

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4
[WiFi-event] event: 4
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.118, MASK: 255.255.255.0, GW: 192.168.1.1

... 

{"mac":"98-F4-AB-78-79-4C","hw_version":"2.0.1","fw_version":"2.4.0","model":"MyModel"}
Sending registration request to http://api.xxxxxxxxxx.io:80/v1/hw/regauth..

[D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: api.xxxxxxxxxx.io port: 80 url: /v1/hw/regauth
[D][HTTPClient.cpp:579] sendRequest(): request type: 'POST' redirCount: 0

[E][WiFiClient.cpp:258] connect(): socket error on fd 55, errno: 113, "Software caused connection abort"
[D][HTTPClient.cpp:1118] connect(): failed connect to api.xxxxxxxxxx.io:80
[W][HTTPClient.cpp:1417] returnError(): error(-1): connection refused
[W][HTTPClient.cpp:1417] returnError(): error(-4): not connected
[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: www.google.co.uk port: 80 url: 
[D][HTTPClient.cpp:579] sendRequest(): request type: 'GET' redirCount: 0

[D][HTTPClient.cpp:1125] connect():  connected to www.google.co.uk:80
[D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 404
[D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 1564
[D][HTTPClient.cpp:603] sendRequest(): sendRequest code=404

[D][HTTPClient.cpp:378] disconnect(): still data in buffer (1564), clean up.

[D][HTTPClient.cpp:387] disconnect(): tcp stop
OK!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-19:
is it possible that the service you are trying to connect to is refusing the connection? Can you try checking google through ETH as well? Then switch and try again?

VojtechBartoska, 2022-05-04:
is this still valid @sglvladi?

sglvladi, 2022-05-05:
@VojtechBartoska I am still unclear about how to clear the cached dns entries

tsctrl, 2022-05-24:
hi @sglvladi sorry to ask here, where does the pin gpio 14 goes to the LAN8720 pin? did you put nc pin? what the hack you did on the lan module to make it to work? make connection to the enable pin? any other inverter ic used?

VojtechBartoska, 2022-10-10:
I'm closing this issue as expired due to no answer. If needed, you can reopen it. Thanks for understanding.

closed 2022-10-10 by VojtechBartoska