espressif/arduino-esp32 #6581: ESP32-S3 native USB CDC stops working after light sleep

Miraculix200, 2022-04-15:
### Board

ESP32-S3-WROOM-1 N8R2

### Device Description

Custom PCB

### Hardware Configuration

GPIO 18 and 19 are connected to USB connector

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When using the Arduino IDE serial monitor or Putty connected to USB (with Hardware CDC and JTAG enabled) at 115200 baud, ESP_LOGD output will stop after light sleep. Doing ESP.restart() will not enable the output again. Only after a hard reset the ESP32-S3 outputs data over USB again.

### Sketch

```cpp
#include "esp32s3/rom/rtc.h"
#include "esp_log.h"

void setup()
{
    while (millis() < 5000)
        delay(100);

    ESP_LOGD("abc", "hello");
    delay(1000);

    for (int i = 0; i < 10; i++) {
        esp_sleep_enable_timer_wakeup(600 * 1000);
        esp_light_sleep_start();
        delay(1);
    }
}

void loop()
{
    ESP_LOGD("abc", "hello");
    delay(1000);
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-20:
@SuGlider Can you please help with triage? Maybe it's a duplicate of already opened USB CDC issues? Thanks

Miraculix200, 2022-04-20:
It's similar to this issue, but I failed to fix it using ` phy_bbpll_en_usb(true);`
 https://github.com/espressif/arduino-esp32/issues/6264

me-no-dev, 2022-05-13:
What happens if you sleep for a shorter time?

What probably happens is that your computer fails to communicate with the ESP during light sleep, so it stops asking it for data (with USB, it's the HOST that asks the questions, the DEVICE can not just send data on it's own). When you hard-reset the ESP, the USB is re-enumerated and  starts working again. The solution might be to call an extra function in the future, or another Arduino functions that will re-enumerate the USB on wakeup from light sleep.

WereCatf, 2022-11-03:
This is months old now and I don't know if this is relevant, but I actually had this same issue with an STM32-board using USB-CDC and it was caused by, indeed, what @me-no-dev there said. What worked for me was as simple as stopping the USB-peripheral, pulling both USB-data lines down to GND for a couple of milliseconds, then restarting the USB-peripheral. I would assume the same would work on the ESP32-S2/-S3/-C3.

N0ury, 2023-04-22:
Found this in USB Serial/JTAG Controller Console chapter of Espressif documentation:
`If the application enters deep sleep mode, the USB Serial/JTAG device will disappear from the system.`
Read it it's more detailed about light sleep.
I think this issue should be closed.

SuGlider, 2023-04-29:
Closing the issue as per the description in the documentation. This is a limitation od the USB and deep/light sleeping.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-serial-jtag-console.html#limitations

closed 2023-04-29 by SuGlider

jstockdale, 2024-02-21:
@WereCatf did you find a way to do this purely on the software / firmware side? I can replicate your workaround (touching ground to D+ alone seems to do it in my case) but when I tried to use the following snippet it didn't work. Do I also have to change the io_mux or other settings?

```
# This does not work to reset the USB bus, but pulling D+ to ground does...
            gpio_reset_pin(GPIO_NUM_19);
            gpio_reset_pin(GPIO_NUM_20);
            gpio_set_direction(GPIO_NUM_19, GPIO_MODE_OUTPUT);
            gpio_set_direction(GPIO_NUM_20, GPIO_MODE_OUTPUT);
            gpio_set_level(GPIO_NUM_19, 0);
            gpio_set_level(GPIO_NUM_20, 0);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            gpio_set_direction(GPIO_NUM_19, GPIO_MODE_INPUT);
            gpio_set_direction(GPIO_NUM_20, GPIO_MODE_INPUT);
            gpio_reset_pin(GPIO_NUM_19);
            gpio_reset_pin(GPIO_NUM_20);
```

I was able to get it to work by using GPIOs to pull down the usb data lines (by soldering a small wire to the pads) but I'd very much prefer to reset usb by messing with the appropriate registers.

IhorNehrutsa, 2024-04-03:
```
...
ESP_ERROR_CHECK(esp_light_sleep_start());
...
uint32_t save = REG_READ(RTC_CNTL_USB_CONF_REG);
SET_PERI_REG_MASK(RTC_CNTL_USB_CONF_REG, RTC_CNTL_USB_PAD_PULL_OVERRIDE);
SET_PERI_REG_MASK(RTC_CNTL_USB_CONF_REG, RTC_CNTL_USB_DP_PULLDOWN);
vTaskDelay(5 / portTICK_PERIOD_MS);
REG_WRITE(RTC_CNTL_USB_CONF_REG, save ) ;
...
```

Rob58329, 2024-09-23:
>ESP32-S3 native USB CDC stops working after light sleep

This is also an issue for the ESP32-C3 with "USB CDC On Boot: Enabled" (when using a Windows10 PC).

The best(?) solution I have found is to do:
```
 Serial.flush();
 Serial.end();
 esp_light_sleep_start();
 Serial.begin(115200);

```
This causes the Arduino-Serial-Monitor to be disconnected at "`Serial.end()`", which you then have to close and then re-open shortly after you hear the USB connection re-starting at the "`Serial.begin(115200)`".  This is similar to what you have to do if using the ESP32-C3 "`esp_deep_sleep_start()`" function.

This works because recent versions of github.com/espressif/arduino-esp32 (eg. v23Sep24) for the ESP32-C3 use 
    ` extern "C" void phy_bbpll_en_usb(bool en);` & `phy_bbpll_en_usb(true);`
in "bool wifiLowLevelInit()"
(and also something different in "`Serial.end()`").