espressif/arduino-esp32 #6584: touchAttachInterrupt kills ULP program

fg-uulm, 2022-04-15:
### Board

ESP32 Dev Module (AZ Delivery)

### Device Description

see above

### Hardware Configuration

External LED at GPIO 2

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

macOS Monterey

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921000

### Description

See attached sketch for a minimal example - I've got a small ULP program to do some crude PWM as a control signal for a motor (ignore the variable naming, please) which is reversible, and needs 50% / 128 as a "stop" value provided by the PWM. This means if the PWM ever stops, the motor starts running. I created the ULP program to generate the PWM also while the ESP sleeps periodically, and everything works as expected so far (PWM continuously running after calling `ulp_start()`, in sleep and awake mode).

However, I tried to add touch wakeup to the sketch as well, and the `touchAttachInterrupt` function call always kills my ULP program / PWM for a short time, but long enough to be noticeable in the motor rpm (I'd guess around 100-200ms). After another call to `ulp_start()` everything works as expected (including touch) until the next wakeup. However even when calling `ulp_start` directly after `touchAttachInterrupt` the small break still exists.

The pins are not related, and the touch wakeup should be only RTC related, so the ULP shouldn't be affected in my opinion? I'm willing to debug this more in-depth, however I'd need some hint where to start. A quick glance into the soure of `touchAttachInterrupt` didn't give me any insights however.

### Sketch

```cpp
#include <Arduino.h>
#include "esp32/ulp.h"
#include "driver/rtc_io.h"

#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  5        /* Time ESP32 will go to sleep (in seconds) */

uint8_t lcdBrightness = 10; // (0-255)

void ulp_start(void) {
  // Slow memory initialization
  memset(RTC_SLOW_MEM, 0, 8192);
  // if LED is connected to GPIO2 (specify by +14)
  const gpio_num_t ledPWMPin = GPIO_NUM_2;
  const int ledPWMBit = RTCIO_GPIO2_CHANNEL + 14;
  // GPIOx initialization (set to output and initial value is 0)
  rtc_gpio_init(ledPWMPin);
  rtc_gpio_set_direction(ledPWMPin, RTC_GPIO_MODE_OUTPUT_ONLY);
  rtc_gpio_set_level(ledPWMPin, 0);
  // Define ULP program
  const ulp_insn_t ulp_prog[] = {
    M_LABEL(1),
    I_WR_REG(RTC_GPIO_OUT_REG, ledPWMBit, ledPWMBit, 1), // on
    I_DELAY(lcdBrightness * 100),
    I_WR_REG(RTC_GPIO_OUT_REG, ledPWMBit, ledPWMBit, 0), // off
    I_DELAY(25500 - lcdBrightness * 100),
    M_BX(1),
  };
  
  // Run ULP program
  size_t size = sizeof(ulp_prog) / sizeof(ulp_insn_t);
  ulp_process_macros_and_load(0, ulp_prog, &size);
  ulp_run(0);
}

void callback(){
  //placeholder callback function
}

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  ulp_start(); //led is on now
  Serial.println("ULP started");
  delay(2000); //just to show that led stays on / ulp is running
  touchAttachInterrupt(T9, callback, 40);
  Serial.println("Touch ISR attached");
  delay(2000); //led is off again for 2000ms + x - even without this led will be off for a short time (approx. 100ms)
  ulp_start(); //restart led / ulp
  Serial.println("ULP re-started");

  esp_sleep_enable_touchpad_wakeup();
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  //esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  
  Serial.println("ESP sleep config done - Going to deep sleep in 1 sec");
  delay(1000);
  esp_deep_sleep_start();
  //led will stay on until wakeup
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
No error message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-06-27:
Closing this issue as expired, if needed you can reopen it. Thanks for understanding.

closed 2022-06-27 by VojtechBartoska

fg-uulm, 2022-06-27:
Hi, I'm very sorry, I missed out on your comments somehow. It seems as the new version has indeed fixed the issue! Thanks!