espressif/arduino-esp32 #6588: I2C write error

NathanKuo0821, 2022-04-18:
### Board

Esp32 devkit_v4

### Device Description

Esp32 devkit_v4 

### Hardware Configuration

GPIO 32 and 33 connect to I2c LCD device.

[**WORK**]
[image]

[**NG**]
[image]

### Version

other

### IDE Name

VSC platformIO

### Operating System

Windows 10

### Flash frequency

40Mhz 

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hi All,

I define the two I2C to control M5UnitLCD in ESP32.

#define I2C1_SDA 15
#define I2C1_SDL 16
#define I2C2_SDA 32
#define I2C2_SDL 33

Now there is a problem that when I use I2C1 to control the LCD everything works fine, but when I switch to I2C2, an error like the following debug message will appear, can anyone help?

Nathan Kuo

### Sketch

```cpp
//I2C BUS
#define I2C1_SDA 15
#define I2C1_SDL 16
#define I2C2_SDA 32
#define I2C2_SDL 33

/* I2C Bus */
TwoWire I2C1 = TwoWire(0); //I2C1 bus
TwoWire I2C2 = TwoWire(1); //I2C2 bus

// LCD
#include <M5UnitLCD.h>
M5UnitLCD display (I2C2_SDA,I2C2_SDL,400000); 
void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200, SERIAL_8N1, RXD1, TXD1);
  Serial1.setTimeout(300);
  Serial.println();

 I2C2.begin(I2C2_SDA,I2C2_SDL);

   display.begin();

  if (display.isEPD())
  {
    display.setEpdMode(epd_mode_t::epd_fastest);
    display.invertDisplay(true);
    display.clear(TFT_WHITE);
  }
  if (display.width() < display.height())
  {
    display.setRotation(display.getRotation() ^ 1);
  }

  // width=240 , height=135
  log_d("width=%d , height=%d",display.width(),display.height());

  display.setColorDepth(1); // mono color
  display.setFont(&fonts::Font4);
  display.setTextWrap(false);
  display.setTextSize(3);
  display.setTextColor(TFT_BLACK);

  display.fillScreen(WHITE);

  display.drawString("Ready" , 0, (display.height()/3));
}
```

### Debug Message

```plain
[D][esp32-hal-i2c.c:1344] i2cProcQueue():  Busy Timeout start=0xe5, end=0x117, =50, max=50 error=3
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbe118
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60027000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb86cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=1
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=3
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8754 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8788
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8730
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[D][esp32-hal-i2c.c:1353] i2cProcQueue():  Gross Timeout Dead start=0x168, end=0x19a, =50, max=50 error=3
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbe118
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60027000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb86cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=1
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=3
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8754 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8788
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8730
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=1
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

NathanKuo0821, 2022-04-18:
More infromation 
If I switch the pin as below, the TwoWire(0) gpio15/16 and gpio 32/33 still working , but the TwoWire(1) neither gpio15/16 nor gpio32/33 work , Does anyone know why?

//#define I2C1_SDA 15 
//#define I2C1_SDL 16
//#define I2C2_SDA 32
//#define I2C2_SDL 33

#define I2C1_SDA 32
#define I2C1_SDL 33
#define I2C2_SDA 15
#define I2C2_SDL 16

/* I2C Bus */
TwoWire I2C1 = TwoWire(0); //I2C1 bus
TwoWire I2C2 = TwoWire(1); //I2C2 bus

mrengineer7777, 2022-04-18:
Your defines aren't applied until you call begin() or setPins().

`I2C2.begin(I2C2_SDA,I2C2_SDL);` should in theory work.  What version of Arduino core are you targeting?

Here are the current prototypes from Wire.h:

```
    bool setPins(int sda, int scl);
    bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
    bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
```

NathanKuo0821, 2022-04-20:
I used the framework-arduinoespressif32 to develop my project , the version info as below

 "name": "framework-arduinoespressif32",
  "version": "3.10006.210326",

I tried to setPins() before call begin() but no luck, i need someone's help.
Thanks.

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200, SERIAL_8N1, RXD1, TXD1);
  Serial1.setTimeout(300);
  Serial.println();

 **I2C2.setPins(I2C2_SDA,I2C2_SDL);**
 I2C2.begin(I2C2_SDA,I2C2_SDL);

...

mrengineer7777, 2022-04-20:
Ah.  You are using Arduino core 1.0.6.  That is the default in PIO, but is very old and unsupported.  Some day they will update to 2.03.

Modify your platformio.ini file to target an updated platform.  Here's mine:

```
[env]
board = featheresp32
monitor_speed = 115200
build_flags = 
	-DCORE_DEBUG_LEVEL=5

[env:featheresp32ard]
framework = arduino
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
```

Note this is a custom build by TASMOTA.  There are lots of web results about how to target different Arduino versions.

NathanKuo0821, 2022-04-21:
> Ah. You are using Arduino core 1.0.6. That is the default in PIO, but is very old and unsupported. Some day they will update to 2.03.
>
> Modify your platformio.ini file to target an updated platform. [...]

Thanks for your advice.
I'm using the tasmota package on our board and it won't start , so I upgraded the code from https://github.com/platformio/platform-espressif32.git.
But the issue still exists, is there a problem with esp32 i2c2? Because I have no problem using gpio 15/16 or 32/36 for i2c1, and vice versa for i2c2.

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
board_build.flash_mode = qio
board_build.f_flash = 40000000L
board_build.partitions = ota_4MB.csv
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-mfix-esp32-psram-cache-issue
extra_scripts = pre:extra_script.py
lib_deps = 
	adafruit/DHT sensor library@^1.4.0
	adafruit/Adafruit Unified Sensor@^1.1.4
	fhessel/esp32_https_server@^1.0.0
	m5stack/M5GFX@^0.0.18
	wifwaf/TCA9548A@^1.1.3
	bblanchon/ArduinoJson@^6.19.4

Thank you sir!

NathanKuo0821, 2022-04-27:
I try using Wire instead of I2C1 and Wire1 instead of I2C2 , the issue fixed.
Thanks.

`
/* I2C Bus */
//TwoWire I2C1 = TwoWire(0); //I2C1 bus
//TwoWire I2C2 = TwoWire(1); //I2C2 bus
//I2C1.begin(I2C1_SDA,I2C1_SDL);
//I2C2.begin(I2C2_SDA,I2C2_SDL);

Wire.begin(I2C1_SDA,I2C1_SDL);
Wire1.begin(I2C2_SDA,I2C2_SDL);
`

closed 2022-04-27 by NathanKuo0821