espressif/arduino-esp32 #6589: ESP32 C3 PDM I2S  Runtime Fail

mrburns42, 2022-04-18:
### Board

M5 Stamp C3

### Device Description

I2C MEM Microphone, SPM1423.   This is connected to GPIO 0 and GPIO 1.  

### Hardware Configuration

Nothing other than the MEM microphone already described.

### Version

latest master

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80MHZ

### PSRAM enabled

no

### Upload speed

115200

### Description

I am trying to use a MEM microphone on an M5 Stamp C3.   The microphone is the only HW attached to the C3.  
Code builds but fails with a boot loop at run time.   There is a message to the serial port:

E (120) I2S: i2s_check_cfg_validity(1490): PDM does not support RX on this chip

This message an a core dump happen repeatedly as the processor attempts to start over and over again. 

However,  I do not think this is a valid statement.   I looked at the Technical Reference Manual 
https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf

It states  " ESP32-C3 I2S supports both TDM RX mode and PDM RX mode. Set I2S_RX_TDM_EN to enable TDM RX
mode, or set I2S_RX_PDM_EN to enable PDM RX mode."   [Section 27.9.1]

### Sketch

```cpp
void i2sInit(){

   i2s_config_t i2s_config = {

    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM),

    .sample_rate =  44100,

    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB

    .channel_format = I2S_CHANNEL_FMT_ALL_RIGHT,

    .communication_format = I2S_COMM_FORMAT_I2S,

    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,

    .dma_buf_count = 2,

    .dma_buf_len = 128,

   };
```

### Debug Message

```plain
E (120) I2S: i2s_check_cfg_validity(1490): PDM does not support RX on this chip
```

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-18:
You should ask esp-idf related questions at https://github.com/espressif/esp-idf/issues.  However, the [PDM recorder example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_audio_recorder_sdcard) does indicate that esp32 and esp32-s3 are the supported platforms.

mrburns42, 2022-04-19:
Sorry for posting here.   I don't really understand any of the hierarchy of where these issues should be posted.  I opened an issue in esp-idf.    Thank you.

closed 2022-04-19 by mrburns42

lbernstone, 2022-04-19:
Includes that are in tools/sdk/espxx/include folder, for example [i2s.h](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32c3/include/driver/include/driver/i2s.h) are part of esp-idf (aka, the sdk).  Headers in arduino, either the core folder or the libraries, are part of this code base.