espressif/arduino-esp32 #6590: esp32 wifi station dosn't got an IPv6 address

luckyjeck, 2022-04-19:
### Board

NodeMCU-32S

### Device Description

NodeMCU-32S,plain module

### Hardware Configuration

nothing else.

### Version

v2.0.2

### IDE Name

Arduion 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

esp32 wifi station dosn't got an IPv6 address

when pc and esp32 connected to the same WIFI Router,
the pc got an IPv6 address correctly,
but esp32 wifi station dosn't got an IPv6 address.
how to fix this issue ? 

### Sketch

```cpp
#include "WiFi.h"

#define STA_SSID "**********"
#define STA_PASS "**********"
#define AP_SSID  "esp32-v6"

static volatile bool wifi_connected = false;

WiFiUDP ntpClient;

void wifiOnConnect(){
    Serial.println("STA Connected");
    Serial.print("STA IPv4: ");
    Serial.println(WiFi.localIP());
    
    Serial.print("STA IPv6: ");
    Serial.println(WiFi.localIPv6());

    ntpClient.begin(2390);
}

void wifiOnDisconnect(){
    Serial.println("STA Disconnected");
    delay(1000);
    WiFi.begin(STA_SSID, STA_PASS);
}

void wifiConnectedLoop(){
  //lets check the time
  const int NTP_PACKET_SIZE = 48;
  byte ntpPacketBuffer[NTP_PACKET_SIZE];

  IPAddress address;
  WiFi.hostByName("time.nist.gov", address);
  memset(ntpPacketBuffer, 0, NTP_PACKET_SIZE);
  ntpPacketBuffer[0] = 0b11100011;   // LI, Version, Mode
  ntpPacketBuffer[1] = 0;     // Stratum, or type of clock
  ntpPacketBuffer[2] = 6;     // Polling Interval
  ntpPacketBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  ntpPacketBuffer[12]  = 49;
  ntpPacketBuffer[13]  = 0x4E;
  ntpPacketBuffer[14]  = 49;
  ntpPacketBuffer[15]  = 52;
  ntpClient.beginPacket(address, 123); //NTP requests are to port 123
  ntpClient.write(ntpPacketBuffer, NTP_PACKET_SIZE);
  ntpClient.endPacket();

  delay(1000);
  
  int packetLength = ntpClient.parsePacket();
  if (packetLength){
    if(packetLength >= NTP_PACKET_SIZE){
      ntpClient.read(ntpPacketBuffer, NTP_PACKET_SIZE);
    }
    ntpClient.flush();
    uint32_t secsSince1900 = (uint32_t)ntpPacketBuffer[40] << 24 | (uint32_t)ntpPacketBuffer[41] << 16 | (uint32_t)ntpPacketBuffer[42] << 8 | ntpPacketBuffer[43];
    //Serial.printf("Seconds since Jan 1 1900: %u\n", secsSince1900);
    uint32_t epoch = secsSince1900 - 2208988800UL;
    //Serial.printf("EPOCH: %u\n", epoch);
    uint8_t h = (epoch  % 86400L) / 3600;
    uint8_t m = (epoch  % 3600) / 60;
    uint8_t s = (epoch % 60);
    Serial.printf("UTC: %02u:%02u:%02u (GMT)\n", h, m, s);
  }

  delay(9000);
}

void WiFiEvent(WiFiEvent_t event){
    switch(event) {

        case ARDUINO_EVENT_WIFI_AP_START:
            //can set ap hostname here
            WiFi.softAPsetHostname(AP_SSID);
            //enable ap ipv6 here
            WiFi.softAPenableIpV6();
            break;

        case ARDUINO_EVENT_WIFI_STA_START:
            //set sta hostname here
            WiFi.setHostname(AP_SSID);
            break;
        case ARDUINO_EVENT_WIFI_STA_CONNECTED:
            //enable sta ipv6 here
            WiFi.enableIpV6();
            break;
        case ARDUINO_EVENT_WIFI_STA_GOT_IP6:
            Serial.print("STA IPv6: ");
            Serial.println(WiFi.localIPv6());
            break;
        case ARDUINO_EVENT_WIFI_AP_GOT_IP6:
            Serial.print("AP IPv6: ");
            Serial.println(WiFi.softAPIPv6());
            break;
        case ARDUINO_EVENT_WIFI_STA_GOT_IP:
            wifiOnConnect();
            wifi_connected = true;
            break;
        case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
            wifi_connected = false;
            wifiOnDisconnect();
            break;
        default:
            break;
    }
}

void setup(){
    Serial.begin(115200);
    WiFi.disconnect(true);
    WiFi.onEvent(WiFiEvent);
    WiFi.mode(WIFI_MODE_APSTA);
    WiFi.softAP(AP_SSID);
    WiFi.begin(STA_SSID, STA_PASS);
}

void loop(){
    if(wifi_connected){
        wifiConnectedLoop();
    }
    while(Serial.available()) Serial.write(Serial.read());
}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
WiFi Event ID: 3
STA Connected
STA IPv4: 172.20.10.2
STA IPv6: 0000:0000:0000:0000:0000:0000:0000:0000
WiFi connected
IP address: 
172.20.10.2
UTC: 01:49:32 (GMT)
UTC: 01:49:42 (GMT)
UTC: 01:49:52 (GMT)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

srochapimenta, 2022-04-20:
Looks like your have problems with DHCP-V6, for me it works great, with few changes... 

Note: GOT_IP6 firing two times, one for AP_START and other for STA_CONNECTED.

Debug 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 20 - GOT_IP6
STA IPv6: 0000:0000:0000:0000:0000:0000:0000:0000
AP IPv6: fe80:0000:0000:0000:32ae:a4ff:fe02:e489
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.27.50, MASK: 255.255.255.0, GW: 192.168.27.1
UTC: 19:59:19 (GMT)
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 20 - GOT_IP6
STA IPv6: fe80:0000:0000:0000:32ae:a4ff:fe02:e488
AP IPv6: fe80:0000:0000:0000:32ae:a4ff:fe02:e489
UTC: 19:59:29 (GMT)

```
#include <WiFi.h>  // CHANGED!!!

#define STA_SSID "**********"
#define STA_PASS "**********"
#define AP_SSID  "esp32-v6"

static volatile bool wifi_connected = false;

WiFiUDP ntpClient;

void wifiOnConnect(){
    Serial.println("STA Connected");
    //Serial.print("STA IPv4: ");   // COMMENT OUT
    //Serial.println(WiFi.localIP()); // COMMENT OUT
    
    //Serial.print("STA IPv6: ");  // COMMENT OUT
    //Serial.println(WiFi.localIPv6()); // COMMENT OUT

    ntpClient.begin(2390);
}

void wifiOnDisconnect(){
    Serial.println("STA Disconnected");
    delay(1000);
    WiFi.begin(STA_SSID, STA_PASS);
}

void wifiConnectedLoop(){
  //lets check the time
  const int NTP_PACKET_SIZE = 48;
  byte ntpPacketBuffer[NTP_PACKET_SIZE];

  IPAddress address;
  WiFi.hostByName("time.nist.gov", address);
  memset(ntpPacketBuffer, 0, NTP_PACKET_SIZE);
  ntpPacketBuffer[0] = 0b11100011;   // LI, Version, Mode
  ntpPacketBuffer[1] = 0;     // Stratum, or type of clock
  ntpPacketBuffer[2] = 6;     // Polling Interval
  ntpPacketBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  ntpPacketBuffer[12]  = 49;
  ntpPacketBuffer[13]  = 0x4E;
  ntpPacketBuffer[14]  = 49;
  ntpPacketBuffer[15]  = 52;
  ntpClient.beginPacket(address, 123); //NTP requests are to port 123
  ntpClient.write(ntpPacketBuffer, NTP_PACKET_SIZE);
  ntpClient.endPacket();

  delay(1000);
  
  int packetLength = ntpClient.parsePacket();
  if (packetLength){
    if(packetLength >= NTP_PACKET_SIZE){
      ntpClient.read(ntpPacketBuffer, NTP_PACKET_SIZE);
    }
    ntpClient.flush();
    uint32_t secsSince1900 = (uint32_t)ntpPacketBuffer[40] << 24 | (uint32_t)ntpPacketBuffer[41] << 16 | (uint32_t)ntpPacketBuffer[42] << 8 | ntpPacketBuffer[43];
    //Serial.printf("Seconds since Jan 1 1900: %u\n", secsSince1900);
    uint32_t epoch = secsSince1900 - 2208988800UL;
    //Serial.printf("EPOCH: %u\n", epoch);
    uint8_t h = (epoch  % 86400L) / 3600;
    uint8_t m = (epoch  % 3600) / 60;
    uint8_t s = (epoch % 60);
    Serial.printf("UTC: %02u:%02u:%02u (GMT)\n", h, m, s);
  }

  delay(9000);
}

void WiFiEvent(WiFiEvent_t event){
    switch(event) {

        //  CHANGED ARDUINO to SYSTEM EVENTS

        case SYSTEM_EVENT_AP_START:
            //can set ap hostname here
            WiFi.softAPsetHostname(AP_SSID);
            //enable ap ipv6 here
            WiFi.softAPenableIpV6();
            break;

        case SYSTEM_EVENT_STA_START:
            //set sta hostname here
            WiFi.setHostname(AP_SSID);
            break;
        case SYSTEM_EVENT_STA_CONNECTED:
            //enable sta ipv6 here
            WiFi.enableIpV6();
            break;
        case SYSTEM_EVENT_GOT_IP6: 
           
            // ADDED PRINT TO DIFERENTMODES

            if ( WiFi.getMode() == WIFI_MODE_STA || WiFi.getMode() == WIFI_MODE_APSTA ) {
              Serial.print("STA IPv6: ");
              Serial.println(WiFi.localIPv6());
            }
            if ( WiFi.getMode() == WIFI_MODE_AP || WiFi.getMode() == WIFI_MODE_APSTA ) {
              Serial.print("AP IPv6: ");
              Serial.println(WiFi.softAPIPv6());
            }
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            wifiOnConnect();
            wifi_connected = true;
            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            wifi_connected = false;
            wifiOnDisconnect();
            break;
        default:
            break;
    }
}

void setup(){
    Serial.begin(115200);   
    Serial.println("");
    WiFi.disconnect(true);
    WiFi.onEvent(WiFiEvent);
    WiFi.mode(WIFI_MODE_APSTA);
    WiFi.softAP(AP_SSID);
    WiFi.begin(STA_SSID, STA_PASS);
}

void loop(){
    if(wifi_connected){
        wifiConnectedLoop();
    }
    while(Serial.available()) Serial.write(Serial.read());
}
```

VojtechBartoska, 2022-05-05:
Is this still valid?

VojtechBartoska, 2022-06-14:
Closing as solved. If needed, please reopen the issue.

closed 2022-06-14 by VojtechBartoska

VojtechBartoska, 2022-06-14:
https://github.com/espressif/esp32-arduino-lib-builder/pull/67

It will be available in next release 2.0.4.