espressif/arduino-esp32 #6593: ESP32 compile error with arduino-esp32 v2.0.3-rc1 (error:  fatal error: bits/error_constants.h: No such file or directory)

samaco634, 2022-04-20:
### Board

ESP32 CAM

### Device Description

ESP32 Wrover Module

### Hardware Configuration

No thing

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

When I try to compile by adding "BluetoothSerial.h", an error occurs and it does not compile.
This only appears when the board option is 'ESP32 wrover module'.
Also, there is no problem in version 2.0.2.

### Sketch

```cpp
#include "BluetoothSerial.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
In file included from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\ios_base.h:46,
                 from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\ios:42,
                 from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\istream:38,
                 from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\sstream:38,
                 from C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\libraries\BluetoothSerial\src\BTAddress.cpp:14:
c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory
 #include <bits/error_constants.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
보드 ESP32 Wrover Module 컴파일 에러.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-23:
Can not reproduce
```
#include <Arduino.h>

#include "BluetoothSerial.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
```
> Executing task: platformio run <

Processing esp-wrover-kit (platform: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip; board: esp-wrover-kit; framework: arduino)
-------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (2.0.3) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 2.0.3 
 - tool-esptoolpy 1.30300.220323 (3.3.0) 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32-arm 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <BluetoothSerial> 2.0.0
Building in release mode
Compiling .pio/build/esp-wrover-kit/src/main.cpp.o
Generating partitions .pio/build/esp-wrover-kit/partitions.bin
Compiling .pio/build/esp-wrover-kit/lib8c0/BluetoothSerial/BTAddress.cpp.o
Compiling .pio/build/esp-wrover-kit/lib8c0/BluetoothSerial/BTAdvertisedDeviceSet.cpp.o
Compiling .pio/build/esp-wrover-kit/lib8c0/BluetoothSerial/BTScanResultsSet.cpp.o
Compiling .pio/build/esp-wrover-kit/lib8c0/BluetoothSerial/BluetoothSerial.cpp.o
Archiving .pio/build/esp-wrover-kit/libFrameworkArduinoVariant.a
Indexing .pio/build/esp-wrover-kit/libFrameworkArduinoVariant.a
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/Print.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/USB.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/USBCDC.cpp.o
Archiving .pio/build/esp-wrover-kit/lib8c0/libBluetoothSerial.a
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/USBMSC.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/WMath.cpp.o
Indexing .pio/build/esp-wrover-kit/lib8c0/libBluetoothSerial.a
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/WString.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/base64.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/esp32-hal-adc.c.o
[... 17 lines cut ...]
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp-wrover-kit/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp-wrover-kit/libFrameworkArduino.a
Indexing .pio/build/esp-wrover-kit/libFrameworkArduino.a
Linking .pio/build/esp-wrover-kit/firmware.elf
Retrieving maximum program size .pio/build/esp-wrover-kit/firmware.elf
Checking size .pio/build/esp-wrover-kit/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.1% (used 13580 bytes from 327680 bytes)
Flash: [=         ]   9.2% (used 120793 bytes from 1310720 bytes)
Building .pio/build/esp-wrover-kit/firmware.bin
esptool.py v3.3
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
=============================== [SUCCESS] Took 4.14 seconds ===============================

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

jmvandervalk, 2022-04-26:
#Compile error occurs when using ESP32_BLE_Arduino library

#include "BLEDevice.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

me-no-dev, 2022-04-26:
@jmvandervalk maybe something went wrong during the installation of the core/toolchain. Your sketch compiles fine for all chips here

VojtechBartoska, 2022-05-05:
Can I consider this as solved or you still need help?

samaco634, 2022-05-09:
[image]

After checking that 2.0.3 was released, I updated it and recompiled it.
The problem still exists.
----------
Arduino: 1.8.19 (Windows 10), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, Huge APP (3MB No OTA/1MB SPIFFS), 80MHz, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Administrator\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Administrator\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Administrator\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32cam:CPUFreq=240,FlashMode=qio,PartitionScheme=huge_app,FlashFreq=80,DebugLevel=none -ide-version=10819 -build-path C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_362678 -warnings=none -build-cache C:\Users\ADMINI~1\AppData\Local\Temp\arduino_cache_217634 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mkspiffs.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -verbose C:\Users\Administrator\Documents\Arduino\sketch_may09a\sketch_may09a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Administrator\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Administrator\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Administrator\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32cam:CPUFreq=240,FlashMode=qio,PartitionScheme=huge_app,FlashFreq=80,DebugLevel=none -ide-version=10819 -build-path C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_362678 -warnings=none -build-cache C:\Users\ADMINI~1\AppData\Local\Temp\arduino_cache_217634 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mkspiffs.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\Administrator\AppData\Local\Arduino15\packages\esp3 [... cut ...]
c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory

 #include <bits/error_constants.h>

          ^~~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

Using library BluetoothSerial at version 2.0.0 in folder: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\BluetoothSerial 

exit status 1

Error compiling for board AI Thinker ESP32-CAM.
-----------------------------------------

JohnGENZ, 2022-05-09:
I have a similar error on a system running under Win10.  An identical system running Win11 works - the file bits/error_constants.h appears to exist.  I am wondering if the composite file name is too long during the compilation process.

VojtechBartoska, 2022-05-26:
is this still valid?

samaco634, 2022-05-27:
That's right.
Even with the newer version 2.0.3 I'm still having this problem.
In 2.0.2, the problem does not occur.

JohnGENZ, 2022-05-27:
Nothing has changed here.  I have two machines.  V2.0.3 does not work on one machine.  V2.0.2 works fine on both.  On the machine where V2.0.3 does nor work the error message from V2.0.3 is as above.  The file does actually exist, but is not being found.

VojtechBartoska, 2022-05-27:
Thanks for updates folks. @PilnyTomas PTAL

Gouzi001, 2022-06-07:
I have sketch where I'm also using BluetoothSerial.h.

On one it's working flawless... on second I'm getting mentioned errors:
arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory

Even the most basic usage of BTserial do the same error.

I tried to copy Arduino15/* between machines and install Arduino IDE 2.0xxx  >> Same error.

Can't it be related to installed version VCC redist ?

JohnGENZ, 2022-06-07:
What OS do you have on both machines - are they the same version?

Gouzi001, 2022-06-07:
W10. They are identical.

JohnGENZ, 2022-06-07:
I have the error running under Win10, but not when running under Win11.   I have been meaning to check the configuration of each to see that the max file name length is the same - but have not done this yet.  (I presume that "bits/error_constants.h"  does actually exisit.)

Nuchaz, 2022-06-08:
I am also experiencing this error. I just installed the Arduino IDE v1.8.19 and the Arduino-ESP32 package, v2.0.3. I'm compiling to a ESP32-S3 and running Windows 10 v1904. I only get this error when compiling with Bluetooth stuff. Other examples seem to work fine.

Benedict-Bsl, 2022-06-13:
**So**

> ### Board
> ESP32 CAM
> 
> ### Device Description
> ESP32 Wrover Module
> 
> ### Hardware Configuration
> No thing
> 
> ### Version
> other
> 
> ### IDE Name
> Arduino IDE
> 
> ### Operating System
> Windows 10
> 
> ### Flash frequency
> 80
> 
> ### PSRAM enabled
> yes
> 
> ### Upload speed
> 921600
> 
> ### Description
> When I try to compile by adding "BluetoothSerial.h", an error occurs and it does not compile. This only appears when the board option is 'ESP32 wrover module'. Also, there is no problem in version 2.0.2.
> 
> ### Sketch
> ```c++
> #include "BluetoothSerial.h"
> 
> void setup() {
>   // put your setup code here, to run once:
> 
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
> 
> }
> ```
> 
> ### Debug Message
> ```
> In file included from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\ios_base.h:46,
>                  from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\ios:42,
>                  from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\istream:38,
>                  from c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\sstream:38,
>                  from C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\libraries\BluetoothSerial\src\BTAddress.cpp:14:
> c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory
>  #include <bits/error_constants.h>
>           ^~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> exit status 1
> 보드 ESP32 Wrover Module 컴파일 에러.
> ```
> 
> ### Other Steps to Reproduce
> _No response_
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Ty the following solution:

[... 155 lines cut ...]

#ifdef _GLIBCXX_HAVE_ETXTBSY
    text_file_busy =              ETXTBSY,
#endif

    timed_out =               ETIMEDOUT,
    too_many_files_open_in_system =       ENFILE,
    too_many_files_open =             EMFILE,
    too_many_links =              EMLINK,
    too_many_symbolic_link_levels =       ELOOP,

#ifdef _GLIBCXX_HAVE_EOVERFLOW
    value_too_large =             EOVERFLOW,
#endif

    wrong_protocol_type =             EPROTOTYPE
    };

_GLIBCXX_END_NAMESPACE_VERSION
} // namespace

#endif
`

Save and run the program again. **Hope this does help someone! cheers!**

PilnyTomas, 2022-06-17:
Hi, I tried compiling under both Linux (Fedora 34) and Window$ (10) using Arduino-ESP32 version 2.0.3 and I could not reproduce any issue - compiled on the first run without error.

Gouzi001, 2022-06-27:
> Anyone able to share a sketch which can help us reproduce this issue?
> Thanks!

"Any" script wil cause error after including <Bluetooth..>  lib.

Jason2866, 2022-06-27:
@Gouzi001  Can not reproduce. See https://github.com/espressif/arduino-esp32/issues/6593#issuecomment-1107478597

PilnyTomas, 2022-06-28:
@Gouzi001 As we pointed out we cannot reproduce the issue. We need more information about your setup. Most likely something went wrong during the installation of the Arduino-esp32.
Please try to reinstall Arduino IDE and the Arduino-esp32 and take notes during the installation - all commands and actions and all the messages and errors you get and share with us.
It is really difficult to guess the reason when we cannot reproduce.

tenchirocom, 2022-07-09:
I too experience this problem. Adding either of these two includes will generate the exact error above:
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

I'm on 1.8.19. Windows 10.

I added BLEScan to an empty sketch, and it failed to compile.

tenchirocom, 2022-07-09:
BTW - I added the file as suggested above and immediately got the bits/messages_members.h error mentioned above. I don't have another install to take that from. I tried copying from the arduino tree, but it didn't work and failed with a completely different problem. Tried an empty file and got the exact same result. Probably something in the esp32 file that's needed. Sketch and error below:

Sketch:
#include <BLEScan.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

Error:

In file included from C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEScan.h:16,
                 from C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\Arduino\sketch_jul09a\sketch_jul09a.ino:1:
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:99:7: error: 'vector' in namespace 'std' does not name a template type
  std::vector<BLEUUID> m_serviceUUIDs;
       ^~~~~~
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:99:2: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:19:1:
+#include <vector>
 
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:99:2:
  std::vector<BLEUUID> m_serviceUUIDs;
  ^~~
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:101:7: error: 'vector' in namespace 'std' does not name a template type
  std::vector<std::string> m_serviceData;
       ^~~~~~
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:101:2: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
  std::vector<std::string> m_serviceData;
  ^~~
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:102:7: error: 'vector' in namespace 'std' does not name a template type
  std::vector<BLEUUID> m_serviceDataUUIDs;
       ^~~~~~
C:\Users\john_4bbk5we\OneDrive\Desktop\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.3\libraries\BLE\src/BLEAdvertisedDevice.h:102:2: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
  std::vector<BLEUUID> m_serviceDataUUIDs;
  ^~~
exit status 1
Error compiling for board WEMOS D1 MINI ESP32.

PilnyTomas, 2022-07-11:
Hi, @samaco634 can you locate the file... ?
`c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\bits\error_constants.h`

PilnyTomas, 2022-07-11:
Can you, please also try the portable version of IDE?
https://docs.arduino.cc/software/ide-v1/tutorials/PortableIDE

samaco634, 2022-07-13:
Yes it is.
error_constants.h file is in c:\users\administrator\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\bits\
And also, portable version has same problem.
it's libs are instaled in 'C:\Users\Administrator\AppData\Local\Arduino15'
[image]

PilnyTomas, 2022-07-15:
@tenchirocom can you please try the change in #6990 if that solves the `vector` error?
As for the "missing" file, I'm still unsure what the problem is. I'm still betting on corrupted installation - mine works ok on Windows.

PilnyTomas, 2022-07-15:
@samaco634 I would like to ask you to try something:
On the problematic Windows machine, please create a new admin user with a very short user name - ideally only 1 character (if Windows allows that).
Install the Arduino IDE as a normal application - NOT with the Micro$oft app store. On the Arduino [software page](https://www.arduino.cc/en/software) please use either [Win 7 and newer](https://downloads.arduino.cc/arduino-1.8.19-windows.exe) or [ZIP file](https://downloads.arduino.cc/arduino-1.8.19-windows.zip).
Add the ESP32 board as described [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html).
Then try to compile the problematic sketch and report if the issue persists.

VojtechBartoska, 2022-07-26:
Any updates on this? If not, we will close this issue soon as we are not able to reproduce it. Thanks for understanding.

nikthefix, 2022-07-30:
I've had this error many times in the past - always after upgrading ESP32 Arduino lib. The solution for me each time is to reduce the path length to the sketch folder. I use portable Arduino IDE for all projects.

closed 2022-08-10 by PilnyTomas

BornaBiro, 2022-10-26:
Sorry to re-open this old thread, but I think I narrow down the problem.

I'm currently using Arduino IDE 1.8.18. with ESP32 Arduino Core v2.0.5 from the stable release.
Selected Board is ESP32 Wrover Module, Flash Frequency 80MHz, Flash Mode QIO, Partition Scheme Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), Core Debug Level None.

If I try to compile SerialToSerialBT example (from ESP32 Examples, Examples->BluetoothSerial), it fails showing error that bits/messages_members.h file is missing (on other PC is bits/error_constants.h). But if I try to compile the same example for ESP32 Dev Module Board, it compiles successfully.

Here's the catch. If I try to compile SerialToSerialBT example for ESP32 Wrover Board, but I remove -mfix-esp32-psram-cache-issue flag from boards.txt file
`esp32wrover.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw`
to
`esp32wrover.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-strategy=memw`
restart the Arduino IDE, it compiles example without any errors. It even runs the Bluetooth example with no problems.

I even try to use PSRAM in the example and again it works without the problems.

So it's definitely a problem with that flag.

fjmsouza, 2023-12-04:
Based on @Benedict-Bsl 's clue:
Copy and Paste: 
FROM:
C:\Users\...\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\xtensa-esp32s3-elf\bits\
TO:
C:\Users\...\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\bits\

Despite the long time, now, compile in 2.2.1 IDE.

Jacounasse, 2024-01-19:
I tried hard to get BluetoothSerial working. I followed Benedic-bsl advice creation a error_constant.h as described. It did not solve the problem. I made the move suggested by fjmsouza. Not successful. I added the messages_members.h file in the same folder. It's worse !

I use a Nano ESP32 and IDE 2.1.1

Too bad BluetoothSerial would be very convenient for my project. I go find another way :( Unless anyone can help.

With thanks