espressif/arduino-esp32 #6595: ESP32-WROOM-32 Serial high speed communication error！

pzh11001, 2022-04-20:
### Board

ESP32 Dev Module

### Device Description

ESP32-WROOM-32
[image]

### Hardware Configuration

ESP32-WROOM-32 and  USB data line

### Version

latest master

### IDE Name

ArduinoIDE

### Operating System

WIN10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When the baud rate of serial port is set to 2500000, the transmission data is wrong. However, when the baud rate is set to 3000000, the transmission data is normal! Repeated tests show that the baud rate of 2000000-2300000 is also normal. But every time it is set to 2500000 baud rate, the data will be wrong. No data is lost, but the data becomes other numbers. For example, I test transmission 0x02. However, the host computer sometimes receives 0x02 and sometimes 0x06.

I tested mega2560 with the same code, and everything was normal.

Is there any way to solve this problem?

### Sketch

```cpp
char cc = 0x02;
void setup() 
{
  Serial.begin(2500000);
  delay(10);
}

void loop() 
{
  Serial.write(&cc);
  delay(500);
}
```

### Debug Message

```plain
06 02 02 06 02 06 02 02 06 02 02 02 06 02 02 06 06 06 02 06 02 06 02 02 02 02 02 06 02 06 02 06 02 02 02 02 02 02 06 06 02 06 02 02 02 02 02 06 02 06 06 02 02 06 02 06 02 06 02 02 02 06 06 02 02 06 02 02 06 02 06 06 02 02 02 02 02 02 02 02 06 06 02 02 02 06 06 02 02 06 06 06 02 02 02 02 06 06 06 06 06 06 02 06 06 02 02 02 06 06 06 02 02 06 02 06 02 06 02 06 06 02 06 06 02 02 02 02 06 02 06 06 02 06 02 06 06 06 06 06 02 06 06 06 02 02 02 02 02 02 02 02 02 02 06 02 02 06 02 02 06 06 06 02 02 06 06 02 06 06 02 06 06 06 02 06 02 02 02 02 06 02 02 02 02 06 06 06 02 02 02 06 06 06 02 06 02 02 06 02 06 06 06 02 06 06 06 06 02 06 06 02 02 06 02 06 02 02 06 06 06 06 06 02 06 06 06 06 06 06 06 02 06 02 06
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pzh11001, 2022-04-20:
[image]
When the baud rate 3000000, everything is normal
How odd!

mrengineer7777, 2022-04-20:
Wow.  2.5Mhz baud is really fast.  Are you sure you need that fast?  The higher the baud rate the shorter your communications lines have to be.  Also, there are hardware limitations on both devices that limit what baud rates are valid.  I don't know what the ESP32 can handle, but I do know that pushing the speed too high can cause signal integrity issues.
Link to baud rate divider issues: https://electronics.stackexchange.com/questions/9264/what-standard-uart-rates-are-there

Ironically, higher baud rates may not give better performance.  Some UARTs can shift out a byte really fast, but if you look on a scope there are large gaps between bytes while the UART waits for another byte to be loaded by the processor.  Something to consider.  Better UARTs implement a small TX buffer, which allows the processor to que up several bytes.

pzh11001, 2022-04-26:
okay, thank you. I think I have to use a lower baud rate

closed 2022-04-27 by pzh11001

capedra, 2023-06-14:
@me-no-dev @pzh11001 
In my case, whenever I choose 2.5mbps, I get this error: "Error: Cannot configure port, something went wrong. Original message: OSError(22, 'Incorrect parameter.', None, 87)". So, the max speed for me is 2.0mbps, not 2.5mbps. And I don't even know why.