espressif/arduino-esp32 #6597: Move to Esp-idf SPI Master Driver

s-hadinger, 2022-04-20:
### Related area

Are there any plans to move `SPI` class to esp-idf native SPI Master interface instead of the current low-level API?

### Hardware specification

ESP32

### Is your feature request related to a problem?

In Tasmota we are using SPI DMA with ILI9341 driver - using Esp-idf SPI Master interface. Unfortunately there is a conflict with Arduino `SPI` class if the same bus is also used (typically for SDCard).

The current work-around is to make sure that DMA transfer is complete and no more transactions are in-flight.

I tried to use the W5500 Ethernet interface, which also relies on SPI Master Interface. There is no easy work-around as both APIs can't coexist on the same bus.

### Describe the solution you'd like

Use `spi_bus_add_device()` and `spi_device_transmit()` APIs to communicate with SPI devices, instead of the current implementation.

Or at least use `spi_device_acquire_bus()` and `spi_device_release_bus()` to make sure there are no conflicts between the 2 APIs.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-04-21:
Hello @s-hadinger, thank you for opening this Feature Request.

We are aware that SPI is awaiting refactoring to use ESP-IDF API. 

I'm adding your request in our Roadmap and we will investigate it.

s-hadinger, 2022-04-21:
Happy to know it's in the roadmap. I'm looking forward to this evolution.

closed 2025-01-16 by Parsaabasi