espressif/arduino-esp32 #6600: Please, add localIPv6 .toShortString()

srochapimenta, 2022-04-20:
### Related area

WIFI IPv6Address

### Hardware specification

ESP32

### Is your feature request related to a problem?

**Feature request**

Most OLED/LCD/TFT have small width, as IPv6 can be shorted/abreviated, so its usefull to display it short as possible.

Normal IPv6 Address: 
    fe80:0000:0000:0000:32ae:a4ff:fe02:e488
    
It can be short as:
    fe80::32ae:a4ff:fe02:e488

### Describe the solution you'd like

I have implemented it at IPv6Address.h/cpp but it is overrited every time i update sdk, so i will apreciate if this feature can be part of SDK.

### Describe alternatives you've considered

I have change IPv6Address class, includding this function, that work as expected, im a beginner, so this function can be better writed.
```
String IPv6Address::toShortString() const {
	unsigned short val;
	std::stringstream stream;
	bool skip;
		
	for(int i = 0; i < 16; i+=2) {
		val = ((_address.bytes[i] << 8) | _address.bytes[i+1]);
		
		if ( val == 0 ) {
			if ( ! skip ) {
				skip = true; 
				stream << ":"; 
			}
		} else {
			if ( skip ) skip = false;
			
			stream << std::right << std::setfill('0') << std::setw(4) << std::hex << static_cast<int>(val);
			
			if ( i < 14 ) stream << ":";
		}
	}
	
	return String( stream.str().c_str() );
}
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-04-22:
Hello, thank you for your suggestion. I added your request to our Roadmap, we will evaluate soon.

sgryphon, 2022-10-12:
There is a pull request for the IPAddress class to add IPv6 support and which already includes the functionality requested here, although in a different class. https://github.com/espressif/arduino-esp32/pull/7174

Once it is merged this issue can probably be closed as redundant / already available.

The new IPAddress code follows the IPv6 IETF canonical format so IPAddress::toString() always outputs in the short format, e.g. "fe80::32ae:a4ff:fe02:e488" (and it also parses the short format in IPAddress::fromString()). It  uses the full IETF algorithm, which is the left-most longest run of two or more zero fields (slightly more complicated than the code above).

(if you want the old long format with the new code you would have to format it yourself, which is not difficult as it is a fixed length format)

This will make IPAddress match ArduinoCore (and so that you don't need two address classes), and already supports. 

The IPv6Address class is no longer necessary, and not widely used, and there is talk of deprecating it after adding support to IPAddress.

closed 2023-12-18 by me-no-dev