espressif/arduino-esp32 #6601: `spiWriteNL` hangs in while loop

johnboiles, 2022-04-20:
### Board

ESP32-PICO-D4

### Device Description

Custom board based on ESP32-PICO-D4 with a SPI-attached display.

### Hardware Configuration

Using default VSPI pins for the SPI display.

### Version

v2.0.2

### IDE Name

PlatformIO + https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

### Operating System

macOS 12.3.1

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Very occasionally (roughly every 1 in 500 screen refreshes) I'm getting lockups in `spiWriteNL` where the [while loop](https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/cores/esp32/esp32-hal-spi.c#L1226) (`while(spi->dev->cmd.usr);`) never exits. After 5s (as configured in my `sdkconfig`) the watchdog timer then kicks in and resets the device. I'm confident this is some deadlock scenario in `spiWriteNL` and not something related to the watchdog timer because if I disable the WDT, then `spiWriteNL` will hang forever when this happens.

I'm using PlatformIO + platform-espressif32 (IDF) + arduino-esp32 2.0.2 (git submodule'd in my `components` dir). Here's the abbreviated relevant bits of my `platformio.ini`:

```
[env:main]
build_type = debug
# ESP-IDF 4.4
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
framework = espidf
board_build.esp-idf.sdkconfig_path = sdkconfig
lib_deps =
	adafruit/Adafruit GFX Library@^1.10.13
	adafruit/Adafruit BusIO@^1.7.2
lib_compat_mode = off
build_flags =
	-D CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
	-D CONFIG_ARDUHAL_LOG_COLORS
	-std=gnu++17
	-include src/config.h
	-D ESP32
	-Wno-unknown-pragmas
build_unflags =
	${common.build_unflags}
	-Werror=all
	-fno-exceptions
	-Wextra
```

Core dump is attached showing `InterruptWDTTimoutCPU1` firing. Would love any speculation as to why this happens and what can be done to mitigate. Probably the next thing I'll try is to drop the Arduino SPI driver and use the IDF methods directly.

[... 276 lines cut ...]
.coredump.tasks.data 0x3ffea740 0x1b8 RW
.coredump.tasks.data 0x3ffd3b98 0x158 RW
.coredump.tasks.data 0x3ffe96e0 0x214 RW
.coredump.tasks.data 0x3fff1618 0x158 RW
.coredump.tasks.data 0x3fff1440 0x1d0 RW
.coredump.tasks.data 0x3ffb7d88 0x158 RW
.coredump.tasks.data 0x3ffba6a0 0x1e8 RW
.coredump.tasks.data 0x3ffdfd64 0x158 RW
.coredump.tasks.data 0x3ffd8030 0x214 RW
.coredump.tasks.data 0x3fff48cc 0x158 RW
.coredump.tasks.data 0x3fff45e0 0x2e4 RW
.coredump.tasks.data 0x3ffec304 0x158 RW
.coredump.tasks.data 0x3ffec100 0x1fc RW
.coredump.tasks.data 0x3ffbd634 0x158 RW
.coredump.tasks.data 0x3ffd2b00 0x1b8 RW
.coredump.tasks.data 0x3ffb5b54 0x158 RW
.coredump.tasks.data 0x3ffde470 0x1f0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-05:
Hello, can you please retest this with 2.0.3 version?

VojtechBartoska, 2022-06-27:
Closing this issue as expired. You can reopen it if needed, thanks for understanding.

closed 2022-06-27 by VojtechBartoska