espressif/arduino-esp32 #6604: Rebooting...

SupremeBy, 2022-04-21:
### Board

ESP32S3 Dev Module

### Device Description

ESP32S3 Dev Module

### Hardware Configuration

no

### Version

latest master

### IDE Name

ARDUINO IDE

### Operating System

win11

### Flash frequency

40M

### PSRAM enabled

no

### Upload speed

921600

### Description

I just updated 2.0.3-rc1 to download the sample program for my esp32-s3-wroom, but after downloading, the development board reboots indefinitely and the serial port prints the following information madly

### Sketch

```cpp
#include <Arduino.h>
#include <Ticker.h>

// attach a LED to pPIO 21
#define LED_PIN 21

Ticker blinker;
Ticker toggler;
Ticker changer;
float blinkerPace = 0.1;  //seconds
const float togglePeriod = 5; //seconds

void change() {
  blinkerPace = 0.5;
}

void blink() {
  digitalWrite(LED_PIN, !digitalRead(LED_PIN));
}

void toggle() {
  static bool isBlinking = false;
  if (isBlinking) {
    blinker.detach();
    isBlinking = false;
  }
  else {
    blinker.attach(blinkerPace, blink);
    isBlinking = true;
  }
  digitalWrite(LED_PIN, LOW);  //make sure LED on on after toggling (pin LOW = led ON)
}

void setup() {
  pinMode(LED_PIN, OUTPUT);
  toggler.attach(togglePeriod, toggle);
  changer.once(30, change);
}

void loop() {
  
}
```

### Debug Message

```plain
Rebooting...
鳨SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376b3c
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
entry 0x403b61d8

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x4037691a:0x3fceb1800x4037913d:0x3fceb1a0 0x4037e471:0x3fceb1c0 0x420050da:0x3fceb2f0 0x40376435:0x3fceb320 0x403bafc7:0x3fceb350 0x403bb24e:0x3fceb380 0x403b622d:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

ELF file SHA256: 0000000000000000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SupremeBy, 2022-04-21:
[image]
This is my download parameter

VojtechBartoska, 2022-04-21:
Hello @SupremeBy, where was the problem?

me-no-dev, 2022-04-21:
> Hello @SupremeBy, where was the problem?

wrong flash mode. Needed to use OPI 80MHz

SupremeBy, 2022-04-25:
> > Hello @SupremeBy, where was the problem?
> 
> wrong flash mode. Needed to use OPI 80MHz

I tried your suggestion, but it didn't work. In addition, I used the same code and parameters to download another esp32s3 board. I found that it can be downloaded and run. I observed that the difference between the two boards is the module. The former is esp32-s3-wroom-2 d2n16r8v and the latter is esp32-s3-wroom-1 m0n8r8. I don't know if this is the main reason

me-no-dev, 2022-04-25:
N8R8 -> QIO 80MHz Flash + OPI PSRAM
N32R8 -> OPI 80MHz Flash + OPI PSRAM

closed 2022-04-25 by me-no-dev