espressif/arduino-esp32 #6608: AnalogRead has been quite slow since v1.0.4

englandsaurus, 2022-04-21:
### Board

ESP32 Dev Module

### Device Description

SparkFun ESP32 Thing

### Hardware Configuration

Microphone on ESP pin 34

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm running an FFT on the signal from a microphone attached to pin 34 (compiling with ESP32 Dev Module) to drive some LEDs. However I've stripped all of that away and written a simple speed test for `analogRead`.

Running the attached sketch on ESP32 Core v1.0.4 completes 916632 loops in the first ten seconds, while running the same code on v2.0.2 only completes 117625 loops in the same amount of time. Is there a reason that `analogRead()` has been slowed down in versions >v1.0.4?

### Sketch

```cpp
#define TIME_BETWEEN_OUTPUTS 1000

unsigned long loopNumber = 0;
unsigned long timeSinceLastOutput;
unsigned long currentTime;

void setup() {
  Serial.begin(115200);
  Serial.println("Speed Test Start");
  currentTime = millis();
  timeSinceLastOutput = currentTime;
}

void loop() {
  currentTime = millis();
  if (currentTime - timeSinceLastOutput > TIME_BETWEEN_OUTPUTS)
  {
    timeSinceLastOutput = currentTime;
    Serial.print(currentTime / 1000);
    Serial.print("s: ");
    Serial.println(loopNumber);
  }
  analogRead(34);
  loopNumber++;
}
```

### Debug Message

```plain
1s: 11754
2s: 23517
3s: 35281
4s: 47045
5s: 58809
6s: 70573
7s: 82335
8s: 94099
9s: 105863
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-22:
Do you see same results on v2.0.3-rc1?

randymortensen, 2022-04-23:
v2.0.3-rc1 is slow too.
On my ESP-WROOM-32, 100000 analogRead's on pin 36 (ADC1 - channel 0) average 87 uSec using 2.0.3-rc1 and only 9 uSec using 1.0.4

lbernstone, 2022-04-23:
@VojtechBartoska It might be worthwhile to include the old analogRead from 1.0.x as an analogReadLegacy function, which will only work on esp32, but will have the speed of the old direct register peeks.
@randymortensen You can copy the old esp32-hal-adc.c and .h from 1.0.4 (or 1.0.6) into your <esp32>/cores/esp32 directory.  This will mean you lose the calibration and readMillivolts functions (along with functionality on any of the newer variants), but should get you back the speed.

randymortensen, 2022-04-23:
Overwriting those files in 2.0.3-rc1 with the ones from 1.0.4 did improve the averaged analogRead time (using the same code as before) to 24 uSec. (Of course, I was hoping for 9 uSec :-)

randymortensen, 2022-04-25:
For kicks I did a further experiment by overwriting those files in platformio which uses the 2.0.1 core. This resulted in 15 uSec per analogRead which is shorter than the 24 uSec I found in Arduino IDE with 2.0.3-rc1.

PilnyTomas, 2022-08-04:
Hi, I can confirm that with the commit https://github.com/espressif/arduino-esp32/commit/d8b209846131119ad6a84af5987e8625f37900df the speed lowered.
The question of getting it back, or having both versions at a time might not be so simple.

VojtechBartoska, 2022-09-21:
@me-no-dev Please write down the resolution for this issue.

me-no-dev, 2022-09-27:
We can not go back to use our own driver for `analogRead`, because there are differences in the ADC between the chips and that would cause a nightmare to support. ESP-IDF's team has already done all that work and do provide support for the feature.

closed 2022-09-27 by me-no-dev