espressif/arduino-esp32 #6610: INADDR_NONE leads to compilation error in Arduino

SuGlider, 2022-04-21:
### Board

Any (ESP32, ESP32S2, ESP32S3, ESP32C3)

### Device Description

Any - it doesn't matter for the issue

### Hardware Configuration

Any - it doesn't matter for the issue

### Version

latest master (checkout manually)

### IDE Name

Any - it doesn't matter for the issue

### Operating System

Any - it doesn't matter for the issue

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

It just need to be compiled and the error will be displayed.
Basically IPAddress INADDR_NONE declared in <IPAddress.h> is replaced by an integer due to any network usage in any Arduino sketch. This issue causes a compilation error.

### Sketch

```cpp
// If it DOESN'T include any Network header file, it works perfectly well! 
// But it makes no sense to do that and use INADDR_NONE...
#include <WiFi.h>
// or just include <ETH.h> 

void setup() {
  Serial.begin(115200);

  // the code line below causes compilation error
  Serial.printf("INADDR_NONE = [%s]", INADDR_NONE.toString());

  // This piece of code compiles, but outputs trash: "IP_NONE addr = [⸮⸮⸮?]" 
  // it should output [0.0.0.0] instead!
  IPAddress addr = INADDR_NONE; 
  Serial.printf("\n IP_NONE addr = [%s]\n", addr.toString());
}

void loop() {
}
```

### Debug Message

```plain
None - just a compilation error message:

Compiling sketch...

C:\Users\espduino\Documents\Arduino\issue_4073_INADDR_UNDEF\issue_4073_INADDR_UNDEF.ino: In function 'void setup()':
issue_4073_INADDR_UNDEF:48:51: error: request for member 'toString' in '4294967295', which is of non-class type 'u32_t' {aka 'unsigned int'}
   Serial.printf("INADDR_NONE = [%s]", INADDR_NONE.toString());
                                                   ^~~~~~~~
Using library WiFi at version 2.0.0 in folder: C:\Users\espduino\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi 
exit status 1
request for member 'toString' in '4294967295', which is of non-class type 'u32_t' {aka 'unsigned int'}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-21:
The main issue is that <lwip/inet.h> declares ```#define INADDR_NONE IPADDR_NONE``` and it includes <lwip/ip4_addr.h> that declares ```#define IPADDR_NONE ((u32_t)0xffffffffUL)``` (IP 255.255.255.255)
When <lwip/inet.h> is included, it makes the Arduino declaration of "IPAddress INADDR_NONE(0,0,0,0)" invalid. This happens in all the Arduino WiFi and ETH libraries.

VojtechBartoska, 2022-04-22:
probably related issue #5220 ?

SuGlider, 2022-04-23:
> probably related issue #5220 ?

Yes, it is possible, once IPADDR_NONE will not work at all in Arduino. I'll verify the issue #5220 as well.

SuGlider, 2022-04-29:
This is another test case used to verify/test this issue and PR Fix:
issue #4732

``` cpp
#include <Arduino.h>
#include <WiFi.h>

void setup() {
  WiFi.begin("Your_SSID", "Your_wifi_password");
  String nodeName = "NODE-" + WiFi.macAddress();
  nodeName.replace(":", "");
  char _nodeName[20]; nodeName.toCharArray(_nodeName, 20);
  WiFi.setHostname(_nodeName);
  WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
  WiFi.mode(WIFI_STA);

  log_i("Connected to: ");
  log_i( "%s", WiFi.SSID() );
  log_i( "%s", WiFi.localIP().toString().c_str() );
  log_i( "%s", WiFi.getHostname());
}

void loop() {
}
```

PR #6659 (2022-04-29): Fixes INADDR_NONE 
## Description of Change

Fixes ```IPAddress INADDR_NONE``` declaration when using Arduino WiFi or ETH.
This symbol was defined as ```0xffffffff``` by ```lwip /inet.h```, making it impossible to use INADDR_NONE correctly.

This PR only works when ```<wifi-provisioning/wifi_config.h>``` has a modification to include ```<lwip/ip4_addr.h>``` instead of ```<lwip/inet.h>```. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.

## Tests scenarios
This PR was  [... cut]
commit ce2cd111 (2022-04-29): Fixes INADDR_NONE  (#6659)

Description of Change

Fixes IPAddress INADDR_NONE declaration when using Arduino WiFi or ETH.
This symbol was defined as 0xffffffff by lwip /inet.h, making it impossible to use INADDR_NONE correctly.

This PR only works when <wifi-provisioning/wifi_config.h> has a modification to include <lwip/ip4_addr.h> instead of <lwip/inet.h>. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.

Tests scenarios

This PR was tested with all Arduino WiFi exa [... cut]
closed 2022-04-29 by me-no-dev