espressif/arduino-esp32 #6614: esp32c3 - SPIFFS, FFat, fail to mount after commit 8ee5f0a

Sys64736, 2022-04-23:
### Board

ESP32-C3-DevKitC-02

### Device Description

Using [ESP32-C3-DevKitC-02](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html)

### Hardware Configuration

Nothing attached to gpio's

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino & PIO

### Operating System

Win 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Example Spiffs, ffat & nvs issues with esp32c3 after commit 8ee5f0a.  All examples work correctly in 2.0.2 up to commit 8ee5f0a

### Sketch

```cpp
Example SPIFFS and FFAT in latest master
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6  
Attempting to boot anyway...
entry 0x403ce000
[  1336][E][FFat.cpp:60] begin(): Mounting FFat partition failed! Error: -1
FFat Mount Failed

Same issues with SPIFFS

All example code works in releases previous to 8ee5f0a

Only changing commit head between tests.  Have used get.py in tools.

Have also used latest Arduino, using 2.0.3-rc1 and 2.0.2.   Same outcome.

With 2.0.3-rc1, other libraries and code work without issue.

No compile errors.   Only err's when trying to access flash from code. e.g. (nvs, spiffs etc.)

Using Arduino latest, littlefs also failed to mount with rc1, same as https://github.com/espressif/arduino-esp32/issues/6579
```

### Other Steps to Reproduce

Using [ESP32-C3-DevKitC-02](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html) and Lolin [C3 mini](https://www.wemos.cc/en/latest/c3/c3_mini.html).   Using example code from repository.

I believe the issue is similar to https://github.com/espressif/arduino-esp32/issues/6572 and https://github.com/espressif/arduino-esp32/issues/6579

Using latest Arduino IDE, littlefs also fails to mount with rc1, same as https://github.com/espressif/arduino-esp32/issues/6579
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
E (115) esp_littlefs: mount failed,  (-84)
E (116) esp_littlefs: Failed to initialize LittleFS
./components/esp_littlefs/src/littlefs/lfs.c:1790:debug: Bad block at 0x0
./components/esp_littlefs/src/littlefs/lfs.c:1796:warn: Superblock 0x0 has become unwritable
E (6481) esp_littlefs: Failed to format filesystem
[  6481][E][LittleFS.cpp:119] format(): Formatting LittleFS failed! Error: -1
[  6488][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
LittleFS Mount Failed
```
Reverting to any commit prior to 8ee5f0a, all example code involving flash read/write work as expected.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-23:
SPIFFS Example (from git master)
```
00:00:00.001-252/52 HDW: ESP32-C3 
00:00:00.016-250/52 UFS: FlashFS mounted with 304 kB free
00:00:00.037 CFG: Loaded from File, Count 6
.ç:B˙.j
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xf4
load:0x403ce000,len:0x898
load:0x403d0000,len:0x2180
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xf4
load:0x403ce000,len:0x898
load:0x403d0000,len:0x2180
entry 0x403ce000
Listing directory: /
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
Hello World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
Hello World!
Deleting file: /foo.txt
- file deleted
Testing file I/O with /test.txt
- writing................................................................
 - 1048576 bytes written in 23150 ms
- reading................................................................
- 1048576 bytes read in 580 ms
Deleting file: /test.txt
- file deleted
Test complete
```

Sys64736, 2022-04-23:
@Jason2866, thanks,  I don't know what I'm including or excluding to get these errors with the examples.  Is there something else needed after S3 support?  Just changing branches from 2.0.2 to the commit with S3 support I can't seem to read/write to the flash/eeprom with 3 devkits and c3 mini.  Is a new platform branch needed?  I'm going to delete the .platformio folder and try again.

Jason2866, 2022-04-23:
To use latest .git master use this setup for platformio.
```
platform = espressif @ 4.1.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

Sys64736, 2022-04-23:
Thanks @Jason2866 , just tried that and couldn't find package, so changed to espressif32 @ 4.1.0
Was using the upstream branch before this.

That installed but still same error with spiffs.  Erased flash again and cleaned.

Here's the upload:  Do you see anything that stands out?
```
PLATFORM: Espressif 32 (4.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.0.3+sha.b3c203d
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkfatfs 2.0.1
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-riscv32-esp 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
```

This is with the upstream branch
```
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32 Dev Module
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.0.3+sha.b3c203d
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-riscv32-esp 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
```
Here's the ini.  Do you see anything in the build flags that would cause problems?

```
;platform = espressif @ 4.1.0
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
framework = arduino

               
monitor_speed = 115200
monitor_filters = default
board = esp32dev
board_build.mcu = esp32c3
board_build.partitions = default.csv
board_build.variant = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = dio
board_build.arduino.ldscript = esp32c3_out.ld

; remove build flags
build_unflags =
  -DARDUINO_ESP32_DEV
  -DARDUINO_VARIANT="esp32"

build_flags =
  -DARDUINO_ESP32C3_DEV
  -DARDUINO_VARIANT="esp32c3"    
  -DCORE_DEBUG_LEVEL=5 
```

Jason2866, 2022-04-23:
I am using esptool.py v3.3. I have builded a framework for project Tasmota with some (small) changes. You can try with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip
```

Sys64736, 2022-04-23:
@Jason2866 
Thank you very much!!  It's all working now.   I did try using:
https://github.com/Jason2866/platform-espressif32.git#Arduino203rc1
in my tests but it still wasn't working.  This one works great!   
Closing the issues.   I really appreciate your help with this.

```
[env:esp32c3]

  platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip

  framework = arduino
board = esp32-c3-devkitm-1

monitor_speed = 115200
```

closed 2022-04-23 by Sys64736

lranders, 2022-07-17:
I ran into exactly the same issue on the ESP32-S3-DevKitC-1U-N8R8 board.
Spent quite some time searching before ending up here.

Switched from the standard `platform = espressif32` to `platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.4/platform-espressif32-2.0.4.zip`

After rebuilding and uploading the firmware.bin file, the error went away.

I am using PlatformIO with the latest (5.0.0) version Espressif 32 framework.

My question then is as follows:
Should the status of this issue really be **Solved**, when it seems that new users will continue
to be baffled (like I was) by the fact that the default deliverable (5.0.0) in PlatformIO still
seems to contain the broken 2.0.3 code and tooling?

Jason2866, 2022-07-17:
@lranders  The issue is solved here. Platformio is a different project. Platformio crew has not released the Platformio core 2.0.4!
Espressif is not responsible for a other company. You can open a issue in Platformio github and ask for release of latest 2.0.4.

Tasmota Platformio release (my part) does differ in some small details. For 99% of users the changes will have no impact. We have choosen some other settings in sdkconfig to build the needed Arduino libs for Tasmota use.

lranders, 2022-07-17:
@Jason2866 That makes sense.

I am still very new to all of this exciting ESP32 stuff, so I am very much poking around and searching for enlightenment.
The reason for my question was because I don't quite know about the relationship with you folks and the PlatformIO team,
save for the fact that they very aggressively link to your project, hence how I ended up here in the first place.

I will search their PR list and open an issue with them as appropriate.

Thank you for your time!