espressif/arduino-esp32 #6615: Capacitive touch not working on esp32-s2

JaimeCababie, 2022-04-23:
### Board

TinyS2, QTPY ESP32 -S2, Weimos S2 mini 

### Device Description

I tried the ESP32 Touch Test on these 3 boards: QTPY ESP32 S2, Wemos S2 mini and the Tiny S2 form Unexpected maker. 

### Hardware Configuration

Nothing is attached to the hardware. 

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I tried using the Touch Test sketch in the examples library but when opened the serial monitor and touched the pin, the values never change. The problem was the same no matter which board I used. 

### Sketch

```cpp
// ESP32 Touch Test
// Just test touch pin - Touch0 is T0 which is on GPIO 4.

void setup()
{
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  Serial.println("ESP32 Touch Test");
}

void loop()
{
  Serial.println(touchRead(T9));  // get value using T0
  delay(10);
}
```

### Debug Message

```plain
No debug message
```

### Other Steps to Reproduce

I uploaded the sketch using 3 different development boards, all of them had the same behavior. The problem was fixed when I downgraded to an older version of the board package. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-24:
Tried your sketch and i got this when touching:
```
69846
70384
70384
70384
70384
70384
70384
72199
72199
72199
72199
72199
72156
72156
72156
72156
72156
72156
79029
79029
79029
79029
79029
79029
51857
51857
51857
51857
51857
51023
51023
51023
51023
27156
27156
27156
27156
7977
7977
7965
7965
7965
7965
20517
20517
20517
20517
20517
20517
28564
28564
28564
28564
28564
28564
28564
28564
7869
7869
7869
7861
7861
7861
7874
7874
7874
7845
7845
7845
7825
7825
7825
7825
7825
7823
7823
7823
7821
7821
7821
7821
7821
7821
7827
7827
7827
8314
8314
8314
8314
7866
7866
7866
7866
7866
11302
11302
11302
11302
22625
22625
22625
22625
22625
33499
33499
33499
33499
33499
38330
38330
38330
38330
38330
38330
42697
42697
42697
42697
42697
32916
32916
32916
32916
32916
21701
21701
21701
21701
7821
7821
7821
7816
7816
7815
7815
7815
7813
7813
7815
7815
7815
7815
7815
7815
7815
7815
7818
7818
8043
8043
8394
8394
8394
8379
8379
8605
8605
8605
8509
8509
8509
8386
8386
8100
8100
8100
8205
8205
8205
8474
8474
8474
8767
8767
8876
8876
8876
8857
8857
8857
8845
8845
8845
8768
8768
7904
7904
7904
7815
7815
7812
7812
7812
7815
7815
7815
7815
7815
7815
7815
7814
7814
7814
7814
7814
7815
7815
7815
7817
7817
7817
7817
7817
7817
7817
7815
7815
7815
7817
7817
7829
7829
7829
7814
7814
7814
7815
7815
7817
7817
7817
7815
7815
7816
7816
7816
7817
7817
7820
7820
7820
7837
7837
7837
7837
8511
8511
8511
8511
13567
13567
13567
13567
13567
33492
33492
33492
33492
33492
16907
16907
16907
16907
18799
18799
18799
18799
7972
7972
7972
7972
7972
49829
49829
49829
49829
49829
49829
49829
59554
59554
59554
59554
59554
59554
7836
7836
7813
7813
7813
7815
7815
7828
7828
7828
9850
9850
8413
8413
8413
8413
10230
10230
10230
10170
10170
10170
10170
10688
10688
10688
10688
10688
14388
14388
14388
14388
14388
14292
14292
14292
14292
29829
29829
29829
19318
19318
19318
12014
12014
12014
8857
8857
8857
9679
9679
9679
10745
10745
10745
10745
10745
10646
10646
10646
10646
11092
11092
11092
11092
12210
12210
12210
12210
12210
12932
12932
12932
12932
12932
13543
13543
13543
13543
13543
13735
13735
13735
13735
13735
13735
12861
12861
12861
12861
7833
7833
7817
7817
7817
7815
7815
7817
7817
7817
7816
7816
7816
7816
7816
7813
7813
7816
7816
7816
7833
7833
7833
20414
20414
20414
20414
20414
20414
20414
20414
20414
20414
20414
20414
20414
20414
20414
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
79481
63009
63009
63009
63009
63009
63009
63009
63009
63009
63009
63009
63009
63009
63009
66889
66889
66889
66889
66889
66889
66889
66889
66889
66889
66889
66889
66889
7909
7909
7819
7819
7819
7816
7816
7817
7817
7817
7816
7816
7819
7819
7819
7818
7818
7817
7817
7817
8062
8062
8479
8479
8479
8955
8955
8955
8955
9114
9114
9114
9347
9347
9347
9697
9697
9697
9697
10339
10339
10339
10339
10725
10725
10725
10973
10973
10973
10973
11196
11196
11196
11392
11392
11392
11392
11378
11378
11378
11048
11048
11048
11048
10402
10402
9363
9363
9363
7865
7865
7825
7825
7825
7820
7820
7819
7819
7819
7817
7817
7819
7819
7819
7818
7818
7819
7819
7819
7817
7817
7819
7819
7818
7818
7818
7819
7819
7817
7817
7817
7821
7821
7818
7818
7818
7818
7818
7819
7819
7819
7818
7818
7826
7826
7826
7826
50960
50960
50960
50960
35683
35683
35683
35683
39093
39093
39093
39093
39093
39093
43111
43111
43111
7913
7913
7913
7919
7919
7938
7938
7938
8149
8149
8149
8149
22566
22566
22566
22566
22566
29240
29240
29240
29240
29240
29240
22758
22758
22758
22758
16231
16231
16231
16231
23010
23010
23010
23010
23010
24711
24711
24711
24711
24711
24784
24784
24784
24784
24784
25570
25570
25570
25570
25570
25995
25995
25995
25995
25995
28873
28873
28873
28873
28873
28873
30466
30466
30466
30466
30466
30466
31121
31121
31121
31121
31121
31121
32080
32080
32080
32080
32080
32080
32752
32752
32752
32752
32752
32752
32752
34536
34536
34536
34536
34536
34536
34536
34536
38525
38525
38525
38525
38525
38525
38525
38525
38525
41238
41238
41238
```

JaimeCababie, 2022-04-24:
Which version of the Arduino Core are you using?

P-R-O-C-H-Y, 2022-04-25:
Hi @JaimeCababie,
I can confirm that there is no problem with Touch. Tested you sketch and got similar results as @Jason2866 had.
Tested with latest dev release 2.0.3-RC1 with Arduino IDE.

VojtechBartoska, 2022-05-04:
Did you have a chance to retest your code @JaimeCababie? Thanks

JaimeCababie, 2022-05-04:
It worked after I reinstalled the Arduino boards in the IDE. I have not been able to recreate the problem since. Thanks for the support.