espressif/arduino-esp32 #6616: Confusing overload of `Wire::begin`

Equidamoid, 2022-04-23:
### Board

Any

### Device Description

Not specific to a device.

### Hardware Configuration

Any I2C device connected

### Version

v2.0.2

### IDE Name

CLion + platformio

### Operating System

Linux

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

The `Wire::begin` method has confusing overload:
```
    bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
    bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
```
The `sda` and `scl` arguments are of type `int`, while the `SDA` and `SCL` constants from `d32_core.h` have type `uint8_t`. A "naive" user might try to call `begin(SDA, SCL)` and get mysterious `"could not acquire lock"` error on first attempt to use the bus.

To make the problem worse, this code worked fine before the "slave" overload was introduced.

As a solution I want to propose renaming the "slave" function to something like `begin_slave`.

### Sketch

```cpp
int setup(){
    // have to explicitly specify the pins to provide the frequency
    Wire.begin(SDA, SCL, 4000);
    Wire.beginTransmission();
}
```

### Debug Message

```plain
[   238][E][Wire.cpp:319] beginTransmission(): could not acquire lock
[   239][E][esp32-hal-i2c.c:142] i2cWrite(): could not acquire lock
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-24:
You can also:
```cpp
Wire.begin();
Wire.setClock(frequency);
```

StickStackxD, 2022-04-24:
> You can also:
> 
> ```c++
> Wire.begin();
> Wire.setClock(frequency);
> ```

I can confirm that the workaround is working for now.

Equidamoid, 2022-04-24:
@me-no-dev 
I came up with this (I know, ugly, but at least straightforward):
```
Wire.begin(static_cast<int>(SDA), static_cast<int>(SCL), static_cast<uint32_t>(4000));
```

me-no-dev, 2022-04-25:
casting also works, but I do see this as a kind of issue :) we will have to "fix" this somehow

mymy-dot, 2022-05-25:
This issue ended up taking me several hours to solve, until by chance I came across this post. Granted, I don't know what I'm doing. But I do know that adding:

`Wire.begin(static_cast<int>(SDA), static_cast<int>(SCL), static_cast<uint32_t>(4000));`

Made the 'beginTransmission(): could not acquire lock' error go away on my ESP-WROOM-32 and I was able to access my I2C device.

bato3, 2022-07-14:
This is because SDA and SCL are defined as 
```cpp
static const uint8_t SDA = 21;
static const uint8_t SCL = 22;
```
and `Wire.begin()` can be:
```cpp
    bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
    bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
```

And [some libriares](https://github.com/enjoyneering/AHTxx/issues/5) defines it initializations as:
```cpp
   bool     begin(uint8_t sda = SDA, uint8_t scl = SCL, uint32_t speed = AHTXX_I2C_SPEED_100KHZ, uint32_t stretch = AHTXX_I2C_STRETCH_USEC);
```

### Working solutions
```cpp
  Wire.begin((int) SDA , (int) SCL);
  Wire.begin(21, 22);
  Wire.begin();
```
Not working solutions:
```cpp
Wire.begin(0x38, (int) SDA, (int) SCL);
Wire.begin((uint8_t) -1, (uint8_t) -1); // display error:
//i2cSlaveInit(): invalid pins sda=-1, scl=22
```

SuGlider, 2022-07-16:
I see some other conflict with overload of `Wire::begin()`:

`Wire.begin(SDA, SCL, 40000)` will end up in I2C Slave initalization... This is a Regression Issue!
```
[   102][E][esp32-hal-i2c-slave.c:224] i2cSlaveInit(): invalid pins sda=9, scl=-128
[   109][E][Wire.cpp:168] begin(): Slave Init ERROR
```

I also noticed that it is necessay to force the right overload by using something like:
``` cpp
Wire.begin(SDA, SCL, (uint32_t)40000);   //  casting forces it to be MASTER
```

By other hand, a typical Arduino Slave initialization such as
`Wirebegin(0x34);`
will end up in a I2C MASTER initialization - also confusing because this is the Arduino I2C Slave deafult setup.

In order to force Slave setup, it is necessary to use a type casting again...
``` cpp
Wire.begin((uint8_t)0x34);   //  casting forces it to be SLAVE
```

@me-no-dev - would you consider changing all of it to the default Arduino way  `Wire::begin()`/`Wire::begin(uint8_t)` and only expose `Wire::setClock()` and `Wire::setPins()` for an Arduino Core 2.1.0, for instance?

I think it may be a good way to "start over" and solve it for good.

SuGlider, 2022-07-16:
@me-no-dev - Please evaluate the proposed solution in #7000

SuGlider, 2022-07-16:
#7000 make all those cases to work fine:

``` cpp
  // I2C MASTER Test Cases
  Serial.println("\nUC 1.1 - Master - default SDA/SCL pins, default 100KHz bus");
  Wire.end(); Wire.begin();

  Serial.println("\nUC 1.2 - Master - default SDA/SCL pins, default 100KHz bus");
  Wire.end(); Wire.begin(-1, -1);
  
  Serial.println("\nUC 1.3 - Master - default SDA/SCL pins, 1KHz bus");
  Wire.end(); Wire.begin(-1, -1, 1000);
  
  Serial.println("\nUC 1.4 - Master - default SDA, SCL pin 2, 2KHz bus");
  Wire.end(); Wire.begin(-1, 2, 2000);
  
  Serial.println("\nUC 1.5 - Master - default SDA/SCL pins, 400KHz bus");
  Wire.end(); Wire.begin(SDA, SCL, 400000);
  
  Serial.println("\nUC 1.6 - Master default SDA pin 8, SCL pin 9, 800KHz bus");
  Wire.end(); Wire.begin(8, 9, 800000);
  
  Serial.println("\nUC 1.7 - Master - SDA pin 8, SCL pin 9, default 100KHz bus");
  Wire.end(); Wire.begin(8,9);
  
  Serial.println("\nUC 1.8 - Master - SDA pin 10, SCL pin 11, 200KHz bus");
  Wire.end(); Wire.begin(SDA_1, SCL_1, 200000);
  
  Serial.println("\nUC 1.9 - Master - default SDA/SCL pins, default 100KHz bus");
  Wire.end(); Wire.begin(SDA, SCL);
  
  Serial.println("\nUC 1.10 - Master - default SDA/SCL pins, 4KHz bus");
  Wire.end(); Wire.begin(-1, -1, 4000);
  
  Serial.println("\nUC 1.11 - Master - default SDA/SCL pins, default 100KHz bus");
  Wire.end(); Wire.begin((int) SDA , (int) SCL);

  // I2C SLAVE Test Cases
  Serial.println("\nUC 2.1 - Slave - ADDR 0x28 - default SDA/SCL pins, default 100KHz bus");
  Wire.end(); Wire.begin(0x28);
  
  Serial.println("\nUC 2.2 - Slave - ADDR 0x21 - default SDA/SCL pins, default 100KHz bus");
  Wire.end(); Wire.begin(SLV_ADDR);
  
  Serial.println("\nUC 2.3 - Slave - ADDR 0x21 - default SDA/SCL pins, 4KHz bus");
  Wire.end(); Wire.begin(SLV_ADDR, -1, -1, 4000);
  
  Serial.println("\nUC 2.4 - Slave - ADDR 0x38 - SDA 10, SCL 11, default 100KHz bus");
  Wire.end(); Wire.begin(0x38, SDA_1, SCL_1, 0);
  
  Serial.println("\nUC 2.5 - Slave - ADDR 0x38 - default SDA/SCL pins, 4KHz bus");
  Wire.end(); Wire.begin(0x38, -1, SCL, 4000);

```

closed 2022-07-26 by SuGlider