espressif/arduino-esp32 #6622: undefined reference to `esp_crt_bundle_attach'

MikesTag, 2022-04-24:
### Board

ESP32 

### Device Description

DevKitC 4

### Hardware Configuration

Plain

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When using the esp_http_client with certificate bundle
The linker from the Arduino IDE throws an error:
 
undefined reference to `esp_crt_bundle_attach'

### Sketch

```cpp
#include "esp_crt_bundle.h"
#include "esp_http_client.h"

void setup() {
  esp_http_client_config_t config = 
  {    
          .url = "https://content.dropboxapi.com",
          .crt_bundle_attach = esp_crt_bundle_attach
  };
  esp_http_client_handle_t client = esp_http_client_init(&config);
}

void loop() {
}
```

### Debug Message

```plain
undefined reference to `esp_crt_bundle_attach'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-24:
The bundle is repackaged in WiFiClientSecure.  #include <WiFiClientSecure.h> first, and it will find the correct header.  Not ideal, but functional for now.

MikesTag, 2022-04-24:
Thanks for the fast reply
Unfortunatly only in the new 2.0.3 RC
In 2.0.2 it is not included in the WiFiClientSecure

As workaround, i replaced it with the new in the library folder from RC and works now.

closed 2022-04-24 by MikesTag