espressif/arduino-esp32 #6623: max() not accepting #defined values

palmerr23, 2022-04-25:
### Board

DevkitC ESP32 WROOM

### Device Description

None

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino 1.8.13

### Operating System

Win 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Max is not accepting #defined values

The second call to max() fails to compile in ESP32, nut OK in AVR.

Issue occurs in both 2.0.2 and 2.0.3_RC1

### Sketch

```cpp
#define MULT 0.7
void setup() {
  float a = 1.1, b = 2.2, c;  
  c = max(a, b);
  c = max(a, MULT);
}

void loop() {
}
```

### Debug Message

```plain
C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\dad\_microcontroller\esp\esp32\max_test\max_test.ino: In function 'void setup()':
max_test:5:17: error: no matching function for call to 'max(float&, double)'
   c = max(a,MULT);
                 ^
In file included from c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\xyzzy.DESKTOP-J4ESB3N\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:155,
                 from sketch\max_test.ino.cpp:1:
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3468:5: note: candidate: 'template<class _Tp, class _Compare> _Tp std::max(std::initializer_list<_Tp>, _Compare)'
     max(initializer_list<_Tp> __l, _Compare __comp)
     ^~~
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3468:5: note:   template argument deduction/substitution failed:
C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\dad\_microcontroller\esp\esp32\max_test\max_test.ino:5:17: note:   mismatched types 'std::initializer_list<_Tp>' and 'float'
   c = max(a,MULT);
                 ^
In file included from c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\xyzzy.DESKTOP-J4ESB3N\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:155,
                 from sketch\max_test.ino.cpp:1:
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3462:5: note: candidate: 'template<class _Tp> _Tp std::max(std::initializer_list<_Tp>)'
     max(initializer_list<_Tp> __l)
     ^~~
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3462:5: note:   template argument deduction/substitution failed:
C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\dad\_microcontroller\esp\esp32\max_test\max_test.ino:5:17: note:   mismatched types 'std::initializer_list<_Tp>' and 'float'
   c = max(a,MULT);
                 ^
In file included from c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\xyzzy.DESKTOP-J4ESB3N\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:155,
                 from sketch\max_test.ino.cpp:1:
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:265:5: note: candidate: 'template<class _Tp, class _Compare> const _Tp& std::max(const _Tp&, const _Tp&, _Compare)'
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^~~
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:265:5: note:   template argument deduction/substitution failed:
C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\dad\_microcontroller\esp\esp32\max_test\max_test.ino:5:17: note:   deduced conflicting types for parameter 'const _Tp' ('float' and 'double')
   c = max(a,MULT);
                 ^
In file included from c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\xyzzy.DESKTOP-J4ESB3N\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:155,
                 from sketch\max_test.ino.cpp:1:
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:219:5: note: candidate: 'template<class _Tp> const _Tp& std::max(const _Tp&, const _Tp&)'
     max(const _Tp& __a, const _Tp& __b)
     ^~~
c:\users\xyzzy.desktop-j4esb3n\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:219:5: note:   template argument deduction/substitution failed:
C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\dad\_microcontroller\esp\esp32\max_test\max_test.ino:5:17: note:   deduced conflicting types for parameter 'const _Tp' ('float' and 'double')
   c = max(a,MULT);
```

### Other Steps to Reproduce

Nothing else, simple compile issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

palmerr23, 2022-04-25:
Sorry, Arduino IDE is 1.8.19

lbernstone, 2022-04-25:
If you want to ensure it is a float, cast it as a float:
`#define MULT (float)0.7

palmerr23, 2022-04-25:
Thanks for the tip, however I'm already aware of the workarounds. BTW, one should not need to cast a clearly floating point number as float, despite the fact that it works! 
I was merely pointing out that this compiler's response to preprocessor insertions is different from "Arduino standard."

lbernstone, 2022-04-26:
Yes, the authors of gcc do not seem to care much about the "Arduino standard"

SuGlider, 2022-04-27:
Arduino Standard defines a macro for ```max()```
#define max(a,b) ((a)>(b)?(a):(b))
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Arduino.h#L93

ESP32 Arduino doesn't define such macro...
it has something else instead:
#define _max(a,b) ((a)>(b)?(a):(b))
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L207

@palmerr23 - does your sketch behave the same as Arduino Standard if you use ```_max()``` instead of ```max()```?

SuGlider, 2022-04-27:
A workaround for now is to add this code to the first line of your sketch:

```#define max(a,b) _max(a,b)```

I'll verify if we can add a patch to solve it for good.

palmerr23, 2022-04-28:
Yes, the workaround does work as you describe.

It would be useful to have _max() and max() doing the same thing, preferably the same as Arduino "standard".

Thanks for your efforts so far on this.

SuGlider, 2022-05-10:
@palmerr23 - There is no way to add `max()/min()` to the `Arduino.h` because it conflicts with functions with the same name used in the C++ environment of the compiler.

Therefore, the work around proposed is the only way to go.

I added a few other functions to `Arduino.h` in the PR #6726

PR #6726 (2022-05-11): Adds Arduino standard macro-functions
## Description of Change
This is a review of macro basic functions from Arduino.h that were missing from the original Arduino version.
This PR adds some missing functions to ESP32 Arduino.h

## Tests scenarios
Tested using ESP32 and a few examples. The main target of testing was to make sure that there is no compiling conflicts.
Also searched for such macro on the source code to verify that those functions are not included anywhere else.

## Related links

Closes #6623
closed 2022-05-11 by palmerr23