espressif/arduino-esp32 #6624: I want the ESP32 default setting to be HT20.

CheolsoonKwon, 2022-04-25:
### Board

ESP32-WROOM-32D

### Device Description

Devkit

### Hardware Configuration

GPIO 16, 17pin is UART 

### Version

v2.0.2

### IDE Name

Arduino

### Operating System

Windows10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello Everyone

I want the ESP32 default setting to be HT20.

I changed the settings because I couldn't capture in Wireshark.

Change to HT20 -> HT40+ to capture.

ESP32 Arduino seems to work with HT40+.

Previously, the test condition was Laptop(Station) <-> ESP32(AP).

It worked without any problems, and when I changed the Laptop(station) to an HT20 dedicated device, a data transmission/reception problem occurred.

Is it possible to change ESP32's HT40 setting to HT20 on Arduino?

### Sketch

```cpp
//Server Configuration
    Serial.println("Configuring access point...");
    // You can remove the password parameter if you want the AP to be open.
    WiFi.softAPConfig(local_IP, gateway, subnet);
    WiFi.softAP(Read_SSID, Read_PW);
    IPAddress myIP = WiFi.softAPIP();
    Serial.print("AP IP address: ");
    Serial.println(myIP);
    server.begin();
    server.setNoDelay(true);
    Serial.println("Server started");
    SSID_PW_Change_Flag = 2;
    SSID_PW_Diff_Flag = 2;
    Serial.println("AP START");
```

### Debug Message

```plain
No Debug Message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-25:
[esp_wifi docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_interface_t)
```
#include <WiFi.h>
#include <esp_wifi.h>

void setup() {
  Serial.begin(115200);
  WiFi.softAP("led101010", "test1234");
  esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
}

void loop() {

```
#include <WiFi.h>
#include <esp_wifi.h>
void setup() {
  Serial.begin(115200);
  WiFi.softAP("esp32", "test1234");
  esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
}
void loop() {delay(-1);}
```

VojtechBartoska, 2022-05-18:
is this still valid @CheolsoonKwon?

grtyvr, 2022-06-27:
Might be nice if that were a paramater in the config struct.

closed 2025-01-16 by Parsaabasi