espressif/arduino-esp32 #6627: Inconsistency in LFS_NAME_MAX settings (LittleFS / mklittlefs)

Jason2866, 2022-04-25:
### Board

not relevant

### Device Description

general issue

### Hardware Configuration

no hardware

### Version

latest master (checkout manually)

### IDE Name

Platformio (custom self build)

### Operating System

macOS 12.3

### Flash frequency

not relevant

### PSRAM enabled

no

### Upload speed

115200

### Description

When a LittleFS partition is downloaded via esptool.py the files can not be extracted. There are different settings choosen for `LFS_NAME_MAX` for LittleFS in IDF/Arduino and in tool `mklittlefs`. It is`64` for IDF / Arduino. For the tool `mklittlefs` it is `32`
As it is now the tool `mklittlefs` is not useable.
See https://github.com/joltwallet/esp_littlefs/issues/78 too.

### Sketch

```cpp
Make a LittleFS partition download this partition and try to extract with mklittlefs.
```

### Debug Message

```plain
not needed
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-26:
@me-no-dev No, there is no tool at the moment which has `LFS_NAME_MAX = 64` as setting. Esp8266 Arduino uses `32` as default. Would be nice to be align.
Idk if this would be a breaking change.

BrianPugh, 2022-04-26:
the least breaking change would probably be to upload pre-compiled binaries with `LFS_NAME_MAX=64`. Changing the arduino default from 64->32 could catch people who rely on the default by surprise and leave broken systems.

The repo owner [provided a nice command](https://github.com/earlephilhower/mklittlefs#cross-compiling) to cross compile all the packages, could simply rebuild them with `-DLFS_NAME_MAX=64`.

BrianPugh, 2023-05-04:
I merged in the PR that correctly passed along the appropriate variables to the compilation of mklittlefs. This is a part of esp_littlefs v1.5.4

BrianPugh, 2023-09-20:
`esp_littlefs` has migrated from mklittlefs to littlefs-python for image generation, further mitigating this problem.

Jason2866, 2023-09-20:
@BrianPugh Thanks for heads up! Will test the next days.

Jason2866, 2023-09-23:
It does work now with the changes made. Thx @BrianPugh
Compiled mklittlefs version with setting `LFS_NAME_MAX = 64`. Now it is possible to build a valid FS image and read the FS image back from a device and extract the files. Extracting files was never possible before.
@me-no-dev The fully esp32 compatible mklittlfs binaries for all platforms are [here](https://github.com/Jason2866/mklittlefs_esp32/releases/tag/030200)
Feel free to use or not ;-)

BrianPugh, 2023-09-23:
@Jason2866 the purpose of the most recent change was to use `littlefs-python` for creation.  littlefs-python is precompiled, so developers don't have to worry about building the binary.

You could either directly use `littlefs-python` in python as a library, or shell it out by changing [this line](get_extract_cmd) to invoke the `littlefs-python` CLI.

Jason2866, 2023-09-23:
@BrianPugh Was too focused on solving mklittlefs. Yes using the Python `littlefs-python` is easier ;-)

Parsaabasi, 2025-01-08:
Hello,

This issue is solved. I go ahead and close it. In case there was reoccurrence, feel free to reopen it. 

Thanks

closed 2025-01-08 by Parsaabasi