espressif/arduino-esp32 #6631: Port name is incorrect - not using board name based on PID/VID - showing C3 instead of S3.

UnexpectedMaker, 2022-04-26:
### Board

TinyS3, ProS3 & FeatherS3

### Device Description

Plain board, no external HW connected.

### Hardware Configuration

Plain board, no external HW connected.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

MacOS 12.3.1

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Flashing the board should set the name in ( ) in the ports list based n the board name using the VID/PID provided - not only does it not put the correct board name, but it says it's an ESP32C3 Dev Module - when in fact they are ESP32-S3 boards
[image]

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Happens in 2.0.3RC1 as well.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

UnexpectedMaker, 2022-04-26:
Oh, actually, it works when `USB Mode` set to OTG - but not when it's set to HW CDC/Serial JTAG.
So just need the `HW CDC/Serial JTAG` path to also use the correct naming.

me-no-dev, 2022-04-26:
@UnexpectedMaker we can not change the PID/VID of the integrated CDC/JTAG. We can only do it on the OTG peripheral. And both S3 and C3 have the same PID/VID so they show as the same board. It is not something we can adjust.

closed 2022-04-26 by me-no-dev

UnexpectedMaker, 2022-04-26:
Ok, bummer, thanks for the confirmation @me-no-dev !