espressif/arduino-esp32 #6632: esp32-hal-misc.c infinite loop.

dynamic21xx, 2022-04-26:
### Board

nodemcu-32s

### Device Description

nodemcu-32s plain.

### Hardware Configuration

No.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

UbuntuLinux 20

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200,9600

### Description

In my program used DallasTemperature, and onewire. espressif32 3.5 working fine.
When I update espressif32 4.1, my program cannot run, and the program always reset.
In first run 0 sensor attached, but espressif32 3.5 working fine.
I found in esp32-hal-misc.c infinite loop.
If I commented out these lines , espressif32 4.1 working.
https://community.platformio.org/t/update-espressif32-4-1-dallastemperature-senzor-begin-freeze/27399
[image]

### Sketch

```cpp
DallasTemperature
senzor.begin()
```

### Debug Message

```plain
always reset
```

### Other Steps to Reproduce

espressif32 3.5 working fine.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-26:
That's a finite loop.  esp_timer_get_time() is a volatile function that (surprise) gives you the current number of microseconds since boot.  You don't say what you are using for DallasTemperature, but wherever it is using delayMicroseconds seems more likely the source of your problem.

dynamic21xx, 2022-04-27:
Thanks, but the dallastemperature call this timer in begin() method. I just call _senzor.begin().
In my debugger 'e' value is 2, and never going out that while-do cycle. This error only came,  when I update espressif32 4.1. If I use espressif32 3.5 , no error.
SilentTemperatureSensor::SilentTemperatureSensor(uint8_t pin): _oneWire( pin ), _sensors( &_oneWire ) {
    _pin = pin;
    _sensors.begin();  //freeze if not commented

mrengineer7777, 2022-04-27:
Looks like you are using PIO.  PIO targets arduino-esp32 version 1.0.4, based on espressif32 3.3.x.  Most of us are manually targeting a newer version of Arduino core.  I'm using 2.0.2.3, custom TASMOTA build:
```
framework = arduino
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
```

Your Dallas library may not be compatible with the newer Arduino framework.

We'll need more info about your platform targets, or even better a sample program that duplicates the issue if you want help.  I recommend creating a Github repository with the sample to make it easy for others to download the project and test.

dynamic21xx, 2022-04-27:
I'm so sorry for the endless loop opinion, it's too early, in my debugging. I will try your suggestion.
Thank you!

VojtechBartoska, 2022-05-18:
any updates on this @dynamic21xx? Thanks!

Parsaabasi, 2025-01-08:
Hello, 

Since there were no replies, I close this issue. In case you would need us to take it further, please feel free to reopen it. 

Thanks

closed 2025-01-08 by Parsaabasi