espressif/arduino-esp32 #6635: two uart (uart0 and uart1) receive interrupt didnot work in the same code

muzahid13, 2022-04-27:
### Board

esp32-s2-devkitm-1

### Device Description

esp32-s2-devkitm-1

### Hardware Configuration

i am using as following pin for uart0 and uart1
 U1RXD 17
 U1TXD 18
 U0RXD 44
 U0TXD 43

### Version

v1.0.6

### IDE Name

Arduino IDE 

### Operating System

Windows 10

### Flash frequency

80mHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am developing a project, here uart1 will receive data then uart0 will sent data finally uart0 will receive data, I need to receive data over uart received interrupt. I tested that intially power up the board uart1 can receive and uart0 can sent data for only one time, then hold the board. if i press the reset button, it follow the same only for one time data receiving. 

please check my following ISR code:

static void UART_ISR_ROUTINE(void *pvParameters)
{
    uart_event_t event;
    bool exit_condition = false;
    while (1) { 
        if(xQueueReceive(uart1_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
             if (event.type == UART_DATA) {
                ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_1, (size_t*)&UART1_data_length));
                UART1_data_length = uart_read_bytes(UART_NUM_1, UART1_data, UART1_data_length, 100);

             uart_write_bytes(UART_NUM_0, UART1_data, 8);
             ESP_ERROR_CHECK(uart_wait_tx_done(UART_NUM_0, 100));

          }
               else {
                //TODO...
                   }     
        }

      if(xQueueReceive(uart0_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
            if (event.type == UART_DATA) {
                ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_0, (size_t*)&UART0_data_length));
                UART0_data_length = uart_read_bytes(UART_NUM_0, UART0_data, UART0_data_length, 100);

             uart_write_bytes(UART_NUM_0, UART0_data, 8);
             ESP_ERROR_CHECK(uart_wait_tx_done(UART_NUM_0, 100));
             
          }  

              else {
                //TODO...
            } 
        
        }

  if (exit_condition) {
            break;
        }      
    }
  
    vTaskDelete(NULL);
}

please help me to find out the problem, please suggest me how can i solve this problem. thanks

### Sketch

```cpp
static void UART_ISR_ROUTINE(void *pvParameters)
{
    uart_event_t event;
    bool exit_condition = false;
    while (1) {

      
        if(xQueueReceive(uart1_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
             if (event.type == UART_DATA) {
                ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_1, (size_t*)&UART1_data_length));
                UART1_data_length = uart_read_bytes(UART_NUM_1, UART1_data, UART1_data_length, 100);

             uart_write_bytes(UART_NUM_0, UART1_data, 8);
             ESP_ERROR_CHECK(uart_wait_tx_done(UART_NUM_0, 100));

          }
               else {
                //TODO...
                   }     
        }

      if(xQueueReceive(uart0_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
            if (event.type == UART_DATA) {
                ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_0, (size_t*)&UART0_data_length));
                UART0_data_length = uart_read_bytes(UART_NUM_0, UART0_data, UART0_data_length, 100);

             uart_write_bytes(UART_NUM_0, UART0_data, 8);
             ESP_ERROR_CHECK(uart_wait_tx_done(UART_NUM_0, 100));
             
          }  

              else {
                //TODO...
            } 
        
        }

  if (exit_condition) {
            break;
        }
        
    }
  
    vTaskDelete(NULL);
}
```

### Debug Message

```plain
two uart receive interrupt didnot work
```

### Other Steps to Reproduce

i tested practically, it received the data only one time then got hold

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-27:
Hello @muzahid13,

Serial was refactored and there were a lot of changes since Arduino ESP32 core v1.0., which is no longer supported.

Please test your code against v2.0.3-rc1. Thanks.

SuGlider, 2022-04-27:
@muzahid13 - I think the report may be wrong because Arduino Core 1.0.6 has no support for ESP32-S2.

I also see that you are trying to create an UART ISR. It won't work with Arduino.
We have a new HardwareSerial API called ```onReceive()``` that runs an user callback function as soon as data arrives to the UART.

Please refer to https://github.com/espressif/arduino-esp32/issues/6102#issuecomment-1010584406 - it is a similar request.

SuGlider, 2022-04-27:
Anyway, in order to be able to use ```onReceive(callBackFunc)```, it is necessary to use Arduino Core 2.0.3-RC1 (latest one).

SuGlider, 2022-05-17:
@muzahid13 - Can we close this issue?

closed 2022-05-18 by SuGlider