espressif/arduino-esp32 #6636: Does ESP32 + RTL8201 support Wake on LAN from Deep sleep

ryanyang83, 2022-04-27:
### Board

ESP32 wrover 

### Device Description

ESP32 wrover + RT8201 (Unuse BT/WIFI) 

### Hardware Configuration

H/W design base on ESP32-Ethernet-Kit V1.2 and the LAN PHY chip change to RLT8201 

### Version

v2.0.2

### IDE Name

no

### Operating System

no

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

RTL8201 LAN PMEB is connected to ESP32 GPIO34, wake up through RTC_GPIO, The system can wake on LAN from light sleep,but can't wake up from deep sleep. Is it a hardware limitation?

### Sketch

```cpp
no
```

### Debug Message

```plain
no
```

### Other Steps to Reproduce

no

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-27:
is the ESP driving the clock of the RTL? Probably it needs something running from the ESP that is otherwise suspended in DeepSleep

ryanyang83, 2022-05-05:
After check your recommend. The 8201 can wake on from deep sleep. Thanks

closed 2022-05-05 by ryanyang83