espressif/arduino-esp32 #6637: enroll_face_id_to_flash_with_name  return -2

davidbcf, 2022-04-27:
### Board

ESP32-S2

### Device Description

ESP32-CAM 

### Hardware Configuration

4 pins：5V ，U0R ， UOT ，GND

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

win 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

enroll_face_id_to_flash_with_name  return -2 

E (8869) fr_flash: Not found
Camera Ready! Use 'http://10.8.4.177' to connect
enroll_face 2
enroll_face 1
E (28570) fr_flash: Not found
enroll_face -2
enroll_face 2
enroll_face 1

### Sketch

```cpp
static inline int do_enrollment(face_id_name_list *face_list, dl_matrix3d_t *new_id)
{
  ESP_LOGD(TAG, "START ENROLLING");
  int left_sample_face = enroll_face_id_to_flash_with_name(face_list, new_id, st_name.enroll_name);
  ESP_LOGD(TAG, "Face ID %s Enrollment: Sample %d",
           st_name.enroll_name,
           ENROLL_CONFIRM_TIMES - left_sample_face);
  Serial.print("enroll_face ");
  Serial.println(left_sample_face);
  return left_sample_face;
}
```

### Debug Message

```plain
E (5872) fr_flash: Not found
Camera Ready! Use 'http://10.8.4.177' to connect
[D][face_test.ino:213] do_enrollment(): START ENROLLING
[D][face_test.ino:217] do_enrollment(): Face ID 1 Enrollment: Sample 1
enroll_face 2
[D][face_test.ino:213] do_enrollment(): START ENROLLING
[D][face_test.ino:217] do_enrollment(): Face ID 1 Enrollment: Sample 2
enroll_face 1
[D][face_test.ino:213] do_enrollment(): START ENROLLING
E (24797) fr_flash: Not found
[D][face_test.ino:217] do_enrollment(): Face ID 1 Enrollment: Sample 5
2
[D][face_test.ino:213] do_enrollment(): START ENROLLING
[D][face_test.ino:217] do_enrollment(): Face ID 1 Enrollment: Sample 1
enroll_face 2
[D][face_test.ino:213] do_enrollment(): START ENROLLING
[D][face_test.ino:217] do_enrollment(): Face ID 1 Enrollment: Sample 2
enroll_face 1
```

### Other Steps to Reproduce

no other steps 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

davidbcf, 2022-04-27:
how to solve the problem : @return -2                   Flash partition not found

me-no-dev, 2022-04-27:
1.0.6 is no longer supported and the face recognition library has since changed much. Unfortunately Face Recognition is also no longer supported on ESP32 and ESP32-S2. My Guess is that you are missing a "data" partition with name "fr" to save the faces.

me-no-dev, 2022-04-27:
Here is an example partition scheme for 4MB camera support (but on the new version): https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/partitions.csv

I do not guarantee  that it will work, but this is what the error suggests. Did you try the CameraWebServer example?

davidbcf, 2022-04-28:
i tryed this partition below, it seems like error occured when save info the  3th time.

enroll_face 2
enroll_face 1
E (141165) fr_flash: Not found
enroll_face -2
enroll_face 2
enroll_face 1
E (144223) fr_flash: Not found
enroll_face -2
[image]

i tryed  CameraWebServer example, when open Face Detetion ,it crashed....

[image]

davidbcf, 2022-04-28:
is somewhere  i can find the definition table  of flash type, flash subtype and flags?

me-no-dev, 2022-04-28:
CameraWebServer is working with github master of this repo. You will have it working when we release 2.0.3

me-no-dev, 2022-05-13:
Closing because 2.0.3 is released.

closed 2022-05-13 by me-no-dev

me-no-dev, 2022-11-07:
@jlarnal you should ask this question in the esp-dl repo: https://github.com/espressif/esp-dl

jlarnal, 2022-11-08:
Thanks for the Link @Me_No_Dev

I will ask this question there, if it isn't answered yet. 😊👍

QianZhongLi, 2024-01-21:
this will be ok，without partition not found error
[image]

# Name,   Type,  SubType, Offset,   Size,    Flags
nvs,      data,  nvs,     0x9000,   0x5000,
otadata,  data,  ota,     0xe000,   0x2000,
app0,     app,   ota_0,   0x10000,  0x3d0000,
fr,0x20,0x20,  0x3e0000, 0x20000,
[image]