espressif/arduino-esp32 #6638: Unstable ethernet events.

zekageri, 2022-04-27:
### Board

esp32-wrover

### Device Description

Board is a custom board with bare ESP32-WROVER16 ( 8 mb ps ram ) chip. The problem does not occour with the package that has no "new events".

### Hardware Configuration

There are a bunch of periferials like Ethernet, I2C RTC device, two hardware UART and things like that. But again, the issue occouring with the package where new wifi events defined.

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Sometimes on bootup, ARDUINO_EVENT_ETH_GOT_IP does not occour when connecting via Ethernet.

### Sketch

```cpp
#define ETH8720_PHY_ADDR  1
#define PIN_PHY_POWER -1
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18
#define ETH_TIMEOUT 20000

void ethEvent(WiFiEvent_t event){
    Serial.printf("\nNS - WiFiEvent: %d\n", event);
    if( event == ARDUINO_EVENT_ETH_START ){
        ETH.setHostname(hsh_fileSystem.config.hostname);
        Serial.println("ETH CONNECTED!");
    }else if( event == ARDUINO_EVENT_ETH_GOT_IP ){
        hsh_networkSystem.stopAP();

        hsh_networkSystem.ethConnected      = true;
        hsh_networkSystem.connectionMode    = ETH_CONNECTION;
        hsh_networkSystem.connected         = true;
        hsh_networkSystem.shouldCheckETH    = false;
        hsh_fileSystem.config.gateway       = ETH.gatewayIP();
        hsh_fileSystem.config.currentIP     = ETH.localIP();
        hshDisplay.firstDisplay             = true;
        hshDisplay.addNoty("Ethernet connected.");

        #if NETWORK_DEBUG_MODE
            Serial.printf("\nConnected to Ethernet\n");
            Serial.printf("NS - IP:\t%s\n",          ETH.localIP().toString().c_str());
            Serial.printf("NS - MAC:\t%s\n",         ETH.macAddress().c_str());
            Serial.printf("NS - Host:\t%s\n",        ETH.getHostname());
        #endif
        hsh_networkSystem.setupMDNS();
    }else if( event == ARDUINO_EVENT_ETH_DISCONNECTED ){
        hsh_networkSystem.ethConnected      = false;
        hshDisplay.addNoty("Ethernet disconnected.");
        hsh_networkSystem.connectionMode    = NOT_CONNECTED;
        hsh_networkSystem.connected         = false;
        hsh_networkSystem.ethStartTime      = millis();
        hsh_networkSystem.shouldCheckETH    = true;
        hshDisplay.firstDisplay             = true;

        #if NETWORK_DEBUG_MODE
            Serial.println("\nNS - Ethernet disconnected");
        #endif
    }
}

void networkSystem::connectETH(){
    stopAP();
    #if NETWORK_DEBUG_MODE
        Serial.println("\nNS - Connecting to Ethernet");
    #endif
    shouldCheckETH = true;
    ETH.begin(
        ETH8720_PHY_ADDR,
        PIN_PHY_POWER,
        PIN_SMI_MDC,
        PIN_SMI_MDIO,
        ETH_PHY_LAN8720,
        ETH_CLK_MODE
    );
    WiFi.onEvent(ethEvent);
    if( !hsh_fileSystem.config.dhcp ){ // DHCP is set to true in config file so this line is never executing but the problem presist.
        ETH.config(hsh_fileSystem.config.ip, hsh_fileSystem.config.gateway, hsh_fileSystem.config.subnet,
            hsh_fileSystem.config.dns, hsh_fileSystem.config.dns2);
    }
    ethStartTime = millis();
}
```

### Debug Message

```plain
There is no debug message. The ethEvent callback just not triggering about 4/10.
```

### Other Steps to Reproduce

There is no other step. I tried with old `WiFiEvent_t event` but these gives me warnings with the following pio configuration:

```
[env:arduino-esp32]
platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board       = esp32dev
framework   = arduino

platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

board_build.f_cpu   = 240000000L

upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = ./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -DCORE_DEBUG_LEVEL=0
                ;-D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                ;-D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
              
extra_scripts = mklittlefs/replace_fs.py
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-27:
Same as https://github.com/espressif/arduino-esp32/issues/6105 ?

zekageri, 2022-04-27:
Hmmm... dont know about the link up flag. I will check on that

zekageri, 2022-04-29:
If i could check the ethernet connection without callback event handler that would be good. Just like on the wifi part:

```C
while (WiFi.status() != WL_CONNECTED) {
 // wait for wifi connection...
}
```

it would be good if i could do something like this with ethernet:

```C
while ( eth.status() != ETH_CONNECTED) {
 // wait for eth connection...
}
```

zekageri, 2022-05-03:
This issue still presist. Ethernet wont call callback function sometimes, at all.

zekageri, 2022-05-04:
Any clue on this? :/

MartinMueller2003, 2022-05-06:
I am in the same situation.

zekageri, 2022-05-17:
I have tried different configurations, DHCP, Static IP, no dns added, two and one dns added to the config,  `WiFi.mode(WIFI_OFF);` before `ETH.begin() ` but nothing seems to work. Callback is never called. :|

MartinMueller2003, 2022-05-17:
With latest version I get connected but have no callbacks. Went to a polling model and I can see that the interface comes up.

zekageri, 2022-05-17:
How do you do a polling model? I have connections with callbacks too btw

zekageri, 2022-05-17:
Do you check for `WiFi.status()` like this: `while (WiFi.status() != WL_CONNECTED) {}` ?

MartinMueller2003, 2022-05-17:
Correct.

Sampozzo, 2022-05-24:
It seems that event number fired was changed to 18 - 20 - 22:

[  3366][V][WiFiGeneric.cpp:352] _arduino_event_cb(): Ethernet Started
[  3367][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 18 - ETH_START
[  3368][V][WiFiGeneric.cpp:344] _arduino_event_cb(): Ethernet Link Up
[  3396][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
ETH Connected (SYSTEM_EVENT_ETH_CONNECTED)
[  3420][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[  3420][D][WiFiGeneric.cpp:916] _eventCallback(): ETH IP: 192.168.7.169, MASK: 255.255.252.0, GW: 192.168.5.1

Before the event numbers  was 26 - 28 - 30

I don't know if it is a bug or a desired change. Anyone know?

zekageri, 2022-05-24:
I printed out the event number and i can confirm that the callback is not even called. No events at all.

Sampozzo, 2022-05-24:
@zekageri i'm using platform.io with the latest platform (espressif32 4.2) and the latest arduino framework released (2.0.2). 

Enabling debug level 5 i have the log of my previous message. The same with dev branch (2.0.4)

me-no-dev, 2022-05-25:
events have changed: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.h

We do not use `SYSTEM_EVENT_` anymore

Sampozzo, 2022-05-25:
@me-no-dev thanks. Can i know why or a link that explain this change?

me-no-dev, 2022-05-25:
ESP-IDF changed the way it enumerates and processes events, so in order for Arduino to provide as similar API as possible, we had to change the event names and actually define them ourselves. This is since 2.0.0

Sampozzo, 2022-05-25:
@me-no-dev thank you very much!

MartinMueller2003, 2022-05-25:
Did the ESP8266 events also change?
Why is that the use of the old mechanism is not throwing compiler errors?

me-no-dev, 2022-05-25:
@MartinMueller2003 we do not follow ESP8266 so no idea. It does compile because the old enums still exist in ESP-IDF, but can not be used the same way as before anymore in Arduino.

zekageri, 2022-05-25:
Thats okay but why is my callback sometimes called and sometimes not? As you can see iam using the new event defines.