espressif/arduino-esp32 #6639: Firmware update via HTTP upload crashes the board after restart.

zekageri, 2022-04-27:
### Board

esp32-wrover

### Device Description

It is a custom board, equipped with i2c, two hardware uart, spi interface.

### Hardware Configuration

It is a custom board, equipped with i2c, two hardware uart, spi interface.

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

After i upload the firmware via HTTP file upload using `AsyncWebserver` and resetting the board and try to access it's web page i got a `pxQueue` crash immidiately, every time. Maybe it's an `AsyncWebserver` bug and not related to esp32?  But it happens only after restart... It seems to me that some flag or register is not emptyed or something.
I'm restarting it via the reset pin with a GPIO output. 

### Sketch

```cpp
void sSystem::firmwareHandler(){
    server.on("/newFirmware", HTTP_POST, [](AsyncWebServerRequest *request) {},
        [](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data,size_t len, bool final) {
            byte errCode = 1;
            if (!index) {
                hsh_Server.firmwareUpdating = true;
                int cmd = (filename.indexOf("LittleFS") > 0) ? U_SPIFFS : U_FLASH;
                int fileSize = UPDATE_SIZE_UNKNOWN;
                if(request->hasArg("fileSize")){fileSize = request->arg("fileSize").toInt();}

                if ( !Update.begin(fileSize, cmd) ) {
                    #if SS_DEBUG_MODE
                        Serial.println("Update begin failed!");
                    #endif
                    errCode = 2;
                    final = true;
                }
            }

            if (Update.write(data, len) != len) {
                #if SS_DEBUG_MODE
                    Serial.println("Update write failed!");
                #endif
                errCode = 3;
                final = true;
            }

            if (final) {
                if( errCode == 2){
                    request->send(500, "text/plain", "Update begin failed! Size mismatch or invalid file!");
                }else if( errCode == 3 ){
                    request->send(500, "text/plain", "Update write failed!");
                    hsh_Server.sendFirmwareProgress(100,true,false);
                    hsh_Server.firmwareUpdating = false;
                }else if ( !Update.end(true) ) {
                    AsyncWebServerResponse *response = request->beginResponse(500, "text/plain", "Failed");
                    hsh_Server.sendFirmwareProgress(100,true,false);
                    request->send(response);
                    hsh_Server.firmwareUpdating = false;
                } else {
                    request->onDisconnect(firmwareRespHandler);
                    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Succeded");
                    hsh_Server.sendFirmwareProgress(100,true,true);
                    request->send(response);
                }
            }
    });
}

void firmwareRespHandler(){
    hsh_Performance.beginRestart();
}

void pSystem::beginRestart(){
    shouldRestart = true;
}

void pSystem::restart(){
    if( shouldRestart ){
        shouldRestart = false;
        Serial.println("Restarting...");
        hsh_Server.closeAllSocket();
        hsh_fileSystem.config.cleanRestart = true;
        hsh_fileSystem.copyConfig();
        zones.save(true);
        vTaskDelay(100);
        digitalWrite(RESTART_PIN,HIGH); // There is a GPIO output which triggers the esp's reset pin.
        //ESP.restart();
        //forcedReset();
    }
}

/*
void pSystem::forcedReset(){
  esp_task_wdt_init(1, true);
  esp_task_wdt_add(NULL);
  while(true); 
}
*/
```

### Debug Message

```plain
assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))    

Backtrace:0x40083e19:0x3ffe98200x4008edc1:0x3ffe9840 0x400943d1:0x3ffe9860 0x4008fd75:0x3ffe9990 0x4018e747:0x3ffe99d0 0x4018e8a8:0x3ffe99f0 0x4018e9ef:0x3ffe9a20 0x4018ed01:0x3ffe9a50 0x4018e2bb:0x3ffe9ab0
  #0  0x40083e19:0x3ffe98200 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
```

### Other Steps to Reproduce

Async webserver -> firmware upload -> restart module -> load webpage -> crash

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-27:
Maybe the browser did not fully close the connection before the ESP rebooted and the server freaked. In any case, running the backtrace through the exception decoder would have given us more info

zekageri, 2022-04-27:
Yeah, iam using PIO and the decoder is turned on. There are some crash logs even the decoder cant decoder. But i will copy it manually

zekageri, 2022-04-27:
Before restart, the esp closing all connections, and if the browser detects that the ping message ( in my case it is time data every 1 sec ) is not given, it closes the connection client side and try to connect again. Its really weird if both party cant close the connection

zekageri, 2022-04-27:
How can the server freak when there is no connection after restart

me-no-dev, 2022-04-27:
there is a connection if the computer does not know that the ESP rebooted and did not close it's connection fully before that.

Jason2866, 2022-04-27:
There is no general problem with HTTP OTA upload. We use it for our project (Tasmota) and it does work. We are NOT using `AsyncWebserver`. So it looks like the combination causes trouble.

zekageri, 2022-04-27:
The whole soft restart thing is stinky from the start. It had so much trouble before. That's why i decided to go with an external solution. My esp pulls high one of it's pin to open a transistor which will let a 3v3 signal to the reset pin of the esp. But even this approach seems a little bit unreliable because of these kind of stuffs.

This problem could occour because my client side script tries to connect to the esp's server as soon as possible and the esp has no breathing room for bootup. Probably it get's a connection request immidiately when the server starts and if it is doing other stuffs on that time it freaks out... I try to delay the client side connection request after restart and see what happens.

zekageri, 2022-04-27:
It just happened again. It seems to me that it is not related to firmware upload but any kind of file upload? I don't know why my pio exception decoder not decoding the backtrace. Is there any decoder which is outside Arduino IDE and PIO?

```C
assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace:0x40083e19:0x3ffe98200x4008edc1:0x3ffe9840 0x400943d1:0x3ffe9860 0x4008fd75:0x3ffe9990 0x4018e567:0x3ffe99d0 0x4018e6c8:0x3ffe99f0 0x4018e80f:0x3ffe9a20 0x4018eb21:0x3ffe9a50 0x4018e0db:0x3ffe9ab0 
  #0  0x40083e19:0x3ffe98200 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402        

ELF file SHA256: 0000000000000000

Rebooting...
```

zekageri, 2022-04-29:
I decoded this backtrace with an empty arduino sketch and i got this:

```C
Decoding stack results
0x40083e19: psram_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32/spiram_psram.c line 864
```

It is possible that psram enable fails. I use it a lot.

I enable PSram with a platformio.ini file like this:

```
build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -DCORE_DEBUG_LEVEL=0
```

Should i enable it in my sketch too?

Jason2866, 2022-04-29:
If you have a ESP32 older than rev.3 you need the PSRAM fix when enabling PSRAM.
Your config should include:
```
build_flags = -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw
```

zekageri, 2022-04-29:
I did not know about that flag: `-mfix-esp32-psram-cache-strategy=memw`.
I will test it, thanks.

zekageri, 2022-04-29:
Did not help unfortunatelly. :/

podaen, 2022-05-10:
If there is an error in the receive of packages it just continues. It could explane your crash after boot.

VojtechBartoska, 2022-08-23:
@zekageri is this still valid?

zekageri, 2022-08-23:
It is solved with some magic. Thank you

closed 2022-08-23 by zekageri

kgobes, 2023-09-05:
@zekageri What was the resolution to this? I am having the same issue.

zekageri, 2023-09-06:
> @zekageri What was the resolution to this? I am having the same issue.

Well, i moved to a lot of forks of ESPAsyncWebserver. They all had isssues. Iam still struggeling with this lib. Now i try a different fork from esphome. This particular problem was solved with an upgrade to a new esp release i think. It wasnt the libs fault. But it just has a lot of problem. I suggest you to try other forks.