espressif/arduino-esp32 #6642: Hardware serial sometimes doesn't send every data.

zekageri, 2022-04-27:
### Board

esp32-wrover-e

### Device Description

Custom board, using ESP32-Wrover-E ( 16mb flash 8mb external ram )

### Hardware Configuration

Ethernet -> ETH_LAN_8720 chip

RTC Module -> DS3231 chip ( i2c )

Transistor for restarting the device -> 2n2222 or similar. ( GPIO 12 )

Two hardware serial UART:
 - Debug Serial 0 : RX: GPIO 3 | TX: GPIO 1
 - Custom Modbus Serial 1 : RX: GPIO 34 | TX:  GPIO 15
 

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Serial1 sometimes does not send out all the bytes and i got a frame timeout in my communication because the modules that my esp communicates doesnt understand the broken message. Happens mostly on web page load or file upload via HTTP.

Communication runs on an available core in a separate task with 115200 baud rate, tries to write as soon as possible.

### Sketch

```cpp
#define MBUS_BAUD 115200
#define MBUS_RX 34  //   6.  PIN
#define MBUS_TX 15  //   23. PIN

void mBusSetup(){
    pinSetup();
    Serial1.begin(MBUS_BAUD,SERIAL_8N1,MBUS_RX,MBUS_TX);
}

void mSystem::pinSetup(){
    pinMode(MBUS_TX_EN, OUTPUT);
    digitalWrite(MBUS_TX_EN, LOW);
}

void mSystem::startWrite(){
    mBusLock();
    //Serial1.flush();
    digitalWrite(MBUS_TX_EN, HIGH);
}

void mSystem::endWrite(){
    Serial1.flush();          // does not help
    //delayMicroseconds(200); // Does not help. Dont want magic delays
    digitalWrite(MBUS_TX_EN, LOW); // TX_ENABLE pin pulled low too soon i assume.
    mBusUnLock();
}

void mSystem::rawWrite(uint8_t * data, int length){
    int crc16 = getCRC_16(data, length);
    startWrite();
    Serial1.write( data, length );
    Serial1.write( lowByte(crc16) );
    Serial1.write( highByte(crc16) );
    endWrite();
}

#define MBUS_DISCOVER_LISTEN_TIMEOUT 150 // Modules will respond back in maximum of 15ms.

void writeReadDevice(int index){
    // This function handles which device address to write and assmebles the writable data and calls
    // rawWrite. If rawWrite returns, this function waits for the response data.
    uint8_t exampleData[] = {0xFF, 0x03, 0x00, 0x06, 0x00, 0x03};
    uint8_t responseData[20]; // Maximum response data is 10-13.
    rawWrite(exampleData, sizeof(exampleData));

    // wait for the response data. The modules will respond back if the data is understandable.
    long startMs = millis();
    while( !Serial1.available() ){
        vTaskDelay(1);
        if( millis() - startMs >= MBUS_DISCOVER_LISTEN_TIMEOUT ){
            Serial.println("FRAME TIMEOUT!");
            return;
        }
    }

    int respDataCount = 0;
    while( Serial1.available() > 0 ){
        responseData[respDataCount] = Serial1.read();
        respDataCount++;
    }
}

void commLoop(){
    writeReadDevice(deviceIndex);
    deviceIndex++; // Used for communicating with different modules
    if( deviceIndex > deviceCounter ){deviceIndex = 0;}
}

void mBusLoopTask(void* parameter){
    mBusSetup();
    for(;;){
        commLoop();
        vTaskDelay(3); // The problem occours even faster if this delay is smaller.
    }
}
```

### Debug Message

```plain
No debug message just broken data on Serial.
```

### Other Steps to Reproduce

Try to write frequently on serial1.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-27:
@zekageri 

I see you are using PlatformIO. Is your build based on Arduino Core 1.0.6?
Latest PIO Arduino Core is still 1.0.6.

Would you mind testing it using Arduino IDE with latest Arduino Core (2.0.3-RC1)?

Thanks.

zekageri, 2022-04-27:
This is my pio ini file

`[env:arduino-esp32]
platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board       = esp32dev
framework   = arduino

platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

board_build.f_cpu   = 240000000L

upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = ./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -DCORE_DEBUG_LEVEL=0
                ;-D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                ;-D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
              
extra_scripts = mklittlefs/replace_fs.py`

zekageri, 2022-04-28:
I downloaded the new Arduino IDE 2.0.0 and i can only download the 1.0.6 espressif framework.

VojtechBartoska, 2022-04-28:
Please check this guide: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

For using v2.0.3-rc1, you need to change your link in Preferences to Development release link.

zekageri, 2022-04-28:
**I know that but with PIO i'm already using the latest with this ini file:**

```
platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board       = esp32dev
framework   = arduino
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

SuGlider, 2022-04-28:
@zekageri - please check these links regarding PIO setup:
https://github.com/espressif/arduino-esp32/issues/6647#issuecomment-1111792625
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-platformio

zekageri, 2022-04-28:
Thank you, i will try this!

zekageri, 2022-04-28:
Okay, deleted cache, packages and platform folder from .platformio inside users. The sketch compiled, i did upload it to my esp32 but the problem presist. Serial.flush() does not wait for all bytes

SuGlider, 2022-04-29:
I'm pretty sure that ```Serial.flush()``` only returns when all bytes are sent.
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L319-L332

There was an issue related to MODBUS and it was fixed there. Fixed by #6133

Are you sure you are building with the latest Arduino Core version?

zekageri, 2022-04-29:
Yeah, its 4.1.0

[image]

zekageri, 2022-04-29:
This is really interesting. Is there something wrong with my code? I configured Serial2 now instead of the Serial1 and commented out the microsec delay and i rarely got frame timeout now.

If there is file system access or other mid heavy process on other tasks, i got one or two frame timeouts.

zekageri, 2022-04-29:
```C
void mBusSetup(){
    pinSetup();
    Serial2.begin(MBUS_BAUD,SERIAL_8N1,MBUS_RX,MBUS_TX);
}

void mSystem::pinSetup(){
    pinMode(MBUS_TX_EN, OUTPUT);
    digitalWrite(MBUS_TX_EN, LOW);
}

void mSystem::startWrite(){
    digitalWrite(MBUS_TX_EN, HIGH);
}

void mSystem::endWrite(){
    Serial2.flush();
    //delayMicroseconds(400);
    digitalWrite(MBUS_TX_EN, LOW);
}

void mSystem::rawWrite(uint8_t * data, int length){
    int crc16 = getCRC_16(data, length);
    startWrite();
    Serial2.write( data, length );
    Serial2.write( lowByte(crc16) );
    Serial2.write( highByte(crc16) );
    endWrite();
}

#define MBUS_LISTEN_TIMEOUT 150 // Modules will respond back in maximum of 15ms.

void writeReadDevice(int index){
    // This function handles which device address to write and assmebles the writable data and calls
    // rawWrite. If rawWrite returns, this function waits for the response data.
    uint8_t exampleData[] = {0xFF, 0x03, 0x00, 0x06, 0x00, 0x03};
    uint8_t responseData[20]; // Maximum response data is 10-13.
    rawWrite(exampleData, sizeof(exampleData));

    // wait for the response data. 
    // The modules will respond back if the data is understandable.

    long startMs = millis();
    while( !Serial1.available() ){
        vTaskDelay(1);
        if( millis() - startMs >= MBUS_LISTEN_TIMEOUT ){
            Serial.println("FRAME TIMEOUT!");
            return;
        }
    }

    int respDataCount = 0;
    while( Serial1.available() > 0 ){
        responseData[respDataCount] = Serial1.read();
        respDataCount++;
    }
}
```

vshymanskyy, 2022-04-29:
Are you absolutely sure that the output data is broken?
I currently have issues reading the data with v2.0.0+, including v2.0.3-rc1.
My issue is that data up to 128 bytes get buffered (`Serial1.available()` returns 0), but then everything arrives in a burst.

SuGlider, 2022-04-30:
There is a new API for UART called ```onReceive(callbackFunc) ```

It works as a sort of ISR... When data arrives and there is short interval of time (as long as about 11 symbols in the current baudrate), it will call right away the callback function.

It may help...

vshymanskyy, 2022-04-30:
@SuGlider thank you. `onReceive` works for the default `Serial`, but doesn't work on the Serial1, which is initialized this way:
```
//Serial1.setRxBufferSize(512); // tried this, doesn't affect anything. data is always read in chunks of 120 bytes
Serial1.begin(9600, SERIAL_8N1, 2, 15);
```
On 1.0.6 i don't see such issues

SuGlider, 2022-04-30:
Actually it works for all Serial, Serial1 and Serial2.

Just use ```Serial1.onReceive() ```

vshymanskyy, 2022-04-30:
Sure. But it looks like i have problems on the signal lavel. Sorry, it is probably unrelated to this issue.

SuGlider, 2022-04-30:
> ```c
> void mBusSetup(){
>     pinSetup();
>     Serial2.begin(MBUS_BAUD,SERIAL_8N1,MBUS_RX,MBUS_TX);
> }
>
> void mSystem::pinSetup(){
>     pinMode(MBUS_TX_EN, OUTPUT);
>     digitalWrite(MBUS_TX_EN, LOW);
> } [...]

It seems OK, if you are sending data through Serial**2** and receiving through Serial**1**.

SuGlider, 2022-05-01:
> Okay, deleted cache, packages and platform folder from .platformio inside users. The sketch compiled, i did upload it to my esp32 but the problem presist. Serial.flush() does not wait for all bytes

I just tested ```Serial.flush()```. It only returns when FIFO is empty and all data has been sent out.
It is very easy to see it when baudrate is 300bps and we send a string with about 200 characters, associating turning on/off builtin LED before/after ```Serial.print()```.

zekageri, 2022-05-02:
Can it be high baud rate problem? ( 115200 ) Or can it be that other tasks interfere, and the pin is not going to low in time?

zekageri, 2022-05-02:
```C
void mSystem::startWrite(){
    digitalWrite(MBUS_TX_EN, HIGH);
}

void mSystem::endWrite(){
    Serial1.flush();
    digitalWrite(MBUS_TX_EN, LOW);
}

void mSystem::rawWrite(uint8_t * data, int length){
    int crc16 = getCRC_16(data, length);
    startWrite();
    Serial1.write( data, length );
    Serial1.write( lowByte(crc16) );
    // Other tasks interrupts this task in here maybe??
    Serial1.write( highByte(crc16) );
    // OR here?
    endWrite();
}
```

SuGlider, 2022-05-02:
@zekageri - Just to make sure it is not bad configuration of the development environment, could you please update your PlatformIO version to the latest as described in https://piolabs.com/blog/news/platformio-oss-april-2022-updates.html

Thanks!

zekageri, 2022-05-02:
Thank you for your response. 
I did update my environmnet. 
Currently using espressif 4.2.0 and PlatformIO 5.2.5.

This results in a strange boot loop when using default_16MB.csv with `esp-wrover-kit` board.
The wrover kit's external ram and it's 16mb flash is absolutly necessary for me.
I can't go forward if this isnt solved.

I opened a new issue for this : [Boot loop issue with 16mb csv](https://github.com/platformio/platform-espressif32/issues/788)

zekageri, 2022-05-04:
Boot loop problem solved. Upgraded to latest framework, issue still presist.

Frame timeouts are inconsistent. Can not lower the baudrate because the modules that i communicate with are tied to 115200.

```C
void mSystem::startWrite(){
    //mBusLock();
    //Serial2.flush();
    digitalWrite(MBUS_TX_EN, HIGH);
}

void mSystem::endWrite(){
    Serial2.flush();
    //delayMicroseconds(200);
    digitalWrite(MBUS_TX_EN, LOW);
    //mBusUnLock();
}

void mSystem::rawWrite(uint8_t * data, int length){
    int crc16 = getCRC_16(data, length);
    startWrite();
    Serial2.write( data, length );
    Serial2.write( lowByte(crc16) );
    Serial2.write( highByte(crc16) );
    endWrite();
}
```

```C
rawWrite(writeBuffer,writeBufferSize);

long startMS = millis();
while ( !Serial2.available() ){
    vTaskDelay(1);
    if( millis() - startMS > READ_WRITE_TIMEOUT ){
        Serial.printf("FRAME TIMEOUT at millis: %d\n",millis());
        return false;
    }
}
```

**Serial:**

```
FRAME TIMEOUT at millis: 71669
FRAME TIMEOUT at millis: 127851
```

It is rarely happening with the new espressif 4.2.0. But it is there

zekageri, 2022-05-04:
Sometimes i get a bunch of timeouts on page close or refresh like this:

```C
FRAME TIMEOUT at millis: 39941
FRAME TIMEOUT at millis: 40124
FRAME TIMEOUT at millis: 40304
FRAME TIMEOUT at millis: 57933
FRAME TIMEOUT at millis: 58296
FRAME TIMEOUT at millis: 58484
FRAME TIMEOUT at millis: 58675
FRAME TIMEOUT at millis: 77164
FRAME TIMEOUT at millis: 85935
FRAME TIMEOUT at millis: 86111
FRAME TIMEOUT at millis: 86289
FRAME TIMEOUT at millis: 86468
FRAME TIMEOUT at millis: 86642
FRAME TIMEOUT at millis: 89668
FRAME TIMEOUT at millis: 89844
FRAME TIMEOUT at millis: 90020
FRAME TIMEOUT at millis: 90196
FRAME TIMEOUT at millis: 90459
```

SuGlider, 2022-05-04:
Questions: 

How much is `READ_WRITE_TIMEOUT` ? 150ms?
Why using `vTaskDelay(1);` ? It will delay for a whole RTOS Tick... I'm not sure but I think that it is about 100ms...

zekageri, 2022-05-04:
- READ_WRITE_TIMEOUT is 150ms. ( it should be max 20-30 ms since a device takes about 10-15 ms to respond )
- vTaskDelay(1) is used to allow other tasks to run while that task waits for incoming bytes. ( it does not change anything if i comment it out )
- We measured it with Oscilloscope. The devices on the bus never missed a single packet in extensive texting. They are not doing any other tasks just sending back their registers if the esp asks them to.

SuGlider, 2022-06-04:
> ```c++
> 120
> ```

It seems that IDF has actually a special behavior when RX FIFO reaches 120 bytes (full threshold).
I'm investigating it and there is a setup for "timeout always on even on FIFO Full" that is disabled by default in IDF driver.
Further information in  https://github.com/espressif/arduino-esp32/issues/6689#issuecomment-1146626896

SuGlider, 2022-06-04:
> * READ_WRITE_TIMEOUT is 150ms. ( it should be max 20-30 ms since a device takes about 10-15 ms to respond )
>
>     * vTaskDelay(1) is used to allow other tasks to run while that task waits for [...]

Possible relationship with https://github.com/espressif/arduino-esp32/issues/6689#issuecomment-1146626896

zekageri, 2022-06-05:
Meanwhile i throttled my communication task and i found out that if i write on serial every 4.7ms the esp sends out all the packets and the modules on the bus are happily answer every time. This delay is fluid because i did not used timers, just the good old millis(). But it definitely helped. Ofc it is a workaround. I want my communication task to be as fast as possible since there can be dozens of modules on the bus and it can take a while to write all of them ( and wait for every response )

zekageri, 2022-06-05:
The processors inside the modules are PICKs, they are not the fastest, so if the esp wants like 20 continous register values, that can delay the whole communication. And there are GPIO-s on some of the modules too. So it would be really good to not include this 4700 micro delay.

zekageri, 2022-06-07:
`identifier "uart_set_always_rx_timeout" is undefined`

Do i have to include something for this to work?

```C
error: 'uart_set_always_rx_timeout' was not declared in this scope    
     uart_set_always_rx_timeout(2, true);
```

zekageri, 2022-06-07:
If i use `Serial2.setRxTimeout(1);` i got frame timeouts even with the 4700 microsec delay.

zekageri, 2022-06-07:
**So my task looks like this:**

```C
void mBusLoopTask(void* parameter){
    hsh_modbus.setup();
    hsh_modbus.ready = true;
    for(;;){
         hsh_modbus.loop();
    }
}
```

**Loop looks like this:**

```C
void mSystem::loop(){
    scanBus();
    reinitSavedDevices();
    if( !shouldScanBus && !reinitingSavedDevices && !hsh_Server.firmwareUpdating && !hsh_Performance.isHwResetRunning() ){
        // MODBUS_MIN_WRITE_TIME_MICRO  -> 4700
        if( micros() - lastWriteMicros >= MODBUS_MIN_WRITE_TIME_MICRO ){
            lastWriteMicros = micros();
            writeReadDevice(deviceIndex);
            deviceIndex++;
            if( deviceIndex > deviceCounter ){ deviceIndex = 0; vTaskDelay(1); }
        }
        hsh_modules.sendPeriodicModuleInfo();
        faultReset();
        consumeMaxAmper();
        consumeThermCalibration();
    }else{
        vTaskDelay(100);
    }
}
```

SuGlider, 2022-06-14:
@zekageri - I found out that some power meters send BREAK (holds the line Low for more than 11 bits in the current baudrate) to UART at the end of a data frame.

It seems to cause this delay that you are reporting. Do you think that this is your case?

zekageri, 2022-06-15:
Thank you for the tremendous help.
The modules does not send BREAK, i'm sure of it. 
They are configured for 8 bit and a stop bit.
I can monitor their response with **multiway v10** which is a software to test modbus communcation for PC.
If the esp is out of the bus and i send commands to the modules manually with multiway, i can see that they answering correctly every time. The Multiway software is configured for the same settings which is 115200 baud, no parity, 8 data bit and 1 stop bit.

I included "driver/uart.h", commented out the delay again and on setup i set both 

```C
Serial2.setRxTimeout(1);
uart_set_always_rx_timeout(2, true);
```

The frame timeout is here again. 

But i noticed something. Regardless of these two lines, my task loop looks like this:

```C
void mBusLoopTask(void* parameter){
    hsh_modbus.setup();
    hsh_modbus.ready = true;
    for(;;){
         //if( micros() - lastWriteMicros >= MODBUS_MIN_WRITE_TIME_MICRO ){
            //lastWriteMicros = micros();
            writeReadDevice(deviceIndex); // <-- actual write function you see above. Assemble, write and wait for the packet back.
            deviceIndex++;
            if( deviceIndex > deviceCounter ){ deviceIndex = 0; vTaskDelay(1); }
       // }
    }
}
```

Notice the `vTaskDelay(1)`; for task switching happens if it writed every module. So i wanted the task switch to happen only if the esp writed every module. This results in a lot of FRAME_TIMEOUT ( usually in every 100ms or so ).

If i modify my task loop to this:

```C
void mBusLoopTask(void* parameter){
    hsh_modbus.setup();
    hsh_modbus.ready = true;
    for(;;){
         //if( micros() - lastWriteMicros >= MODBUS_MIN_WRITE_TIME_MICRO ){
            //lastWriteMicros = micros();
            vTaskDelay(1);
            writeReadDevice(deviceIndex); // <-- actual write function you see above. Assemble, write and wait for the packet back.
            deviceIndex++;
            if( deviceIndex > deviceCounter ){ deviceIndex = 0; }
       // }
    }
}
```

The FRAME_TIMEOUTS happens only a couple of seconds on a random module, but now i have CRC Mismatch too.

Sample from the debug output:

```C
FRAME TIMEOUT at millis: 3431308 module id: 20
new write CRC Mismatch. Got: FC00 Calculated: E140
new write CRC Mismatch. Got: FC00 Calculated: 21B1
new write CRC Mismatch. Got: FC00 Calculated: E140
IO - Disconnected!
new write CRC Mismatch. Got: FC00 Calculated: 21B1
new write CRC Mismatch. Got: FC00 Calculated: 21B1
IO - Disconnected!
FRAME TIMEOUT at millis: 3435624 module id: 20
IO - Disconnected!
new write CRC Mismatch. Got: FD00 Calculated: ED80
FRAME TIMEOUT at millis: 3438019 module id: 30
new write CRC Mismatch. Got: FD00 Calculated: ED80
new write CRC Mismatch. Got: FC00 Calculated: E140
IO - Disconnected!
IO - Disconnected!
new write CRC Mismatch. Got: FC00 Calculated: E140
new write CRC Mismatch. Got: FC00 Calculated: 21B1
```

**So if i let the task switch frequently ( after every write ) the response data is broken a lot of the times, and sometimes it is even timeouts.**

zekageri, 2022-06-15:
If i let `TickType_t 2` delay for the task between every write, no CRC mismatch and no frame timeout. ( only if i load a heavy webpage )

**Like this:** 

```C
void mBusLoopTask(void* parameter){
    hsh_modbus.setup();
    hsh_modbus.ready = true;
    for(;;){
        vTaskDelay(2);
        writeReadDevice(deviceIndex); // <-- actual write function you see above. Assemble, write and wait for the packet back.
        deviceIndex++;
        if( deviceIndex > deviceCounter ){ deviceIndex = 0; }
    }
}
```

SuGlider, 2022-06-15:
Just to get us in the same page:

FreeRTOS runs the tasks at 1000Hz by default in Arduino.
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L943
`vTaskDelay(1)` means delaying **at least 1ms**, depending on how the processing goes. 

But, this time depends on the other Tasks that are running in the ESP32.
I think that your project has ETH as well... The ISRs and other routines with higher priority may interfere with this timing.
It is hard to make Arduino work in a very deterministic way with several peripherals, tasks and so on.

### About the MODBUS environment:

How exactly is this wired?
Are you using a RS-485 module? Which one?
Does it use ESP32 RTS pin in order to control DE/RE pin of the RS485 module?
Or DE/RE (if any) are in shortcut and controlled by `MBUS_TX_EN` pin?

zekageri, 2022-06-15:
Yes, i aggree it can be hard to tell with so many peripherials.
Yes, i use ETH and a lot of other things, and i have a lot of other tasks.

Modbus is using an RS-485 converter. I can not tell now which one it is because it itn't a complett module but an SMD IC. ( i will look it up )
The converter uses the esp `MBUS_TX_EN` pin to determine if it has to switch to rx or tx mode, which will be pulled low by my software when it sent out all the packets and high when it starts sending. I don't use any built in pin for this. Should i?

And again, i can not use the built in modbus library because the protocoll slighly modified in my environment and some things can not work. It isn't a standard modbus protocoll. But the timing and the packets are the same.

zekageri, 2022-06-16:
As you can see in my examples before, i have these functions to write to the serial:

```C
// Pull the converter pin to HIGH to indicate the start of the packets.
void mSystem::startWrite(){
    digitalWrite(MBUS_TX_EN, HIGH);
}

// Pull the converter pin to LOW to indicate the end of the packets and flush Serial to send all the data that is left in the buffer.
void mSystem::endWrite(){
    Serial1.flush();
    digitalWrite(MBUS_TX_EN, LOW);
}

void mSystem::rawWrite(uint8_t * data, int length){
    // Compute the crc for the data.
    int crc16 = getCRC_16(data, length);

    startWrite();

    // Add the whole data to Serial buffer.
    Serial1.write( data, length );

    // Add CRC low and high byte to the Serial buffer.
    Serial1.write( lowByte(crc16) );
    Serial1.write( highByte(crc16) );

    endWrite();
}
```
I did not specify any pin in `Serial1.begin()`

SuGlider, 2022-06-17:
These fucntions should work fine... 
What exactly is the issue you are seeing with it?

I tested it here for about 12 horus sending data, using ESP32 with DIO (QIO doesn't work with default pins 9 and 10 for Serial1 beacuse these pins are used for accessing the flash).
It works fine and fast, no delay, no data loss.

zekageri, 2022-06-18:
Sorry i meant i did not specify any data controll pin which would internally go high on write and low on flush. I do it myself. My pins are the following for serial

    `#define MBUS_RX 34  //   6.  PIN 
     #define MBUS_TX 15  //   23. PIN 
     #define MBUS_TX_EN 5 
     #define MODBUS_MIN_WRITE_TIM.E_MICRO 5000`

zekageri, 2022-06-28:
Thank you i will try it now.

SuGlider, 2022-06-28:
@zekageri 

This will work:

``` cpp
// Necessary include for testing the fix
#include "driver/uart.h"

void setup() {
  // for example, start Serial2 - UART2
  Serial2.begin(115200);

  // right after starting UART2, add this code:
  uart_intr_config_t uart_intr = {
      .intr_enable_mask = (0x1<<0) | (0x8<<0),  // UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT,
      .rx_timeout_thresh = 1,
      .txfifo_empty_intr_thresh = 10,
      .rxfifo_full_thresh = 112,
  };
  uart_intr_config((uart_port_t) 2, &uart_intr);  // Two is the UART number for Arduino Serial
}
```

Thanks!

zekageri, 2022-06-28:
Thank you for the suggestions.
I will try that tomorrow.

zekageri, 2022-06-29:
As soon as i run this, i get some FRAME_TIMEOUT and no communication after that at all.
I will debug it now.

zekageri, 2022-06-29:
Here is how my code relevant part looks like.

```C
#include "driver/uart.h"

boolean mSystem::newWriteToBuffer(int index, uint16_t registerAddr){
    int byteCount       = modules[index]->writeNumBytes*2;
    int writeBufferSize = byteCount + 7;
    int respDataCount   = 0;
    boolean isSuccess   = false;

    uint8_t writeBuffer[writeBufferSize];
    uint8_t respBuffer[150];
    writeBuffer[0] = uint8_t(modules[index]->address);
    writeBuffer[1] = 0x11;
    writeBuffer[2] = highByte(registerAddr);
    writeBuffer[3] = lowByte(registerAddr);
    writeBuffer[4] = highByte(modules[index]->writeNumBytes);
    writeBuffer[5] = lowByte(modules[index]->writeNumBytes);
    writeBuffer[6] = uint8_t(byteCount);

    // put the buffer data into writeBuffer
    for(int i = 0; i < modules[index]->writeNumBytes; i++){
        writeBuffer[i*2+7] = highByte(modules[index]->writeBuffer[i]);
        writeBuffer[i*2+8] = lowByte(modules[index]->writeBuffer[i]);
    }
    rawWrite(writeBuffer,writeBufferSize);
    long startMS = millis();
    while ( !Serial1.available() ){
        //vTaskDelay(1);
        // READ_WRITE_TIMEOUT == 55ms
        if( !Serial1.available() && ( (millis() - startMS) > READ_WRITE_TIMEOUT) ){
            Serial.printf("FRAME TIMEOUT at millis: %d module id: %d\n",millis(),modules[index]->id);
            return false;
        }
    }
    while ( Serial1.available() ){
        respBuffer[respDataCount] = Serial1.read();
        respDataCount++;
    }
    //Serial1.flush();
    if( respBuffer[0] == uint8_t(modules[index]->address) ){
        uint8_t gotCRC_High = respBuffer[respDataCount-2];
        uint8_t gotCRC_Low  = respBuffer[respDataCount-1];
        uint16_t gotCRC     = (gotCRC_Low << 8) | gotCRC_High;
        uint16_t calcCRC    = getCRC_16(respBuffer,respDataCount-2);
        if( gotCRC == calcCRC ){
            if( respBuffer[1] == 0x90 || respBuffer[1] == 0x83 ){
                Serial.printf("\nNew write failed with code: %02X\n",respBuffer[2]);
                return false;
            }
            for(int i = 0; i < modules[index]->readNumBytes; i++){
                modules[index]->readBuffer[i] = (respBuffer[i*2+3] << 8) | respBuffer[i*2+4];
            }
            return true;
        }else{
            Serial.printf("new write CRC Mismatch. Got: %04X Calculated: %04X\n",gotCRC,calcCRC);
            return false;
        }
    }
    return false;
}

void mSystem::writeReadDevice(int index){
    if( modules[index] != 0 ){
        if( modules[index]->isActive ){

            if( modules[index]->onError ){
                if( millis() - modules[index]->wentToError >= 60000 ){//modules[index]->onErrorTryCounter >= 1000 ){
                    modules[index]->onError = false;
                }else{
                    return;
                }
            }

            uint16_t regWriteAddr = getRegWriteAddr(modules[index]->id);
            setThermosDefaultDisplay(index,regWriteAddr);
            if( !newWriteToBuffer(index,regWriteAddr) ){
                modules[index]->failCounter++;
                moduleCommError(index);
            }else{
                modules[index]->failCounter = 0;
                modules[index]->onError     = false;
                checkModuleFaultCode(index);
            }
        }
    }
}

void mSystem::testSerialSetup(){
    uart_intr_config_t uart_intr = {
        .intr_enable_mask = (0x1<<0) | (0x8<<0),  // UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT,
        .rx_timeout_thresh = 1,
        .txfifo_empty_intr_thresh = 10,
        .rxfifo_full_thresh = 112,
    };
    uart_intr_config((uart_port_t) 1, &uart_intr);
}

void mSystem::setup(){
    Serial1.begin(MBUS_BAUD,SERIAL_8N1,MBUS_RX,MBUS_TX);
    testSerialSetup();  // WITHOUT THIS SERIAL SETUP, IT IS WORKING.
}

void mSystem::loop(){
  writeReadDevice(deviceIndex);
  deviceIndex++;
  if( deviceIndex > deviceCounter ){ deviceIndex = 0; }
}

void mBusLoopTask(void* parameter){
    for(;;){
        hsh_modbus.loop();
        vTaskDelay(2);
    }
}
```

SuGlider, 2022-06-30:
Please try this change:

```cpp

void mSystem::testSerialSetup(){
    uart_intr_config_t uart_intr = {
        .intr_enable_mask = (0x1<<0) | (0x8<<0),  // UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT,
        .rx_timeout_thresh = 1,
        .txfifo_empty_intr_thresh = 10,
        .rxfifo_full_thresh = 1,
    };
    uart_intr_config((uart_port_t) 1, &uart_intr);
}
```

It will force the driver to read each byte, one by one very fast.

zekageri, 2022-07-01:
**This causes CRC mismatches in every response**

```C
new write CRC Mismatch. Got: 1311 Calculated: 807E
new write CRC Mismatch. Got: 110B Calculated: FFFF
new write CRC Mismatch. Got: 943D Calculated: 97BC
new write CRC Mismatch. Got: C03D Calculated: 4C34
new write CRC Mismatch. Got: 94F1 Calculated: 79B2
new write CRC Mismatch. Got: C03D Calculated: 6FDA
new write CRC Mismatch. Got: 943D Calculated: 965C
new write CRC Mismatch. Got: C03D Calculated: E6A2
new write CRC Mismatch. Got: 943D Calculated: C9C7
new write CRC Mismatch. Got: 11C0 Calculated: FFFF
new write CRC Mismatch. Got: 943D Calculated: EECA
new write CRC Mismatch. Got: C03D Calculated: BF9E
Making config from sun times calculated 
new write CRC Mismatch. Got: 943D Calculated: 838A
new write CRC Mismatch. Got: 11C0 Calculated: FFFF
new write CRC Mismatch. Got: 0B3D Calculated: 53E9
new write CRC Mismatch. Got: 943D Calculated: 81E6
new write CRC Mismatch. Got: C03D Calculated: EEE3
new write CRC Mismatch. Got: 943D Calculated: 5DCA
new write CRC Mismatch. Got: 1101 Calculated: FFFF
new write CRC Mismatch. Got: 8CF1 Calculated: 522D
new write CRC Mismatch. Got: C0F1 Calculated: 3C9C
new write CRC Mismatch. Got: 0002 Calculated: 7CA5
new write CRC Mismatch. Got: F110 Calculated: 3021
FS - (copy config) Firmware version: 1.34
```

It is possible that i have to adjust my code in order for this to work.

zekageri, 2022-07-20:
With my workaround to put a bigger delay in the loop it was working.
Now i have upgraded to the latest update and these timeouts are back again with my workaround.
The ESP misses the packets...

```
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP-WROVER-KIT
esp-wrover-kit (platform: espressif32; board: esp-wrover-kit; framework: arduino)

PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

zekageri, 2022-07-20:
In my PIO ini i did not modified the framework or the platform.

```C
[env:esp-wrover-kit]
platform = espressif32

board       = esp-wrover-kit
framework   = arduino
```

zekageri, 2022-07-22:
Can i jump between versions with this line in the PIO ini?

`platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3`

If i modify the last version number

SuGlider, 2022-07-22:
I'm not familiar with PIO... I think that maybe @Jason2866 can help on it.

zekageri, 2022-07-22:
I tested with 
```
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
```

same in every case

Jason2866, 2022-07-22:
Platformio actual latest release is 5.0 (which is Arduino core 2.0.3)
the way you did to tag the version is working and fetches the corresponding commit.
To use latest core 2.0.4 you can do
`platform = https://github.com/platformio/platform-espressif32.git`
since this branch (develop) alfeady uses core 2.0.4

zekageri, 2022-07-25:
Thank you for the suggestion.

What is the difference beetween platform_packages, platform, and framework?

Framework should be the ESP IDF, but my default PIO configuration for the framework is arduino, but i have seen arduino_espressif too since Arduino framework is top on IDF.

Jason2866, 2022-07-25:
`platforms` includes all needed toolchains, tools and the framework (here Arduino) since defined as wanted in platformio setup.
With `platform_packages` you can define (override) anything what is default defined in `platforms`

zekageri, 2022-07-25:
I see. Thank you for the explanation. I will investigate it, it might be a problem from my side.

VojtechBartoska, 2022-08-23:
@zekageri Hi, can I consider this as solved?

zekageri, 2022-08-23:
Sorry. It seems the update solved the problem. Thank you.

closed 2022-08-23 by zekageri