espressif/arduino-esp32 #6643: How to connect esp s3 and sensors through spi/iic communication?

South-River, 2022-04-27:
### Board

ESP-S3-12K-kit

### Device Description

ESP-S3-12K-kit

### Hardware Configuration

## SPI
pin
- 14 - CS0
- 15 - CS1
- 16 - MISO
- 17 - MOSI
- 18 - SCLK
 
## IIC 
- 18 - SCL
- 17 - SDA

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

240Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

It had confused me several days. I can't get my oled screen display words via SPI/IIC and I can't find any tutorial on how to use SPI/IIC on ESP S3 in arduino. I think the screen's library(Adafruit_SSD1306) should be fine because I can use it on Arduino UNO and ESP32. 

### Sketch

```cpp
/* Definition of SPI Pins */
#define HSPI_MISO 16 //19 
#define HSPI_MOSI 17 //23 
#define HSPI_SCLK 18
#define HSPI_CS0 14  // accel chip
#define HSPI_CS1 15  // gyro chip

SPIClass hspi(HSPI);
void setup() 
{
  /* SPI init */
  hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_CS0); //SCLK, MISO, MOSI, SS 
  hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_CS1); //SCLK, MISO, MOSI, SS 

  int status;
  /* USB Serial to print data */
  Serial.begin(115200);

  /* maximize cpu frequency */
  setCpuFrequencyMhz(240);
  Serial.print("CPU Frequency: ");
  Serial.println(getCpuFrequencyMhz());
  
  delay(1000);
}
```

```#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED 寬度像素
#define SCREEN_HEIGHT 64 // OLED 高度像素

// 設定OLED
#define OLED_RESET     4 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

//自訂圖形，從Image2LCD轉過來的，尺寸是128x48，少掉的16個像素是上方黃色區塊(假設是用藍黃雙色的OLED)
//本範例為了在上方區塊顯示文字，所以圖形就沒有涵蓋整個高度

static const unsigned char PROGMEM logo_bmp[] =
{ 
0XFF,0XFF,0XFF,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFF,0XFF,0XFF,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFF,0XFF,0XF8,0X7F,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFF,0XFF,0XF8,0X3F,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFE,0XFF,0XF8,0X3F,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFE,0X1F,0XFC,0X7F,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFC,0X03,0XFD,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XE4,0X00,0XFB,0XFF,0XFF,0X83,0X00,0X18,0X00,0X60,0X00,0X38,0X02,0X00,0X00,0X00,
0XE4,0X00,0X1B,0XFF,0XFF,0X83,0X7E,0X18,0X3F,0XFF,0XC0,0XFE,0X03,0X0F,0XFF,0XFC,
0XCC,0X00,0X03,0XFF,0XFF,0X83,0X7F,0XFE,0X7F,0XFF,0XC3,0XC7,0XB3,0X0F,0XFF,0XFC,
0XC8,0X7C,0X00,0XFF,0XFF,0X83,0X46,0XFC,0X01,0XF8,0X07,0XFE,0XB3,0X00,0X0C,0X00,
0XC8,0XFE,0X00,0X1F,0XFF,0X86,0XE6,0XD8,0X03,0XFE,0X00,0XFE,0X33,0X00,0X0C,0X00,
0X99,0XFF,0X00,0X07,0XFF,0X87,0XF4,0XD8,0X1F,0X67,0X80,0X00,0X33,0X00,0X0C,0X00,
0X91,0XFF,0X00,0X00,0XFF,0X8E,0X1C,0XD8,0X7C,0X61,0XE3,0XFF,0X33,0X00,0X0C,0X00,
0X91,0XFF,0X00,0X00,0X1F,0X8E,0X1B,0XFE,0X64,0X62,0X23,0XFF,0XB3,0X00,0X0C,0X00,
0XE1,0XFF,0X00,0X00,0X1F,0X9E,0X31,0XFE,0X0C,0X26,0X03,0XFF,0XB3,0X00,0X0C,0X00,
0XE0,0XFE,0X01,0X80,0X1F,0X9E,0XE1,0X18,0X7F,0X9F,0XE3,0XFF,0XB3,0X00,0X0C,0X00,
0XE0,0X7C,0X03,0XE0,0X3F,0X86,0X43,0X08,0XFF,0X9F,0XE3,0X01,0XB3,0X00,0X0C,0X00,
0XC0,0X38,0X07,0XE0,0X3F,0X86,0XFF,0XFC,0X0C,0X07,0X03,0XFF,0XB3,0X00,0X0C,0X00,
0XF0,0X00,0X07,0XF0,0X4F,0X86,0XFF,0XFC,0X1E,0X0F,0X03,0XFF,0X33,0X00,0X0C,0X00,
0XFE,0X00,0X07,0XE0,0X4F,0X86,0X07,0X80,0X3F,0X9F,0X83,0X00,0X33,0X00,0X0C,0X00,
0XC7,0XC0,0X07,0XE0,0X9F,0X86,0X0F,0XC0,0X3D,0X9E,0XC2,0X7F,0X33,0X00,0X0C,0X00,
0X81,0XF0,0X03,0XC0,0X9F,0X86,0X3F,0X70,0X6C,0X36,0XE6,0X7F,0X83,0X00,0X0C,0X00,
0X80,0X3E,0X00,0X01,0X9F,0X86,0XF3,0X3C,0XCC,0X66,0X66,0X61,0X83,0X00,0X0C,0X00,
0XC0,0X07,0X80,0X01,0X3F,0X86,0XE3,0X0C,0X0C,0X06,0X06,0X7F,0X83,0X1F,0XFF,0XFE,
0XF8,0X01,0XF0,0X01,0X31,0X86,0X03,0X00,0X0C,0X06,0X06,0X7F,0XBE,0X1F,0XFF,0XFE,
0XBF,0X00,0X3E,0X03,0XF1,0X86,0X03,0X00,0X04,0X02,0X00,0X41,0X1C,0X00,0X00,0X00,
0X8F,0XC0,0X0F,0X83,0XF1,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XCC,0XE0,0X01,0XF7,0XF3,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XCC,0X1C,0X00,0X3F,0XF1,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X8D,0XC7,0XC0,0X0F,0XF0,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X8E,0X38,0XF8,0X0F,0XF8,0X83,0X00,0X00,0X30,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X83,0X8E,0X1E,0X0F,0XFF,0X82,0X00,0X00,0X30,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X80,0X71,0XC7,0XDF,0XF3,0X80,0X00,0X00,0X30,0X00,0X00,0X00,0X00,0X00,0X30,0X00,
0X80,0X1C,0X67,0XFF,0XF9,0X83,0X7D,0XCF,0X33,0X39,0XE0,0XE3,0X9F,0X70,0X7A,0X26,
0X81,0XE3,0X8C,0X7F,0XC7,0X83,0X6E,0XC9,0XB6,0X6D,0XE1,0X86,0XDD,0XD8,0X33,0X66,
0X81,0XFC,0XF8,0X1F,0X83,0X83,0X66,0X61,0XBC,0X4D,0X81,0X0C,0X59,0X98,0X33,0X74,
0X80,0XFE,0X13,0X1C,0X47,0X83,0X66,0X6F,0XBC,0XFD,0X83,0X0C,0X79,0X98,0X33,0X54,
0X80,0X3E,0X00,0X24,0X6F,0X83,0X66,0X69,0XB6,0XC1,0X81,0X0C,0X59,0X98,0X31,0XDC,
0X83,0X3E,0X00,0X64,0X7F,0X83,0X66,0X79,0XB6,0X61,0X8D,0X86,0XD9,0X9B,0X31,0X98,
0X83,0X70,0X00,0X47,0XFF,0X83,0X66,0X6F,0XB3,0X39,0X8C,0XE3,0X99,0X9B,0X19,0X98,
0X87,0XF0,0X00,0XC7,0XFF,0X83,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X87,0XF0,0X00,0X8F,0XFF,0X86,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X81,0XE0,0X01,0XFF,0XFF,0X84,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X80,0X60,0X03,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X80,0X00,0X03,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0X80,0X00,0X67,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,
0XFF,0XFF,0XFF,0XFF,0XFF,0X80,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00,0X00, };

#define SCL_PIN 18
#define SDA_PIN 17

void setup() {
  Serial.begin(115200);
  Wire.begin(SDA_PIN, SCL_PIN, 400000UL);

  // 偵測是否安裝好OLED了
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // 一般1306 OLED的位址都是0x3C
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }

  // 顯示Adafruit的LOGO，算是開機畫面
  display.display();
  delay(1000); // 停1秒

  // 清除畫面
  display.clearDisplay();

  testdrawstyles();    // 測試文字

  testdrawbitmap();    // 顯示圖形

}

void loop() {
}

void testdrawstyles(void) {
  unsigned long timer=micros();
  display.clearDisplay();
  display.setTextSize(2);             // 設定文字大小
  display.setTextColor(1);        // 1:OLED預設的顏色(這個會依該OLED的顏色來決定)
  display.setCursor(0,0);             // 設定起始座標
  display.print("Hello OLED");        // 要顯示的字串
  display.display();                  // 要有這行才會把文字顯示出來
  Serial.println("testdrawstyles");
  Serial.println(micros()-timer);
  delay(1000);
}

void testdrawbitmap(void) {
  //display.clearDisplay();
  //顯示圖形，x,y,圖形文字,寬,高,1:OLED預設的顏色(這個會依該OLED的顏色來決定)
  unsigned long timer=micros();
  display.drawBitmap(0,16,logo_bmp, 128, 48, 1);
  display.display();  // 要有這行才會把文字顯示出來
  Serial.println("testdrawbitmap");
  Serial.println(micros()-timer);
  delay(1000);
}
```
```

### Debug Message

```plain
ehhh?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-27:
Try `Wire.setPins(SDA_PIN, SCL_PIN);`, then `Wire.begin();`. Adafruit's library switches the I2C clock itself. SPI code looks fine. You can call `uint8_t in_data = hspi.transfer(out_data);`

South-River, 2022-04-27:
> Try `Wire.setPins(SDA_PIN, SCL_PIN);`, then `Wire.begin();`. Adafruit's library switches the I2C clock itself. SPI code looks fine. You can call `uint8_t in_data = hspi.transfer(out_data);`

Thank you for your answer.
But sadly, I had tried your idea and failed to light up my screen(ORZ).

South-River, 2022-04-27:
> @South-River -- Regarding the I2C SSD1306:
> 
> I can confirm that your sketch works fine exactly as you posted it. Maybe the I2C address is wrong? Please try 0x3D instead:
> 
> ```c++
>   // 偵測是否安裝好OLED了
>   if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3D)) { // 一般1306 OLED的位址都是0x3C
> ```
> 
> [image]

AMAZING!!! However I tried your suggestion and failed.
Since our code are same, I'm considering maybe this problem comes from the board:(? I would try to ask the ai-thinker's engineers that is there any difference between their's ESP-S3-12K-kit and Espressif's ESP32-S3-DevKitC-1. I used to thought that they are same because they both annouced to use ESP-S3, and a tutorial guides me to choose "ESP32S3 DEV MODULE" to compile it.

SuGlider, 2022-04-27:
@South-River 

You can also try to use the I2C default pins (SDA 8 and SCL 9) - just remove the ```Wire.begin(SDA_PIN, SCL_PIN, 400000UL);``` line from the sketch and use these 2 default pins.

Again, it may be worth to try both I2C addresses 0x3C and 0x3D.
Another thing you can do to confirm which I2C address is the correct one, would be to run the ```Wire-->WireScan.ino``` example from ESP32S3 examples folder. It will tell you which I2C addresses are available in the bus.

``` cpp
#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.begin();
}

void loop() {
  byte error, address;
  int nDevices = 0;

  delay(5000);

  Serial.println("Scanning for I2C devices ...");
  for(address = 0x01; address < 0x7f; address++){
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if(error != 2){
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0){
    Serial.println("No I2C devices found");
  }
}
```

The ouput I get with the OLED (with 0x3C address) that I'm using is:

```
SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x428
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29ac
entry 0x403b61d8
Scanning for I2C devices ...
I2C device found at address 0x3C
Scanning for I2C devices ...
I2C device found at address 0x3C
```

South-River, 2022-04-28:
> @South-River
>
> You can also try to use the I2C default pins (SDA 8 and SCL 9) - just remove the ```Wire.begin(SDA_PIN, SCL_PIN, 400000UL);``` line from the sketch and use these 2 default pins. [...]

Thank you(orzorz). I tried your suggestion  to run "WireScan.ino" with sda connected to 8 and scl connected to 9， and also sda connected to 17，scl connected to 18. But it tells me that i2c find nothing.

South-River, 2022-04-28:
I solved this problem because I get a wrong picture and link sensor on wrong pins.

closed 2022-04-28 by South-River