espressif/arduino-esp32 #6644: error in updating Firmware OTA on esp32 using LTE module

mudgalp, 2022-04-27:
### Board

ESP32 DEV MODULE

### Device Description

The device is a RS485 Based Modbus to 4G_LTE Dataloger device.
Consist of esp32-wroom module 4MB Flash Size.
MAX485CSA IC for Modbus Communication >> connected on serial1 at 9600 baudrate
Quectel 4G-LTE module EC200S for Operating in remote area(There is no wifi network) >> connected on serial2 at 115200 baudrate.
and other 12-24v to different power regulations.

all the debugging is done from serial0.

### Hardware Configuration

#define ESP32_U1_TX   25  // MAX485_RX --> 
#define ESP32_U1_RX   27  // MAX485_TX -->
#define ESP32_U2_TX   17  // GSM_RX --> 
#define ESP32_U2_RX   16  // GSM_TX -->
#define MAX485_EN     26
#define LED           2
#define BUZZ          33
#define CONFIG        35
#define ADC           34
#define GSMPWR        32
#define GSMDTR        4

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

hello everyone,
I'm working on a device in which I'm trying to do Firmware Over the air update of esp32-wroom module,
but i'm using another 4G-LTE module from Quictel (EC200S) to download the bin file from server.

everything looks fine but in the last it gives me error to update as >>Error Occurred. Error #: 6

here is the ota Function i'm using.
please help

### Sketch

```cpp
bool FirmwareVersionCheck()
{
  bool value;
  int httpCode ;
[... 674 lines cut ...]
[4-25 14:47:32.3]line:Content-Type: application/octet-stream
[4-25 14:47:32.3]Got application/octet-stream payload.
[4-25 14:47:32.3]line:Content-Length: 848560
[4-25 14:47:32.3]Got 848560 bytes from server
[4-25 14:47:32.3]line:Last-Modified: Sun, 24 Apr 2022 15:25:09 GMT
[4-25 14:47:32.3]line:Connection: keep-alive
[4-25 14:47:32.3]line:ETag: "62656bd5-cf2b0"
[4-25 14:47:32.3]line:Accept-Ranges: bytes
[4-25 14:47:32.3]line:
[4-25 14:47:32.3]This Line Was empty one times
[4-25 14:47:32.3]httpResponseLineCount is :1
[4-25 14:47:32.3]This Line Was empty two times
[4-25 14:47:32.3]Breaking While() Now
[4-25 14:47:32.3]contentLength : 848560, isValidContentType : 1
[4-25 14:47:32.3][D][Updater.cpp:132] begin(): OTA Partition: app1
[4-25 14:47:32.3]Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!
[4-25 14:54:16.9]Written only : 784906/848560. Retry?
[4-25 14:54:16.9]Error Occurred. Error #: 6
[4-25 14:54:23.1]<----------BOOT PROCESS COMPLETE---------->
[4-25 14:54:23.1]
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-27:
@mudgalp 
We are not supporting Arduino Core 1.0.6 anymore.
Would it be possible for you to test it using the latest Arduino Core Version (2.0.2 or 2.0.3-RC1)?

Thanks!

mudgalp, 2022-04-27:
tried updating the Esp32 Arduino core to 2.0.2
also updated the arduino ide to 1.8.19

still getting same response. please help

here is the new log recorded out of the device.

Thank you 

[4-28 01:42:27.3]Starting FOTA to update New Firmware
[4-28 01:42:27.3]
[4-28 01:42:27.3]<<<  ATTENTION !!! DO NOT TURN OFF THE DEVICE...  >>>
[4-28 01:42:27.3]
[4-28 01:42:27.3]Updating the Device Firmware Now!
[4-28 01:42:27.3]
[4-28 01:42:27.3]Please Wait untill the Device Restarts
[4-28 01:42:27.3]
[4-28 01:42:27.3]http://data.silicisroadindustries.com/silicis_fota/buslog4g_fota_lpa/update.bin
[4-28 01:42:27.3]
[4-28 01:42:28.3]GSM REGISTERED ON NETWORK
[4-28 01:42:28.3]
[4-28 01:42:29.4]Signal Strength : 20
[4-28 01:42:29.4]
[4-28 01:42:30.5]GPRS CONNECT SUCCESSFULLY
[4-28 01:42:30.5]
[4-28 01:42:35.1]URL CONNECT SUCCESSFULLY
[4-28 01:42:35.1]
[4-28 01:42:37.8]WAITING FOR GET COMMAND RESPONSE
[4-28 01:42:37.8]
[4-28 01:42:38.8]WAITING FOR GET COMMAND RESPONSE
[4-28 01:42:38.8]
[4-28 01:42:40.4]GET RESPONSE IS:200
[4-28 01:42:40.4]
[4-28 01:42:40.4]GET RESPONSE IS OK
[4-28 01:42:40.4]
[4-28 01:42:41.4]line:
[4-28 01:42:41.4]This Line Was empty one times
[4-28 01:42:41.4]httpResponseLineCount is :0
[4-28 01:42:41.4]line:CONNECT
[4-28 01:42:41.4]line:HTTP/1.1 200 OK
[4-28 01:42:41.4]line:Server: nginx/1.18.0 (Ubuntu
[4-28 01:42:41.4]line:Date: Wed, 27Apr 2022 20:12:3 GMT
[4-28 01:42:41.4]line:Content-Type: application/octet-stream
[4-28 01:42:41.4]Got application/octet-stream payload.
[4-28 01:42:41.4]line:Content-Length: 848560
[4-28 01:42:41.4]Got 848560 bytes from server
[4-28 01:42:41.4]line:Last-Modified: Sun, 24 Apr 2022 15:25:09 GMT
[4-28 01:42:41.4]line:Connection: kee`-alive
[4-28 01:42:41.4]line:ETag: "62656bd5-cf2b0"
[4-28 01:42:41.4]line:Accept-Ranges: bytes
[4-28 01:42:41.4]line:
[4-28 01:42:41.4]This Line Was empty one times
[4-28 01:42:41.4]httpResponseLineCount is :1
[4-28 01:42:41.4]This Line Was empty two times
[4-28 01:42:41.4]Breaking While() Now
[4-28 01:42:41.4]contentLength : 848560, isValidContentType : 1
[4-28 01:42:41.4][ 75057][D][Updater.cpp:133] begin(): OTA Partition: app1
[4-28 01:42:41.4]Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!
[4-28 01:49:25.8]Written only : 743006/848560. Retry?
[4-28 01:49:25.8]Error Occurred. Error #: 6

mudgalp, 2022-05-04:
waiting for response...

mudgalp, 2022-05-10:
### Tried printing the error using **Update.printError(Serial);** here are the logs.
### its says Error Occurred. Error #: 6
### Stream Read Timeout

//-----------------------------------------------------------------------------------------------------------------

[5-10 09:42:22.3]<----------BOOTING UP THE DEVICE---------->
[5-10 09:42:22.3]
[5-10 09:42:22.3]EEPROM Begin Successfully
[5-10 09:42:22.3]
[5-10 09:42:22.3]Last Data Upload Interval = 1
[5-10 09:42:22.3]
[5-10 09:42:22.3]Files Stored In Flash = 0
[5-10 09:42:22.3]
[5-10 09:42:22.3]SSID :
[5-10 09:42:22.3]
[5-10 09:42:22.3]PASS :
[5-10 09:42:22.3]
[5-10 09:42:22.3][  2064][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[5-10 09:42:22.5][  2166][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[5-10 09:42:22.5][  2167][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[5-10 09:42:24.0][  3667][E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long
[5-10 09:42:24.0][  3667][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: silicis
[5-10 09:42:24.0][  3669][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: PM12
[5-10 09:42:24.0][  3676][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: PM13
[5-10 09:42:24.0]CONNECTING TO WIFI...
[5-10 09:42:24.0]
[5-10 09:42:26.2][  5918][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 2
[5-10 09:42:26.2][  5919][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[5-10 09:42:26.2][  5923][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[5-10 09:42:26.2][  5928][I][WiFiMulti.cpp:119] run(): [WIFI] 2 networks found
[5-10 09:42:26.2][  5933][D][WiFiMulti.cpp:151] run():        0: [11][52:9B:68:86:29:7B] PM (-45) *
[5-10 09:42:26.2][  5941][D][WiFiMulti.cpp:151] run():        1: [6][40:95:BD:0C:05:0C] JioFi_10C050C (-88) *
[5-10 09:42:26.3][  5949][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
[5-10 09:42:27.3]UNABLE TO CONNECT ANY WIFI
[5-10 09:42:27.3]
[5-10 09:42:27.4]CHECKING GSM MODULE IN 10Sec.
[5-10 09:42:27.4]
[5-10 09:42:39.9]MODULE OK
[5-10 09:42:39.9]
[5-10 09:42:39.9]SETTING OTHER PARAMETERS TO GSM
[5-10 09:42:39.9]
[5-10 09:42:42.1]IMEI : 861190059737848
[5-10 09:42:42.1]
[5-10 09:42:43.7]CHECKING FOR SIM CARD
[5-10 09:42:43.7]
[5-10 09:42:44.8]SIM CARD FOUND OK
[5-10 09:42:44.8]
[5-10 09:42:45.9]GSM INITIALIZATION COMPLETE
[5-10 09:42:45.9]
[5-10 09:42:47.0]GSM IS REGISTERED ON NETWORK
[5-10 09:42:47.0]
[5-10 09:42:48.2]NETWORK TIME SYNC SUCCESSFUL
[5-10 09:42:48.2]
[5-10 09:42:49.3]SETTING SMS COMMANDS
[5-10 09:42:49.3]
[5-10 09:42:49.9]SETTING HTTP COMMANDS
[5-10 09:42:49.9]
[5-10 09:42:53.2]<--APN SET FOR JIO-->
[5-10 09:42:53.2]
[5-10 09:42:55.8]STARTING GPRS/PDP CONTEXT CONNECTION
[5-10 09:42:57.8]CONTEXT PROFILE IS OK
[5-10 09:42:57.8]
[5-10 09:42:58.8]GPRS/PDP CONTEXT PROFILE IS ACTIVATED
[5-10 09:42:58.8]
[5-10 09:42:59.8]
[5-10 09:42:59.8]+QIACT: 1,1,1,"25.117.218.4"
[5-10 09:42:59.8]
[5-10 09:42:59.8]OK
[5-10 09:42:59.8]
[... 71 lines cut ...]
[5-10 09:43:22.2]line:Date: Tue, 10 May 2022 04:13:20 GMT
[5-10 09:43:22.2]line:Content-Type: application/octet-stream
[5-10 09:43:22.2]Got application/octet-stream payload.
[5-10 09:43:22.2]line:Content-Length: 774096
[5-10 09:43:22.2]Got 774096 bytes from server
[5-10 09:43:22.2]line:Last-Modified: Mon, 09 May 2022 03:38:48 GMT
[5-10 09:43:22.2]line:Connection: keep-alive
[5-10 09:43:22.2]line:ETag: "62788cc8-bcfd0"
[5-10 09:43:22.2]line:Accept-Ranges: bytes
[5-10 09:43:22.2]line:
[5-10 09:43:22.2]This Line Was empty one times
[5-10 09:43:22.2]httpResponseLineCount is :1
[5-10 09:43:22.2]This Line Was empty two times
[5-10 09:43:22.2]Breaking While() Now
[5-10 09:43:22.2]contentLength : 774096, isValidContentType : 1
[5-10 09:43:22.2][ 61899][D][Updater.cpp:133] begin(): OTA Partition: app1
[5-10 09:43:22.2]Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!
[5-10 09:50:00.0]Written only : 686073/774096. Retry?
[5-10 09:50:00.0]Stream Read Timeout
[5-10 09:50:00.0]Error Occurred. Error #: 6
[5-10 09:50:00.0]Stream Read Timeout
[5-10 09:50:06.2]<----------BOOT PROCESS COMPLETE---------->
[5-10 09:50:06.2]

//-----------------------------------------------------------------------------------------------------------------

SuGlider, 2022-05-10:
I have no LTE modem/module in order to test and reproduce the issue with your sketch.
Could you simplify it to the smallest possible sketch that reproduces the error, not using the LTE module?

SuGlider, 2022-05-10:
>[5-10 09:50:00.0]Error Occurred. Error #: 6
>[5-10 09:50:00.0]Stream Read Timeout

Apparently there is an issue with the communication to the OTA server over LTE that is timing out.

You can try to increase the timeout, by using `Serial2.setTimeout(milliseconds)` -- default of Stream Class is 1000 (1 second) time out. https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/

Do it right after `Serial2.begin()`

mudgalp, 2022-05-11:
ok I'll try and get back to you.
thanks

mudgalp, 2022-05-11:
> 

there is no error without the GSM/LTE module, it only occurs when the .bin file is being downloaded using GSM/LTE module and provided to the Update.WriteStream() function.

i'm performing OTA using WIFi and httpUpdate, its works fine everytime.

mudgalp, 2022-05-11:
> > [5-10 09:50:00.0]Error Occurred. Error #: 6
> > [5-10 09:50:00.0]Stream Read Timeout
> 
> Apparently there is an issue with the communication to the OTA server over LTE, which is timing out.
> 
> You can try to increase the timeout, by using `Serial2.setTimeout(milliseconds)` -- default of Stream Class is 1000 (1 second) time out. https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/
> 
> Do it right after `Serial2.begin()`

i'll try this and get back to you.

mudgalp, 2022-05-11:
> Apparently there is an issue with the communication to the OTA server over LTE, which is timing out.
>
> You can try to increase the timeout, by using `Serial2.setTimeout(milliseconds)` -- default of [...]

tried this , Serial.setTimeout(5000).
slowed every serial2 write and read event.
and didn't solve the issue.

SuGlider, 2022-05-11:
I see.
What do you use as OTA server of the firmware?

SuGlider, 2022-05-11:
From #6727 there is a hint...
Can you set a big RxBuffer for the Serial2?

```
  Serial2.setRxBufferSize(4096);
  Serial2.setTimeout(0);
  Serial2.begin(115200);
```

primus192, 2022-05-11:
> > > [5-10 09:50:00.0]Error Occurred. Error #: 6
> > > [5-10 09:50:00.0]Stream Read Timeout
> > 
> > 
> > Apparently there is an issue with the communication to the OTA server over LTE, which is timing out.
> > You can try to increase the timeout, by using `Serial2.setTimeout(milliseconds)` -- default of Stream Class is 1000 (1 second) time out. https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/
> > Do it right after `Serial2.begin()`
> 
> tried this , Serial2.setTimeout(5000). slowed every serial2 write and read event. and didn't solve the issue.

While on arduino 1.0.6 Try setting setRxBufferSize(1024); and flush(); directly before executing Update.writeStream(Serial2); It fixed the problem for me in arduino 1.0.6. However this method did not work for me in new framework even if i set buffer size to 4096 or more before begin of the serial. GSM modules tend to hang once in a while and then throwing large chunks of data at once and the buffer overflows.

Try this way (arduino 1.0.6)

Serial2.setRxBufferSize(1024);
Serial2.flush();
 if (Update.begin(_contentLength)) {
        size_t written = Update.writeStream(Serial2);
        if (written == _contentLength) {
            log_i("Written: %s successfully.", String(written));
        }

mudgalp, 2022-05-12:
> I see. What do you use as OTA server of the firmware?

im using linux based http server made by my team.

mudgalp, 2022-05-12:
> While on arduino 1.0.6 Try setting setRxBufferSize(1024); and flush(); directly before executing Update.writeStream(Serial2); It fixed the problem for me in arduino 1.0.6. However this method did [...]

will try this and let you know.

Thank you

mudgalp, 2022-05-12:
> From #6727 there is a hint... Can you set a big RxBuffer for the Serial2?
> 
> ```
>   Serial2.setRxBufferSize(4096);
>   Serial2.setTimeout(0);
>   Serial2.begin(115200);
> ```

tried this, no response.

mudgalp, 2022-05-12:
> Serial2.setRxBufferSize(1024);
> Serial2.flush();

I think this is the problem in your new framework not working issue.

**### code**

`Serial2.setRxBufferSize(1024);
                Serial2.flush();
                // Check if there is enough to OTA Update
                bool canBegin = Update.begin(contentLength, 0, 2, 1, NULL);`

**debug output**>>

[ 77722][E][HardwareSerial.cpp:494] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.

primus192, 2022-05-12:
> > > > > [5-10 09:50:00.0]Error Occurred. Error #: 6
> > > > > [5-10 09:50:00.0]Stream Read Timeout
> > > > 
> > > > 
> > > > Apparently there is an issue with the communication to the OTA server over LTE, which is timing out.
> > > > You can try to increase the timeout, by using `Serial2.setTimeout(milliseconds)` -- default of Stream Class is 1000 (1 second) time out. https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/
> > > > Do it right after `Serial2.begin()`
> > > 
> > > 
> > > tried this , Serial2.setTimeout(5000). slowed every serial2 write and read event. and didn't solve the issue.
> > 
> > 
> > While on arduino 1.0.6 Try setting setRxBufferSize(1024); and flush(); directly before executing Update.writeStream(Serial2); It fixed the problem for me in arduino 1.0.6. However this method did not work for me in new framework even if i set buffer size to 4096 or more before begin of the serial. GSM modules tend to hang once in a while and then throwing large chunks of data at once and the buffer overflows.
> > Try this way (arduino 1.0.6)
> > Serial2.setRxBufferSize(1024); Serial2.flush(); if (Update.begin(_contentLength)) { size_t written = Update.writeStream(Serial2); if (written == _contentLength) { log_i("Written: %s successfully.", String(written)); }
> 
> I think this is the problem in your new framework not working issue.
> 
> **### code**
> 
> `Serial2.setRxBufferSize(1024);` `Serial2.flush();`// Check if there is enough to OTA Update `bool canBegin = Update.begin(contentLength, 0, 2, 1, NULL);`
> 
> **debug output**>>
> 
> [ 77722][E][HardwareSerial.cpp:494] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.

I know that, i moved this before serial begin and it still did not works in new framework.

SuGlider, 2022-05-12:
Possible Arduino Stream Class issue related to https://github.com/espressif/arduino-esp32/issues/6733#issuecomment-1125349326

Investigating it...

mudgalp, 2022-05-13:
> > > > > > [5-10 09:50:00.0]Error Occurred. Error #: 6
> > > > > > [5-10 09:50:00.0]Stream Read Timeout
> > > > > 
> > > > > 
> > > > > Apparently there is an issue with the communication to the OTA server over LTE, which is timing out.
> > > > > You can try to increase the timeout, by using `Serial2.setTimeout(milliseconds)` -- default of Stream Class is 1000 (1 second) time out. https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/
> > > > > Do it right after `Serial2.begin()`
> > > > 
> > > > 
> > > > tried this , Serial2.setTimeout(5000). slowed every serial2 write and read event. and didn't solve the issue.
> > > 
> > > 
> > > While on arduino 1.0.6 Try setting setRxBufferSize(1024); and flush(); directly before executing Update.writeStream(Serial2); It fixed the problem for me in arduino 1.0.6. However this method did not work for me in new framework even if i set buffer size to 4096 or more before begin of the serial. GSM modules tend to hang once in a while and then throwing large chunks of data at once and the buffer overflows.
> > > Try this way (arduino 1.0.6)
> > > Serial2.setRxBufferSize(1024); Serial2.flush(); if (Update.begin(_contentLength)) { size_t written = Update.writeStream(Serial2); if (written == _contentLength) { log_i("Written: %s successfully.", String(written)); }
> > 
> > 
> > I think this is the problem in your new framework not working issue.
> > **### code**
> > `Serial2.setRxBufferSize(1024);` `Serial2.flush();`// Check if there is enough to OTA Update `bool canBegin = Update.begin(contentLength, 0, 2, 1, NULL);`
> > **debug output**>>
> > [ 77722][E][HardwareSerial.cpp:494] setRxBufferSize(): RX Buffer can't be resized when Serial is already running.
> 
> I know that, i moved this before serial begin and it still did not work in new framework. This solution only works for me in 1.0.6. Tried doing this in 1.0.6?

i'll try that now.

mudgalp, 2022-05-15:
> I know that, i moved this before serial begin and it still did not work in new framework. This solution only works for me in 1.0.6. Tried doing this in 1.0.6?

tried this... no success.

mudgalp, 2022-05-20:
I'm not using GSM Library and if i want to that this library does not support the LTE Module I'm using in my project.

mudgalp, 2022-05-20:
but i tried this method using my AT commands and getting issue in this loop.

````
while (Serial 2.available()) { // read file data to spiffs if (!file.print(char(client.read()))) { Serial.println("Appending file"); } //Serial.print((char)c); // Uncomment this to show data //crc.update(c); readLength++; if (readLength % (contentLength / 13) == 0) { printPercent(readLength, contentLength); } timeout = millis(); }
```

here the problem is that this while loop break before all data bytes downloads.

that means the data available on the server will not be downloaded completely and this while loop will break.

SuGlider, 2022-06-06:
There is a possible work around that may fix this issue.

Pease try this code/example when starting Serial2 (used to communicate to the Modem and then receive the OTA firmware):
```
#include "driver/uart.h"  // necessary for the function uart_set_always_rx_timeout()
void setup(){
        Serial2.setRxBufferSize(4096);
        Serial2.begin(115200);
        Serial2.setRxTimeout(1);
        uart_set_always_rx_timeout(2, true); // 2 is for Serial1, in this case
        Serial2.flush();
}
```

Please let me know if this solves the issue.

mudgalp, 2022-07-03:
Good Day all.
There are few updates with the problem.

1-----------------------------------------

> #include "driver/uart.h"
void setup(){
        Serial1.setRxBufferSize(1024);
        Serial1.begin(115200,SERIAL_8N1,26,27);
        Serial1.setRxTimeout(1);
        uart_set_always_rx_timeout(1, true); // 1 is for Serial1, in this case
        Serial1.flush();
}

 dear @SuGlider , I tried this but there is no difference in the problem.

2-------------------------------------------------------------------------------------------
Other thing is this that i discussed this problem with the Quectel's Technical support team and they asked me to try using slower baud rate like for example 9600 and i tried so, and it worked for some part.

now the esp32 takes all the binary file and writes it to the ota partition successfully but it does not gets restart on the newer firmware version instead it gives me a different error as follows.

the function i used
`

> bool DoFirmwareUpdate()
{
  Serial.println("<<<  ATTENTION !!! DO NOT TURN OFF THE DEVICE...  >>>"); Serial.println();
  Serial.println("Updating the Device Firmware Now!"); Serial.println();
  Serial.println("Please Wait untill the Device Restarts"); Serial.println();
  bool value;
  String fwurl = "";
  fwurl = URL_fw_Bin;
  Serial.println(fwurl); Serial.println();
  if (WiFi.status() != WL_CONNECTED)
  {
    Network_Check_GSM();
    if (GSM_Network_STS == true)
    {
      Serial2.write("AT+QIACT?\r\n");
      delay(100);
      input = Serial2.readString();
      if (input.indexOf("OK") >= 0)
      {
        Serial.println("GPRS CONNECT SUCCESSFULLY");
        Serial.println();
        input = Serial2.readString();
        input.remove(0);
        Serial2.print("AT+QHTTPURL=79,30\r\n");
        delay(100);
        input = Serial2.readString();
        if (input.indexOf("CONNECT") >= 0)
        {
          Serial2.print(fwurl);
          delay(100);
          input = Serial2.readString();
          if (input.indexOf("OK") >= 0)
          {
            Serial.println("URL CONNECT SUCCESSFULLY");
            Serial.println();
            input = Serial2.readString();
            input.remove(0);
            int count = 0;
            Serial2.write("AT+QHTTPGET=30\r\n");
            input = Serial2.readString();
            while (input.indexOf("+QHTTPGET") < 0)
            {
              Serial.println("WAITING FOR GET COMMAND RESPONSE");
              Serial.println();
              if (count >= 5)
              {
[... 318 lines cut ...]
[6-28 19:37:01.6]Got application/octet-stream payload.
[6-28 19:37:01.6]line:Content-Length: 213296
[6-28 19:37:01.6]Got 213296 bytes from server
[6-28 19:37:01.6]line:Last-Modified: Fri, 20 May 2022 08:03:51 GMT
[6-28 19:37:01.6]line:Connection: keep-alive
[6-28 19:37:01.7]line:ETag: "62874b67-34130"
[6-28 19:37:01.7]line:Accept-Ranges: bytes
[6-28 19:37:01.7]line:
[6-28 19:37:01.7]This Line Was empty one times
[6-28 19:37:01.7]httpResponseLineCount is :1
[6-28 19:37:01.7]This Line Was empty two times
[6-28 19:37:01.7]Breaking While() Now
[6-28 19:37:01.7]contentLength : 213296, isValidContentType : 1
[6-28 19:37:01.7][ 78868][D][Updater.cpp:133] begin(): OTA Partition: app1
[6-28 19:37:01.7]Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!
[6-28 19:40:43.9]Written : 213296 successfully
[6-28 19:40:43.9]E (301042) esp_image: invalid segment length 0xf4400827
[6-28 19:40:43.9]Error Occurred. Error #: 9
[6-28 19:40:43.9]Could Not Activate The Firmware
[6-28 19:40:50.1]<----------BOOT PROCESS COMPLETE---------->
[6-28 19:40:50.1]

dear @SuGlider  please check the progress and help me get the solution.

Thank you

SuGlider, 2022-09-19:
We have done a few updates to UART in the new Arduino Core 2.0.5 version.
There are new functions that may help your OTA application:

- `void onReceive(OnReceiveCb function, bool onlyOnTimeout = false);`
    `onReceive()` will setup a callback that will be called whenever an UART interruption occurs (UART_INTR_RXFIFO_FULL or 
    UART_INTR_RXFIFO_TOUT). This means that it is possible to set up a callback that will be executed as soon as data is 
    received.
    UART_INTR_RXFIFO_FULL interrupt triggers at UART_FULL_THRESH_DEFAULT bytes received (defined as 120 bytes by 
    default in IDF). This default value of 120 can be changed using `setRxFIFOFull()` with a value from 1 to 127.
    UART_INTR_RXFIFO_TOUT interrupt triggers at UART_TOUT_THRESH_DEFAULT symbols passed without any reception 
    (defined as 2 symbols by default). This can also be changed with `setRxTimeout()`
    
   `onlyOnTimeout` parameter will define how onReceive will behave:
                 `true` -- The callback will only be called when RX Timeout happens. 
                              Whole stream of bytes will be ready for being read on the callback function at once.
                              This option may lead to Rx Overflow depending on the Rx Buffer Size and number of bytes received
                              in the streaming
   **_Default:_** `false` -- The callback will be called when FIFO reaches `RXFIFO_FULL` bytes and also on RX Timeout.
                              The stream of incommig bytes will be "split" into blocks of minimum `RXFIFO_FULL` bytes on each callback.
                              This option avoids any sort of Rx Overflow, but leaves the UART packet reassembling work 
                              to the Application.

- `void onReceiveError(OnReceiveErrorCb function);`
   `onReceiveError()` will be called on error events (see hardwareSerial_error_t) and it will also call the `onReceive()` callback in 
   case some was defined. By this, it is possible to read data using `onReceive()` callback when a BREAK is sent to UART, for 
   instance.

-  `void setRxTimeout(uint8_t symbols_timeout);`
    `setRxTimeout()` sets the timeout after which onReceive callback will be called (after receiving data, it waits for this time of UART rx inactivity to call the callback fnc)
    param symbols_timeout defines a timeout threshold in uart symbol periods. 
    Setting 0 (zero) symbol timeout disables the callback call by timeout.
    Maximum timeout setting is calculacted automatically by IDF. If set above the maximum, it is ignored and an error is printed on Serial0 (check console).
    For example symbols_timeout=1 defines a timeout equal to transmission time of one symbol (~11 bit) on current baudrate. 
    For a baudrate of 9600, SERIAL_8N1 (10 bit symbol) and symbols_timeout = 3, the timeout would be 3 / (9600 / 10) = 3.125 ms
    
-  `void setRxFIFOFull(uint8_t fifoBytes);`
    `setRxFIFOFull()` will set the number of bytes that will trigger UART_INTR_RXFIFO_FULL interrupt and fill up RxRingBuffer
    This affects some functions such as Serial::available() and Serial.read() because, in a UART flow of receiving data, 
    Serial internal RxRingBuffer will be filled only after these number of bytes arrive or a RX Timeout happens.
    This parameter can be set to  a low value, such as 1 in order to receive byte by byte, but it will also consume more 
    CPU time as the ISR will be activates often.

SuGlider, 2022-09-19:
Using Arduino Core 2.0.5, please try to setup the Serial used for OTA/LTE like this:

``` cpp
void setup(){
        Serial.setRxBufferSize(1024);
        Serial.begin(115200,SERIAL_8N1,26,27);
        Serial.setRxTimeout(1);
        Serial.setRxFIFOFull(1);
        Serial.flush();
}
```

This will force that whenever a single byte arrives to the Serial Port, it will be made immediately available for `Serial.read()`.

SuGlider, 2022-11-20:
After some testing with Arduino Core 2.0.5, the best configuration shall be:

change is on  `Serial.setRxFIFOFull(5);` - it  will produce a better result with UART.

``` cpp
void setup(){
        Serial.setRxBufferSize(1024);
        Serial.begin(115200,SERIAL_8N1,26,27);
        Serial.setRxTimeout(1);
        Serial.setRxFIFOFull(5);
        Serial.flush();
}

```

primus192, 2022-12-01:
I tested the code above. Maybe i can record a video showing difference in updating on arduino 1.0.6 vs 2.0.5 so we can investigate and understand the problem better.

SuGlider, 2022-12-01:
After verifying the differences from UART v1.0.6 to 2.0.x, I see that because 2.0.x uses IDF UART driver, it is slower.
It takes almost the double of the time that 1.0.6 takes to read a single byte.

Unfortunatelly, we can't move away from IDF. 
In order to mitigate part of this issue, I have submitted a PR to make reading more than 1 byte from UART faster, using HardwareSerial::read(buffer, size). 
But this PR doesn't help on reading a single byte, and as result, it also doesn't help `Stream::readBytes()` as well.

The issue with Update Library and Arduino Core 2.0.x is that Stream API is slower now... 
That may lead to timeout issues such as in the initial example of this post.

```
/*
      Writes the remaining bytes from the Stream to the flash
      Uses readBytes() and sets UPDATE_ERROR_STREAM on timeout
      Returns the bytes written
      Should be equal to the remaining bytes when called
      Usable for slow streams like Serial
*/
    size_t writeStream(Stream &data);
```

`readBytes(buffer, size)` shall be replaced by `HardwareSerial::read(buffer, size)` in an alternative version of Update.cpp.
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Updater.cpp#L371

I don't know how possible would be for you guys to make such code change and test it...

primus192, 2022-12-01:
I am goin

> After verifying the differences from UART v1.0.6 to 2.0.x, I see that because 2.0.x uses IDF UART driver, it is slower. It takes almost the double of the time that 1.0.6 takes to read a single [...]

That is a pity. I am going to test that

SuGlider, 2022-12-04:
@mudgalp @primus192 

Please try using lastest master branch from Arduino ESP32 Github to build you application.
There is a new PR that may fix the issue: https://github.com/espressif/arduino-esp32/pull/7525

This PR fixes `Stream::readBytes()` to make it faster using IDF in the new Core 2.0.x

I think that it may solve the timeout issue.

Please use this setup:
```
void setup(){
        Serial.setRxBufferSize(2048);
        Serial.begin(115200,SERIAL_8N1,26,27);  // RxPin = 26 & TxPin = 27
        Serial.setRxTimeout(1);
        Serial.setRxFIFOFull(120);
        Serial.flush();
}
```

closed 2022-12-13 by me-no-dev

Sanjeet-Dhankhar, 2023-01-04:
Dear @mudgalp,

Please provide me the files of the project. I don't need GSM library which you are using for this project rest library kindly share.

Thanks