espressif/arduino-esp32 #6647: Last version arduino-esp32 not support esp32 board, build error

Yeharold, 2022-04-27:
### Board

esp32dev

### Device Description

esp32dev

### Hardware Configuration

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
upload_port = COM3
upload_speed = 921600
monitor_speed = 115200

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

win10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Last version arduino-esp32 build error when set my board as  esp32dev ( c3 s2 s3  chips are fine ) 

### Sketch

```cpp
every examples
```

### Debug Message

```plain
In file included from C:/Users/harold/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/adc.h:14:0,
                 from C:/Users/harold/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-adc.c:20:
C:/Users/harold/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:27: error: expected ',' or '}' before '__attribute__'
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
                           ^
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
*** [.pio\build\esp32dev\FrameworkArduino\esp32-hal-adc.c.o] Error 1
C:/Users/harold/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp: In function 'void setToneChannel(uint8_t)':
C:/Users/harold/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:99:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
*** [.pio\build\esp32dev\FrameworkArduino\Tone.cpp.o] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-27:
Maybe @Jason2866 can help you with this issue... 
I know he has made Tasmota work fine with PIO and the latest Arduino Core version.

Jason2866, 2022-04-27:
Give this a try. Before delete in hidden folder `.platformio` the folder `platforms` and `packages`
```
platform = espressif32 @ 4.1.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board = esp32dev
```

SuGlider, 2022-04-27:
I also found tihs comment regarding a different issue, but related to PlatformIO. 
I think it can also help you:
https://github.com/espressif/arduino-esp32/issues/6632#issuecomment-1111016155

Yeharold, 2022-04-28:
> Maybe @Jason2866 can help you with this issue... I know he has made Tasmota work fine with PIO and the latest Arduino Core version.

thank u ，Jason2866's  fork  help me with this issue...

Yeharold, 2022-04-28:
> Give this a try. Before delete in hidden folder `.platformio` the folder `platforms` and `packages`
> 
> ```
> platform = espressif32 @ 4.1.0
> platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
> board = esp32dev
> ```

thank u so much, everything work fine

VojtechBartoska, 2022-04-28:
Closing as solved, thanks for help with answering that issue.

closed 2022-04-28 by VojtechBartoska

SuGlider, 2022-05-02:
New PIO version released with ESP Arduino Core 2.0.2
https://piolabs.com/blog/news/platformio-oss-april-2022-updates.html

https://github.com/espressif/arduino-esp32/issues/6674#issuecomment-1114753254