espressif/arduino-esp32 #6648: FreeRTOS stream buffer missing

aheinous, 2022-04-27:
### Board

ESP32 Dev module

### Device Description

Select Espressif ESP32 Dev Module when creating a new project. 

### Hardware Configuration

NA

### Version

v2.0.2

### IDE Name

VSCode - PlatformIO

### Operating System

macOS 12.1

### Flash frequency

NA

### PSRAM enabled

no

### Upload speed

NA

### Description

StreamBufferHandle_t etc. are no where to be found despite appearing in the [Espressif documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#stream-buffer-api)

### Sketch

```cpp
#include <Arduino.h>

#include "FreeRTOS.h"

// these includes can't be found
// #include "stream_buffer.h"
// #include "freertos/stream_buffer.h"

StreamBufferHandle_t x;

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
src/main.cpp:5:1: error: 'StreamBufferHandle_t' does not name a type
 StreamBufferHandle_t x;
```

### Other Steps to Reproduce

Start a new PlatformIO project. 
Choose Espressif ESP32 Dev Module, and Arduino Framework.
Code above won't compile.

Note: stream_buffer.h exists at ```.platformio/packages/framework-espidf/components/freertos/include/freertos/stream_buffer.h```

However ```#include FreeRTOS.h``` maps to something from a BLE library ```.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/FreeRTOS.h```
and ```#include "freertos/FreeRTOS.h"``` maps to ```.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/FreeRTOS.h```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-27:
It is best to use double quote includes for your local headers, and use gt/lt for system libraries.  This will make things clearer as to where they should be located.  This compiles for me (in master):
```
#include <freertos/stream_buffer.h>
void setup() {
  Serial.begin(115200);
  Serial.println(sizeof(StreamBufferHandle_t));
}
```

aheinous, 2022-04-28:
You're example didn't work for me with my original fairly up-to-date environment. But it did after doing "Update All." There was a lot of new stuff though, which was surprising. It looks like the update broke both styles of FreeRTOS includes in another project ... but I guess that's a different issue.

closed 2022-04-28 by aheinous