espressif/arduino-esp32 #6650: How do I set BLE_uart with 2M phy?

tatsutaigu, 2022-04-28:
### Board

ESP32-C3-DevKitM-1

### Device Description

ESP32-C3-DevKitM-1

### Hardware Configuration

ESP32-C3-DevKitM-1 only

### Version

v2.0.1

### IDE Name

Arduino IDE 1.8.19

### Operating System

windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I would like to use 2M phy on ESP32-C3.
I tried to combine BLE_uart and BLE5_periodic_advertising.
I could find "ESP_MULTI_ADV_2M", but cannot connect to "ESP_MULTI_ADV_2M".

I use LightBlue on iPhone for BLE debugging.

### Sketch

```cpp
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <BLEAdvertising.h>

esp_ble_gap_ext_adv_params_t ext_adv_params_2M = {
    .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_NONCONN_NONSCANNABLE_UNDIRECTED,
    .interval_min = 0x40,
    .interval_max = 0x40,
    .channel_map = ADV_CHNL_ALL,
    .own_addr_type = BLE_ADDR_TYPE_RANDOM,
    .filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy = ESP_BLE_GAP_PHY_1M,
    .max_skip = 0,
    .secondary_phy = ESP_BLE_GAP_PHY_2M,
    .sid = 1,
    .scan_req_notif = false,
};

static uint8_t raw_scan_rsp_data_2m[] = {
        0x02, 0x01, 0x06,
        0x02, 0x0a, 0xeb,
        0x12, 0x09, 'E', 'S', 'P', '_', 'M', 'U', 'L', 'T', 'I', '_', 'A',
        'D', 'V', '_', '2', 'M', 0X0
};

static esp_ble_gap_periodic_adv_params_t periodic_adv_params = {
    .interval_min = 0x320, // 1000 ms interval
    .interval_max = 0x640,
    .properties = 0, // Do not include TX power
};

static uint8_t periodic_adv_raw_data[] = {
        0x02, 0x01, 0x06,
        0x02, 0x0a, 0xeb,
        0x03, 0x03, 0xab, 0xcd,
        0x11, 0x09, 'E', 'S', 'P', '_', 'P', 'E', 'R', 'I', 'O', 'D', 'I',
        'C', '_', 'A', 'D', 'V'
};

uint8_t addr_2m[6] = {0xc0, 0xde, 0x52, 0x00, 0x00, 0x02};

BLEMultiAdvertising advert(1); // max number of advertisement data 

BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("UART Service");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
										CHARACTERISTIC_UUID_TX,
										BLECharacteristic::PROPERTY_NOTIFY
									);
                      
  pTxCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
											 CHARACTERISTIC_UUID_RX,
											BLECharacteristic::PROPERTY_WRITE
										);

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  //pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
  
  advert.setAdvertisingParams(0, &ext_adv_params_2M);
  advert.setAdvertisingData(0, sizeof(raw_scan_rsp_data_2m), &raw_scan_rsp_data_2m[0]);
  advert.setInstanceAddress(0, addr_2m);
  advert.setDuration(0, 0, 0);

  delay(100);
  advert.start();
  advert.setPeriodicAdvertisingParams(0, &periodic_adv_params);
  advert.setPeriodicAdvertisingData(0, sizeof(periodic_adv_raw_data), &periodic_adv_raw_data[0]);
  advert.startPeriodicAdvertising(0);
  
}

void loop() {

    if (deviceConnected) {
        pTxCharacteristic->setValue(&txValue, 1);
        pTxCharacteristic->notify();
        txValue++;
		delay(10); // bluetooth stack will go into congestion, if too many packets are sent
	}

    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
		// do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}
```

### Debug Message

```plain
LightBlue on iPhone shows below message

Connection Alert
Timeout interrogating the peripheral
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-28:
@tatsutaigu 

Why is it set this way?
``` cpp
esp_ble_gap_ext_adv_params_t ext_adv_params_2M = {
    .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_NONCONN_NONSCANNABLE_UNDIRECTED,
```

Try this:
``` cpp
    .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_SCANNABLE,
```

tatsutaigu, 2022-05-11:
@SuGlider 
I'm sorry for delaying and thank you for your reply.

I tried new code.
However, I got below error.

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x484
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2398
SHA-256 comparison failed:
Calculated: e0c357829f601add6a50d6ea19f89687ec5acf531610506a38b33d09b4a71c7e
Expected: 77332826ec7bc491df854452f675071e3a094e80858dd7d91dd7e27940204216
Attempting to boot anyway...
entry 0x403ce000
Waiting a client connection to notify...
E (483) BT_BTM: LE EA SetAdvData: cmd err=0x12
E (584) BT_BTM: LE EA En=1: cmd err=0xc
E (584) BT_BTM: BTM_BlePeriodicAdvSetParams, instance = 0, Before set the periodic adv parameters, please configure the the                 extend adv to nonscannable and nonconnectable first, and it shouldn't include the legacy bit.
E (598) BT_BTM: LE PA SetData: cmd err=0xc
E (601) BT_HCI: CC evt: op=0x203f, status=0xc
E (605) BT_BTM: LE PA En=1: cmd err=0x42
E (609) BT_HCI: CC evt: op=0x2040, status=0x42
```

SuGlider, 2022-05-15:
@lbernstone - Do you have any hint or can you help on it?

lbernstone, 2022-05-16:
Nope, I know nothing about bluetooth.  I know @chegewara has gotten 2M to work, but probably with IDF or NimBLE.

chegewara, 2022-05-16:
This is self explanatory:
```
E (584) BT_BTM: BTM_BlePeriodicAdvSetParams, instance = 0, Before set the periodic adv parameters, please configure the the                 extend adv to nonscannable and nonconnectable first, and it shouldn't include the legacy bit.
```
Periodic advertising is `nonscannable and nonconnectable`, but dont ask me why. I didnt work too much with 2M.
You should find other options for 2M, but this is not most important, because you can use legacy advertising and set prefered PHY when connected. Again, i know it shouyld be possible, but i dont remember which function to use, and if i implemented it in arduino.

tatsutaigu, 2022-05-16:
Hi,
Thank you for your reply.
I would like to use BLE_uart with 2M ( or coded ).
However, BLE5 examples are only advertising.
Please add BLE5 examples to not only advertising but also other function (BLE_uart and so on).
for example, BLE_uart with 2M or coded.

chegewara, 2022-05-17:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_ble.html#_CPPv436esp_ble_gap_set_prefered_default_phy22esp_ble_gap_phy_mask_t22esp_ble_gap_phy_mask_t

VojtechBartoska, 2022-06-14:
Can I consider this as covered? @tatsutaigu

tatsutaigu, 2022-06-15:
@VojtechBartoska 
I tried it, but I cannot set BLE_uart with 2M ( or coded ) yet.
I don't know coding about it.

VojtechBartoska, 2022-12-15:
Hello, I'm closing this as expired to keep reasonable amount of tasks in issue tracker. Unfortunately there is not much to be recommended.

closed 2022-12-15 by VojtechBartoska

chegewara, 2023-02-28:
I am building app with nimble library now, but just in case i am adding a tip here one more time.
This function after `BLEDevice::init()`
```
esp_err_t esp_ble_gap_set_preferred_default_phy(esp_ble_gap_phy_mask_t  tx_phy_mask, esp_ble_gap_phy_mask_t rx_phy_mask)
```
or this function in `onConnect()` callback:
```
esp_err_t esp_ble_gap_set_preferred_phy(esp_bd_addr_t bd_addr, esp_ble_gap_all_phys_t all_phys_mask, esp_ble_gap_phy_mask_t tx_phy_mask, esp_ble_gap_phy_mask_t rx_phy_mask, esp_ble_gap_prefer_phy_options_t phy_options)
```
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/bluetooth/esp_gap_ble.html#_CPPv429esp_ble_gap_set_preferred_phy13esp_bd_addr_t22esp_ble_gap_all_phys_t22esp_ble_gap_phy_mask_t22esp_ble_gap_phy_mask_t32esp_ble_gap_prefer_phy_options_t

In arduino those functions may have different name, because has been renamed.

[image]