espressif/arduino-esp32 #6652: My cheap SD cards no longer mount

brunojoyal, 2022-04-28:
### Board

Custom PCB

### Device Description

Custom PCB with micro-SD card reader. 

### Hardware Configuration

MicroSD adapter is connected on VSPI on pins 18,19,23,5.

### Version

v2.0.2

### IDE Name

PlatformIO

### Operating System

Linux Mint

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have been using a handful of cheap 256mb SD cards for months with this custom PCB with no problems whatsoever. However, after updating to 2.X, literally none of these cards will mount on any of my boards:

`21:12:30.697 > [  6215][V][Setup.h:77] setup(): Attempting to mount SD card.
21:12:30.999 > [  6516][W][sd_diskio.cpp:104] sdWait(): Wait Failed
21:12:30.999 > [  6516][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
21:12:30.999 > [  6516][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
21:12:30.999 > [  6516][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
21:12:31.298 > [  6817][W][sd_diskio.cpp:104] sdWait(): Wait Failed
21:12:31.299 > [  6817][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed`

However, I have no problem with a name-brand 8gb microSD card.
All of the cards were properly formatted.

I'm not sure what is happening here, nor how it could have been caused by the update... But all of these cards were working before, and none of them are working now, so I'm quite confident that it's related...

Thanks for any ideas.

### Sketch

```cpp
#include <SD.h>

SPIClass spiSD(VSPI);

void setup()
{

  spiSD.begin(18,19,23,5);

}
```

### Debug Message

```plain
`21:12:30.697 > [  6215][V][Setup.h:77] setup(): Attempting to mount SD card.
21:12:30.999 > [  6516][W][sd_diskio.cpp:104] sdWait(): Wait Failed
21:12:30.999 > [  6516][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
21:12:30.999 > [  6516][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
21:12:30.999 > [  6516][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
21:12:31.298 > [  6817][W][sd_diskio.cpp:104] sdWait(): Wait Failed
21:12:31.299 > [  6817][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brunojoyal, 2022-04-28:
Just verified that the problem seems to be fixed on the master branch, so I think this can be closed.

P-R-O-C-H-Y, 2022-05-02:
Hi @brunojoyal, there was a fix in 2.0.3-RC1 for some sd cards that was not able to initialise from version 2.0.0. 
So if its working for you in master branch, I'm closing this :) 
If you still have this issue, feel free to reopen.

closed 2022-05-02 by P-R-O-C-H-Y