espressif/arduino-esp32 #6653: A fatal error occurred: Invalid head of packet (0x46)

hicham-vv, 2022-04-28:
### Board

esp32 dev

### Device Description

Nothing only some Bottons and leds

### Hardware Configuration

GPIO0:Pulledup to 3.3V via 10K resistor,
GPIO2: Connected to led 
GPIO6---->GPIO11 Connected to Flash Memory 
GPIO12: Not connected

### Version

other

### IDE Name

Platfrom IO

### Operating System

windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

default speed of platform IO

### Description

Hello ESP32 lovers,
I am facing a problem with my esp32 Chip is ESP32-D0WDQ6 (revision 1), I tried to flash it but I didn't succeed.
so if i let it to enter to the mode boot automatically , it stuck in "connecting .......___......  after some seconds it give me a the message error "A fatal error occurred: Timed out waiting for packet header "
but when i try to hold  Boot botton so i can connect boot to GND, it begin flashing but suddenly the operation stops and it give me this error, please any ideas 
thank you 
"
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 551.3 kbit/s)...
Hash of data verified.
Compressed 232192 bytes to 126659...
Writing at 0x00010000... (12 %)
Writing at 0x0001e4e8... (25 %)

A fatal error occurred: Invalid head of packet (0x46)
"

### Sketch

```cpp
#include <Arduino.h>
uint64_t comp=0;

void setup() {
  // put your setup code here, to run once:
  pinMode(2,OUTPUT);
  Serial.begin(115200);
  while(!Serial);
  delay(1000);
  
}

void loop() {
  comp++;
  digitalWrite(2,HIGH);
  delay(500);
  digitalWrite(2,LOW);
  delay(500);
  Serial.println(comp);

}
```

### Debug Message

```plain
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (4.1.0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.20001.0 (2.0.1)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkfatfs 2.0.1
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\az-delivery-devkit-v4\firmware.elf
Checking size .pio\build\az-delivery-devkit-v4\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.1% (used 16444 bytes from 532480 bytes)
Flash: [==        ]  17.7% (used 231805 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM85
Uploading .pio\build\az-delivery-devkit-v4\firmware.bin
esptool.py v3.1
Serial port COM85
Connecting........___
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:f4:ab:6c:61:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00048fff...
Compressed 17216 bytes to 11867...
Writing at 0x00001000... (100 %)
Wrote 17216 bytes (11867 compressed) at 0x00001000 in 0.6 seconds (effective 248.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 420.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 572.9 kbit/s)...
Hash of data verified.
Compressed 232192 bytes to 126659...
Writing at 0x00010000... (12 %)
Writing at 0x0001e4e8... (25 %)
Writing at 0x000242d9... (37 %)
Writing at 0x000299a7... (50 %)
Writing at 0x0002ec31... (62 %)
Writing at 0x00037489... (75 %)
Writing at 0x0003f33e... (87 %)

A fatal error occurred: Timed out waiting for packet header
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-28:
I'm not expert on PlatformIO, but is it possible to change the upload speed to a lower one and try again?

I know that with Arduino IDE there is a menu option for it ("Menu->tools->Upload Speed->921600"). It can be set for the following options: 921600, 115200, 256000, 230400 and 512000.
Maybe, just to verify your ESP chip / UART-USB chip, you can give it a try with the Arduino IDE. 

Other possible problems are related to the USB cable quality, ESP board issues, USB port/connector on the PC,  Power stability.

hicham-vv, 2022-04-28:
hello @SuGlider thank you for your reply  but i tried changing speed with both platformIO and Arduino IDE but still getting the same issues

SuGlider, 2022-04-28:
Maybe it is a hardware/cable issue?

hicham-vv, 2022-04-28:
maybe,:( i have checked all the wires and voltages, nothing doubtful

SuGlider, 2022-04-28:
Something else to try would be to wipe out the flash with Flash Download Tool.

SuGlider, 2022-04-28:
http://iot-bits.com/erase-esp32-flash-completely/

SuGlider, 2022-04-28:
https://www.espressif.com/en/support/download/other-tools

hicham-vv, 2022-04-28:
thank you let me test that

SuGlider, 2022-04-28:
it is also possible with CLI esptool.py
https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#erase-flash-erase-flash-erase-region

hicham-vv, 2022-04-28:
Sir it didn't succeed

hicham-vv, 2022-04-28:
stuck here 
[image]

hicham-vv, 2022-04-28:
Error 
[image]

SuGlider, 2022-04-28:
I see. I think that maybe the flash in the board is gone... 
Do you have another board to test it?

SuGlider, 2022-04-28:
A last resource woud to try different combinations of SpiFlashConfig for SPI SPEED and SPI MODE in the Flash Tool when trying to erase it.

SuGlider, 2022-04-28:
I think that the board with the issue has now a hardware failure in its flash. 
If so, there is nothing we can do about it.

hicham-vv, 2022-04-29:
I'm noticing that if I touch boot button it will boot then flashing memory will start.
sometimes the flashing done and the sketch work perfectly, but in the most of case uploading code stops at the midle of the operation, as you see below, 
I'm wondering if the issue is do with floating pins needing to float to ground and holding the BOOT button simply extends the period the pins are given to reach ground. or the the esp32 communicate badly with the flash memory.

Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:f4:ab:6c:61:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00048fff...
Compressed 17216 bytes to 11867...
Writing at 0x00001000... (100 %)
Wrote 17216 bytes (11867 compressed) at 0x00001000 in 0.6 seconds (effective 244.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 402.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 556.5 kbit/s)...
Hash of data verified.
Compressed 232192 bytes to 126659...
Writing at 0x00010000... (12 %)
Writing at 0x0001e4e8... (25 %)
Writing at 0x000242d9... (37 %)

A fatal error occurred: Invalid head of packet (0x46)
*** [upload] Error 2

podaen, 2022-05-01:
Try to install an other driver

SuGlider, 2022-05-02:
> hello, i tried with another board that have the same esp,driver,flash,memory, and its succeed but with that one who have the problem i still got nothing [image]

If another board works with exactly the same configurations you are using for the board that doesn't work and both boards are the same, I think that it makes it clear that the issue is in the board itself.

I'll close the issue in a couple days in case there are no more news regarding this issue.

podaen, 2022-05-02:
you need to select the files to been upload. Take a look at this basic setting that worked for me.

[image]

If you want to advance it you could do it like this...

[image]

VojtechBartoska, 2022-05-18:
Closing the issue.

closed 2022-05-18 by VojtechBartoska

hicham-vv, 2022-05-24:
i think it's an hardware issue, i just supply the esp from 3.3V to 2.6V, then i can upload the code normally !!!

johntamplin, 2023-02-28:
Slowing the upload speed fixed this for me. 
Tools -> Upload speed

jordanrw, 2024-07-24:
> I'm not expert on PlatformIO, but is it possible to change the upload speed to a lower one and try again?
>
> I know that with Arduino IDE there is a menu option for it ("Menu->tools->Upload [...]

Thank you so much for this suggestion @SuGlider! This worked for me. I was having trouble getting a SparkFun ESP32 Thing Plus C working on this [SparkFun IoT Brushless Motor Driver](https://docs.sparkfun.com/SparkFun_IoT_Brushless_Motor_Driver/) board. 

**I'm running macOS Sonoma (14.3.1) on an Apple M3 Pro.**
Below are the steps I took to get my basic _Hello World_ sketch to upload properly:

- I was following this [Hookup Guide - SparkFun IoT Brushless Motor Driver](https://docs.sparkfun.com/SparkFun_IoT_Brushless_Motor_Driver/software_overview/)
- Installed the **esp32** _by Espressif_ board definitions in the Board Manager 
- Installed the [CH340 Driver](https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers#mac-osx)
- Then updated the [security settings on the startup disk of a Mac with Apple silicon](https://support.apple.com/guide/mac-help/change-security-settings-startup-disk-a-mac-mchl768f7291/mac) in order for the Mac to allow a 3rd party .kext file.

However I was still getting this error when trying to upload a basic sketch:
```
A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xA6): Possible serial noise or corruption.).
Failed uploading: uploading error: exit status 2
```

Upload Speed of 115200, 230400, or 460800 all work but 921600 doesn't.

I'm actually unsure if I even needed to do the CH340 driver installation or if there is already some sort of support on Apple silicon Macs for the CH340 chip. SparkFun didn't have any information specifically about Apple silicon.