espressif/arduino-esp32 #6654: WiFi connection unstable after updating to 2.0.X

ToMe25, 2022-04-28:
### Board

ESP32DevkitCVE/IE

### Device Description

The devkit with an external antenna connector(CVIE) an external antenna is connected, in the case of the one with internal antenna nothing is connected besides a PC using an USB-A to USB-Micro-B cable.

### Hardware Configuration

Nothing connected.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO - Eclipse

### Operating System

Ubuntu 20.04

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

After updating a sketch from 1.0.6 to 2.0.2 I noticed that the WiFi in STA mode doesn't get an IP until it times out and reconnects.
This can already take a few minutes, and is the case even tho the first WiFi method called in that sketch is `WiFi.disconnect(true);`.

After finally connecting WiFi works for a few minutes, before silently stopping(No log output even with `CORE_DEBUG_LEVEL` set to 5).
After not working for a few minutes it causes a disconnect event, on which my sketch would then reconnect.
This actually works, and doesn't take a few minutes.

After some useless debugging I then tried to find a minimal sketch to reproduce this.
My end result was pretty much just a web server, but I will post this sketch below.

I then tried the `SimpleWiFiServer` example. which seems to have the same issues, except it didn't seem to ever reconnect.

Restarting the ESP did not seem to change anything at all.

This might be related to #4980

In the log of my custom sketch below I used a bash script to request the content of the web server every 3 seconds.
As far as I know that is the only thing that interacted with the ESP during the time of the test.
The log doesn't seem to change much even without the script, however I wanted to know when it stops working.
Here the script: 
```bash
while true
do
    curl -s --connect-timeout 5 http://192.168.2.138/index.html > /dev/null
    if [ $? == 0 ]; then
        echo OK
    else
        echo ERROR
    fi
    sleep 3
done
```
[... 241 lines cut ...]
WiFi Disconnected
[438212][V][WiFiGeneric.cpp:286] _arduino_event_cb(): STA Stopped
[438339][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[438340][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[438359][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 3 - STA_STOP
[438359][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[438362][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[438419][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: WLAN-8C1784, BSSID: 84:9c:a6:8c:17:50, Channel: 7, Auth: WPA2_PSK
[438421][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[438448][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got Same IP:192.168.2.138
[438449][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[438452][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.2.138, MASK: 255.255.255.0, GW: 192.168.2.1
IP Address: 192.168.2.138
WiFi ready after 438433ms.
```
</details>
```

### Other Steps to Reproduce

It works correctly with version 1.0.6, if the event names are changed to account for the version change.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-03:
@PilnyTomas, PTAL
Could you please check this issue? Thanks.

me-no-dev, 2022-05-10:
> However even if it is something that only occurs with some routers, considering how quickly someone else reported it on the EMS repo I believe its still worth trying to fix it.

The ESP-IDF WiFi library comes as a blob to us (both Arduino and ESP-IDF). In order to be able to forward the issue to our radio team, we need first to have a proper information of the reason. Given that we can not reproduce it, chances are that they will not be able either (else they would have not caused a regression). Good luck with testing. See if you can change settings on your router and will they also change the connection time.

ToMe25, 2022-05-10:
> See if you can change settings on your router and will they also change the connection time.

Unfortunately it doesn't have that many settings that I could see changing this, thats why I want to try it with different hardware, like a RPi or my phone.
Also the issue isn't the connection time, its more how long it takes for it to realize it doesn't have a connection anymore.

I do also intend to test this with the ESP-IDF eventually, but since I don't have an ESP-IDF workspace set up yet, and will probably take a while to get used to it enough to test this, that might be quite a while in the future.

ToMe25, 2022-05-10:
I'm sorry, but I don't quite get what this means.

podaen, 2022-05-10:
LWIP...  You can't connect to the wifi without it.

podaen, 2022-05-10:
Me ether, we have to find someone with a good knowledge of Linux to solve this.

ToMe25, 2022-05-13:
Sorry for taking so long, but I finally tested this with a hotspot on my phone.
Result is it works as expected.
I will try messing with some router settings later today, and hope to find something interesting.
I also wanted to test this with a RPi 4 as the AP, however I don't have an empty SD card rn, and the image I have has makes some problems when I try to use it as an AP.

ToMe25, 2022-05-13:
I played around a bit with my router settings, not expecting anything to help since there wasn't a setting that seemed related, and I can no longer reproduce this.
I was careful to keep track of all changes I did to be able to undo them, because I wanted to know what caused this.
However even after undoing all the changes i can't seem to reproduce this issue anymore, with the same router and ESP32.

I see a decent chance that this issue will just magically re-appear after waiting for a night, like most of my WiFi issues so far, but atm there is nothing I can think of to reproduce this.

The ESP also seems to be fully connected and ready 130-1600ms after startup, which is faster than I can remember it ever being in the past.

ToMe25, 2022-05-13:
I just noticed I didn't make this clear in my last messages:
I tested the issue before changing the router settings to make sure it isn't just fixed because of a new framework version, and was able to reproduce the issue.
I just can't reproduce it since then.
I will try it again tomorrow after everything except the router was shut down for the night, and after another router restart, to see if the issue re-occurs.

me-no-dev, 2022-05-13:
Is it possible that you have a small pool of addresses in your router's DHCP config and they get filled by the computers and devices you have at home? One way to check this is if you get it to start failing again to restart the router and if it get's "fixed" for some time, that could be one clue.

ToMe25, 2022-05-13:
The router is set to give out dhcp addresses from 192.168.2.100 to 192.168.2.200, and the highest I can remember seeing was 192.168.2.156 or something similar.
And that was months ago, I'm pretty sure before just now it didn't have anything above 142.
Also the device list of the router never contains more than 20 devices, so I don't even know how it got to IPs that high.

Another thing is, it didn't work after a restart, it worked after changing the WiFi band.
AFAIK neither the router nor the DHCP server restarted during that change, but even after changing that back it stayed fixed.

The router can use b/g, b/g/n, or a/n.
Default is b/g/n, changing it to a/n for some reason means n capable devices without a support can't connect anymore.
I first changed it to a/n without knowing that, resulting in a lot of devices, including the ESP, no longer being able to connect.
After that I first changed it to b/g, then b/g/n, which made everything work again.

ToMe25, 2022-05-14:
I just checked, and the issue is back without me changing anything.
I just turned the PC I'm using for ESP programming back on, and thus also the ESP used to test this and everything seems to be back in its old not working state.

SuGlider, 2022-05-14:
It seems to be about the DHCP service from the AP Router that may reassign ESP STA IP to some other STA in the network... 

Try this:

``` cpp
// call back function when IP is Lost by Station -- needs to reconnect to AP
void WiFiStationLostIP(arduino_event_id_t event)
{
  log_d("Enter into LOST IP Event Call Back...");
  while (! WiFi.reconnect()) {
    log_d("Trying to Reconnect due to STA__LOST_IP...");
    delay(500); // wait a bit for the necessary processing
  }
}

void setup() {
...
  // Set a callback function when the event STA_LOST_IP happens
  WiFi.onEvent(WiFiStationLostIP, ARDUINO_EVENT_WIFI_STA_LOST_IP);
...
}
```

ToMe25, 2022-05-14:
When I originally tested things for this issue I had a lot of WiFi callbacks that just print debug info, one of them being LOST_IP.
As far as I can remember nothing useful came of that one, however I wanted to be sure so I tested it with your exact code.

The test.sh file I am using here is the connection check script I posted in the original issue.
After adding your LOST IP callback and reflashing the ESP I tested it again.
This time I ran `./test.sh & pio device monitor` to get a combined log of the connection status and the ESP log.
Here the result: <details><summary>Click to expand</summary>
```
[1] 3671
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ERROR
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[␀␘␂␂␂��m␑um����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
setup start
[    63][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   154][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   152][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Started
[   160][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 2 - STA_START
setup end
[   220][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Connected: SSID: WLAN-8C1784, BSSID: 84:9c:a6:8c:17:50, Channel: 7, Auth: WPA2_PSK
[   222][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   260][V][WiFiGeneric.cpp:314] _arduino_event_cb(): STA Got New IP:192.168.2.136
[   261][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   264][D][WiFiGeneric.cpp:938] _eventCallback(): STA IP: 192.168.2.136, MASK: 255.255.255.0, GW: 192.168.2.1
IP Address: 192.168.2.136
WiFi ready after 246ms.
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
ERROR
[360050][V][WiFiGeneric.cpp:307] _arduino_event_cb(): STA Disconnected: SSID: WLAN-8C1784, BSSID: 84:9c:a6:8c:17:50, Reason: 7
[360050][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[360058][W][WiFiGeneric.cpp:897] _eventCallback(): Reason: 7 - NOT_ASSOCED
WiFi Disconnected
[360077][V][WiFiGeneric.cpp:288] _arduino_event_cb(): STA Stopped
[360205][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[360203][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Started
[360224][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 3 - STA_STOP
[360224][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 0 - WIFI_READY
[360227][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 2 - STA_START
[360270][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Connected: SSID: WLAN-8C1784, BSSID: 84:9c:a6:8c:17:50, Channel: 7, Auth: WPA2_PSK
[360272][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[360304][V][WiFiGeneric.cpp:314] _arduino_event_cb(): STA Got Same IP:192.168.2.136
[360305][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[360308][D][WiFiGeneric.cpp:938] _eventCallback(): STA IP: 192.168.2.136, MASK: 255.255.255.0, GW: 192.168.2.1
IP Address: 192.168.2.136
WiFi ready after 360290ms.
OK
OK
OK
OK
OK

--- exit ---
```
</details>

This time the initial connection works instantly, which it usually doesn't, but it still fails after a while without calling LOST IP and then reconnects and works again.
Note that while I made sure that the LOST IP callback is registered correctly it is never called.

SuGlider, 2022-05-14:
How far is the ESP32 from the AP Router?
Could it be some RF failure due to RF noise and then it gets disconnected?
Or even an unstable ESP32 power supplier that causes a variation on the RSSI or even to the RF directly? 

It seems like the ESP gets "lost in the space" for a while...

SuGlider, 2022-05-14:
Is there some sort of `keep alive` so that the ESP can keep the connection to the AP stable?

Maybe a nice test would be adding a PING call in the sketch loop to check the connection to the AP Router IP address.

ToMe25, 2022-05-14:
> How far is the ESP32 from the AP Router?

Roughly 3-4 meters, though there is a ceiling in between.

> Could it be some RF failure due to RF noise and then it gets disconnected?

No idea honestly, tho i would hope it would notice that sooner.

> Or even an unstable ESP32 power supplier that causes a variation on the RSSI or even to the RF directly?

The power supply used is an USB port of a desktop PC for these tests, which it has to be for me to see the logs.
Last time I checked the RSSIs seemed stable, tho that was with 1.0.6.
I did have the same issue with the ESP powered by a USB charger rated for 2.4A tho, so I don't think it's lack of power.
Also I don't have any WiFi issues with 1.0.6 when powering the ESP the same way using the PC.

> Maybe a nice test would be adding a PING call in the sketch loop to check the connection to the AP Router IP address.

I'll try that tomorrow.

SuGlider, 2022-05-14:
For Ping you may want to use one of these Libraries: 
https://github.com/pbecchi/ESP32_ping
https://github.com/marian-craciunescu/ESP32Ping

This link has an example: https://techoverflow.net/2021/06/19/how-to-ping-gateway-in-esp32/

ToMe25, 2022-05-15:
I tested this now, and it seems like the times when the ping doesn't work and the times when the web server isn't reachable are the same.
It also does seem like both the RSSI and the ping fluctuate, however I don't think its enough to cause a disconnect.

I changed the sketch to this: <details><summary>Click to show</summary>
```cpp
#include <ESPAsyncWebServer.h>
#include <ESP32Ping.h>

extern const char WIFI_SSID[] = "WIFI_SSID";
extern const char WIFI_PASS[] = "WPA_PASSPHRASE";

uint64_t start = 0;
AsyncWebServer server(80);

void onWiFiGotIp(WiFiEvent_t event, WiFiEventInfo_t eventInfo) {
	Serial.print("IP Address: ");
	Serial.println((IPAddress) eventInfo.got_ip.ip_info.ip.addr);

	Serial.printf("WiFi ready after %llums.\n", millis() - start);
}

void onWiFiDisonnected(WiFiEvent_t event) {
	Serial.println("WiFi Disconnected");
	WiFi.disconnect(true);
	delay(200);
	WiFi.begin(WIFI_SSID, WIFI_PASS);
}

void WiFiStationLostIP(WiFiEvent_t event) {
	Serial.println("Enter into LOST IP Event Call Back...");
	while (!WiFi.reconnect()) {
		Serial.println("Trying to Reconnect due to STA__LOST_IP...");
		delay(500);
	}
}

void setupWebServer() {
	server.on("/index.html", HTTP_GET, [] (AsyncWebServerRequest *request) {
		request->send(200, "text/plain", "OK");
	});

	server.onNotFound([] (AsyncWebServerRequest *request) {
		request->send(404, "text/plain", "Not Found");
	});

	server.begin();
}

void setup() {
	start = millis();
	Serial.begin(115200);
	Serial.println("setup start");

	WiFi.disconnect(true);
	WiFi.onEvent(onWiFiGotIp, ARDUINO_EVENT_WIFI_STA_GOT_IP);
	WiFi.onEvent(onWiFiDisonnected, ARDUINO_EVENT_WIFI_STA_DISCONNECTED);
	WiFi.onEvent(WiFiStationLostIP, ARDUINO_EVENT_WIFI_STA_LOST_IP);
	WiFi.begin(WIFI_SSID, WIFI_PASS);

	setupWebServer();
	Serial.println("setup end");
}

void loop() {
	if(Ping.ping(WiFi.gatewayIP(), 1)) {
		Serial.printf("Ping OK(%.3fms)\n", Ping.averageTime());
	} else {
		Serial.println("Ping Error");
	}
	Serial.printf("WiFi RSSI: %hd\n", WiFi.RSSI());
[... 1748 lines cut ...]
[553050][I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes

[553061][D][ping.cpp:222] ping_recv(): 60 bytes from 0.0.0.0: icmp_seq=1 time=7.699 ms

[553062][I][ping.cpp:344] ping_start(): 1 packets transmitted, 1 packets received, 0.0% packet loss

[553067][I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 7.699/7.699/7.699/0.000 ms

Ping OK(7.699ms)
WiFi RSSI: -63
HTTP OK
[556088][I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes

[556102][D][ping.cpp:222] ping_recv(): 60 bytes from 0.0.0.0: icmp_seq=1 time=10.312 ms

[556103][I][ping.cpp:344] ping_start(): 1 packets transmitted, 1 packets received, 0.0% packet loss

[556108][I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 10.312/10.312/10.312/0.000 ms

Ping OK(10.312ms)
WiFi RSSI: -65
HTTP OK

--- exit ---
```

SuGlider, 2022-05-15:
I think that ESP32 considers that it has being with no connection to the AP Router for time enough to consider it has disconnected.
There may be a way to set a higher timeout or something like that in the ESP32 WiFi in order to avoid it from disconnecting...

But there is a question.... if you run this same example with 1.0.6, evething works perfectly well?

ToMe25, 2022-05-15:
I tested that a few times when I initially made this issue, and I tested it again just now.
With only one small change that sketch works flawlessly in 1.0.6.
That change being the `ARDUINO_EVENT_WIFI` instances have to be replaced with `SYSTEM_EVENT`.
It does take 2.2 to 2.5 seconds to connect to WiFi, which I know is kinda a lot, but its consistent and doesn't stop working as far as I can tell.

ToMe25, 2022-05-15:
Oh btw I wont be able to test the version 2.0.X things for a while, because my PC died today.
I can test version 1.0.6 using a RPi, but version 2.0.X needs `espressif/toolchain-xtensa-esp32` version `8.4.0+2021r2-patch3` which isn't available for aarch64.

SuGlider, 2022-05-15:
The ping results are the same with 1.0.6?

ToMe25, 2022-05-15:
I can get you a full combined log tomorrow, I assumed that wasn't relevant since its fully working.
I only looked at the first few pings to see if they work at all, which they did, I can't remember the times tho.
I looked at the output of the HTTP test script for at least 15mins, and it worked the whole time.
That's why I thought the pings didn't matter.
I also remember from my initial testing that the 1.0.6 version didn't reconnect every few minutes, but I didn't check that today.
The combined log should show that tho.

SuGlider, 2022-05-15:
That 1.0.6 log would help in comparing both version results.
Thanks.

ToMe25, 2022-05-16:
I just ran the test again with 1.0.6.
I just ran `bash -c "./test.sh 192.168.2.136 & pio device monitor" | tee esp.log` and this is the resulting log:<details><summary>Click to Show</summary>

```
HTTP OK
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
[I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes

[D][ping.cpp:222] ping_recv(): 60 bytes from 215.19.14.128: icmp_seq=1 time=12.752 ms

[I][ping.cpp:344] ping_start(): 1 packets transmitted, 1 packets received, 0.0% packet loss

[I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 12.752/12.752/12.752/0.000 ms

Ping OK(12.752ms)
WiFi RSSI: -59
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
setup start
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
setup end
[I][ping.cpp:324] ping_start(): PING 0.0.0.0: 32 data bytes

[D][ping.cpp:233] ping_recv(): Request timeout for icmp_seq 1

[I][ping.cpp:344] ping_start(): 1 packets transmitted, 0 packets received, 100.0% packet loss

[I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 1000000000.000/0.000/0.000/nan ms

WiFi RSSI: 0
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.2.136, MASK: 255.255.255.0, GW: 192.168.2.1
.168.2.136
WiFi ready after 2287ms.
HTTP OK
[I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes

[D][ping.cpp:222] ping_recv(): 60 bytes from 215.19.14.128: icmp_seq=1 time=10.415 ms

[I][ping.cpp:344] ping_start(): 1 packets transmitted, 1 packets received, 0.0% packet loss

[I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 10.415/10.415/10.415/0.000 ms

Ping OK(10.415ms)
WiFi RSSI: -59
HTTP OK
[I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes

[D][ping.cpp:222] ping_recv(): 60 bytes from 215.19.14.128: icmp_seq=1 time=8.438 ms

[I][ping.cpp:344] ping_start(): 1 packets transmitted, 1 packets received, 0.0% packet loss

[I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 8.438/8.438/8.438/0.000 ms

Ping OK(8.438ms)
WiFi RSSI: -66
HTTP OK
[I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes
[... 3390 lines cut ...]

[I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 6.279/6.279/6.279/0.000 ms

Ping OK(6.279ms)
WiFi RSSI: -60
HTTP OK
[I][ping.cpp:324] ping_start(): PING 192.168.2.1: 32 data bytes

[D][ping.cpp:222] ping_recv(): 60 bytes from 215.19.14.128: icmp_seq=1 time=8.222 ms

[I][ping.cpp:344] ping_start(): 1 packets transmitted, 1 packets received, 0.0% packet loss

[I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 8.222/8.222/8.222/0.000 ms

Ping OK(8.222ms)
WiFi RSSI: -60
HTTP OK

--- exit ---
HTTP OK
```
</details>

While the ping still fluctuates it seems like the RSSI is consitently higher.
The ESP was within a few centimeters of its position from the 2.0.X tests tho, so I can't imagine why the RSSI is significantly higher.

podaen, 2022-05-16:
The RSSI questionable... When I did some tests I found out that when I was close I had a higher RSSI, than I was far from the transmitter (ps: I only print it at bootup). And a longer antenna didn't improve the RSSI. I installed an app on my phone to scan RSSI arround me and the fluctuate too. 

I m curious... can you set a delay between the pings you send from the server, big engough to see if it has infuence on that. Let say 2second between each ping. I'm curious if it stablizes.

ToMe25, 2022-05-16:
> The RSSI questionable... When I did some tests I found out that when I was close I had a higher RSSI, than I was far from the transmitter

The RSSI is signal strength, its supposed to be higher when you're closer.

> And a longer antenna didn't improve the RSSI

A lot more things than just the antenna length matter for the RSSI.
I did two tests with the same antenna: 
1. Using it without an antenna stand directly connected to an adapter and then the ESP.
    This way the RSSI was higher(better) than with an ESP with a PCB antenna.
2. Using the same ESP and antenna with an antenna stand in between.
    This way the RSSI roughly matched that of an ESP with PCB antenna.

> I installed an app on my phone to scan RSSI arround me and the fluctuate too.

Well, yeah, RF stuff always fluctuates a bit due to environmental factors.
The only way your RSSI is stable is for you to place your device directly next to the AP.

> I m curious... can you set a delay between the pings you send from the server, big engough to see if it has infuence on that. Let say 2second between each ping. I'm curious if it stablizes.

What? Currently my sketch pings once every 3 seconds.
The 1.0.6 log was roughly 15 minutes of run time.

SuGlider, 2022-05-16:
>I just ran the test again with 1.0.6.
>I just ran bash -c "./test.sh 192.168.2.136 & pio device monitor" | tee esp.log and this is the resulting log:

Thanks @ToMe25 for the 1.0.6 log. It is really very different from the 2.0.0+ log in terms of ping success.

>While the ping still fluctuates it seems like the RSSI is consitently higher.
>The ESP was within a few centimeters of its position from the 2.0.X tests tho, so I can't imagine why the RSSI is significantly higher.

Yes, I agree that this RSSI difference is something that may explain the 2.0.0+ issue with connection stability, but really not conclusive.
I think that something on the IDF layer of the WiFi radio may have changed from IDF 3.3 (used in 1.0.6) to IDF 4.4 (used in 2.0.0+).
It needs further investigation, here internally.

lubasi01, 2022-11-19:
Having the same behaviour on my side - after upgrading to 2.0.5 Wifi is completely unstable.
My access point is a Unifi-UAP-AC-lite. When the ESP32 is connecting the AP drops all other connected devices (other ESPs, Firestick, Rasips) too.
When using a TP-Link RE200 as access point everything is working fine.

drmpf, 2022-11-23:
I have the same issue on ESP32 (sparkfun ESP32 Thing)
ESP32 V2.0.2 mostly works, higher versions have problems up to and including V2.0.5
ESP32C3 on V2.0.5 works fine as does ESP8266 V3.0.2 boards

Testing with Window10 running Bluestack5 Android emulator running pfodApp

Power cycle the sparkfun ESP32 Thing, 
connect, 
connection made, but outgoing packet dropped
connection times out on the pfodApp side and closed
pfodApp automatically re-connects

Second and third connections usually send more outgoing packets before dropping one.

After about 3 re-connections the connection appears to be stable until the ESP32 is power cycled  again.

Attached is the wireshark capture (zipped because github won't upload .pcapng)
[sparkfunESP32.zip](https://github.com/espressif/arduino-esp32/files/10079772/sparkfunESP32.zip)

Also attached is the Serial monitor output of the received / sent msgs
[sparkfunESP32_monitor.txt](https://github.com/espressif/arduino-esp32/files/10079807/sparkfunESP32_monitor.txt)
The connection timeout on the pfodApp side is 15sec.

The ESP32 send msgs are buffered and the written using
       client->write((const uint8_t *)sendBuffer, sendBufferIdx);

podaen, 2022-11-24:
I having troubles too when my shetch gets bigger. A special when more small packages are send and received after eachother. I have two clients in that shetch one with a high volume of data (receiving) and the other one that is sending and receiving small packages. It's hard to tell what the exact problem is. But if I put the Verbose log on I always get ASSOC_LEAVE. I think we need to make a simple shetch that is sending and receiving between two esps and where the load and the time between can be adjusted and one client can be added afterwards. The problem is I don't have time for a couple of weeks.

drmpf, 2022-11-25:
> Edit 2 -- Rolled back to ESP32 V2.0.0 works fine on that version

I had problems with ESP32 connections using V2.0.0 so have rolled back to V1.0.6 now

podaen, 2022-11-25:
That's OK if you don't use tsl.

drmpf, 2022-11-25:
> That's OK if you don't use tsl.

[pfodApp](https://www.forward.com.au/pfod/index.html) uses a secure hash appended to each message to provide wifi security against modification (not obscurity) so not a problem going back to V1.0.6 for pfodApp connections