espressif/arduino-esp32 #6655: Wifi Provisioning in ESP32-C3 fails in IOS

arihantdaga, 2022-04-28:
### Board

ESP32-C3 Dev Module

### Device Description

ESP-c3-12f 2MB Flash from AI thinker - based on esp32-c3

### Hardware Configuration

N/A

### Version

v2.0.2

### IDE Name

Platformio

### Operating System

Windows

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am using example code from wifi provisioning library. 
I am setting #define CONFIG_IDF_TARGET_ESP32 1 in my .ino 
And trying to provision using BLE. 

I am dumping the same code, once in esp32 dev kit and then in esp32-c3 dev module. 
In ESP32 Dev kit, the provisioning is successful without any errors. 
But in ESP32-C3 module, it prints this error in serial on connecting from esp ble provisioning app(both android and IOS). 
```
E (7299) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
```
From Android app, despite this error, i am able to provision the device. 
But in the IOS it asks for the pairing and then it fails. I am not able to provision. 

The same setup in esp32 devkit works perfectly fine. 

### Sketch

```cpp
#include <esp_wifi.h>
#include "WiFiProv.h"
#include "WiFi.h"
#define CONFIG_IDF_TARGET_ESP32 1
// #define CONFIG_BLUEDROID_ENABLED 1
void SysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        Serial.print("\nConnected IP address : ");
        Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
        break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        Serial.println("\nDisconnected. Connecting to the AP again... ");
        break;
    case ARDUINO_EVENT_PROV_START:
        Serial.println("\nProvisioning started\nGive Credentials of your access point using \" Android app \"");
        break;
    case ARDUINO_EVENT_PROV_CRED_RECV: { 
        Serial.println("\nReceived Wi-Fi credentials");
        Serial.print("\tSSID : ");
        Serial.println((const char *) sys_event->event_info.prov_cred_recv.ssid);
        Serial.print("\tPassword : ");
        Serial.println((char const *) sys_event->event_info.prov_cred_recv.password);
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL: { 
        Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
        if(sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR) 
            Serial.println("\nWi-Fi AP password incorrect");
        else
            Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");        
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
        Serial.println("\nProvisioning Successful");
        break;
    case ARDUINO_EVENT_PROV_END:
        Serial.println("\nProvisioning Ends");
        break;
    default:
        break;
    }
}

void setup() {
  WiFi.mode(WIFI_STA);
     esp_wifi_restore();
  Serial.begin(115200);
  //Sample uuid that user can pass during provisioning using BLE
  /* uint8_t uuid[16] = {0xb4, 0xdf, 0x5a, 0x1c, 0x3f, 0x6b, 0xf4, 0xbf,
                   0xea, 0x4a, 0x82, 0x03, 0x04, 0x90, 0x1a, 0x02 };*/
  WiFi.onEvent(SysProvEvent);
#if CONFIG_IDF_TARGET_ESP32 && CONFIG_BLUEDROID_ENABLED
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, "abcd1234", "Prov_123");
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, "abcd1234", "Prov_123");
#endif
}

void loop() {
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x0 (USB_BOOT)
wait usb download
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x484
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2398
entry 0x403ce000
[   220][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   259][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[   260][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
[   262][V][WiFiGeneric.cpp:275] _arduino_event_cb(): STA Stopped
[   267][D][WiFiGen⸮⸮[   280][V][WiFiGeneric.cpp:429] _arduino_event_cb(): Provisioning Initialized!
[   281][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 33 - PROV_INIT
[   284][I][WiFiProv.cpp:142] beginProvision(): Starting AP using BLE. service_name : Prov_123, pop : abcd1234
[   295][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[   300][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
[   325][V][WiFiGeneric.cpp:435] _arduino_event_cb(): Provisioning Start!
[   325][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 35 - PROV_START

Provisioning started
Give Credentials of your access point using " Android app "
E (13580) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
E (16130) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

arihantdaga, 2022-04-28:
Checked with the latest git. 
I am able to provision the esp32-c3 device via ios also. However, on serial we are still getting the same message during the process.

rotematari, 2024-06-17:
i get the same error

closed 2025-01-16 by Parsaabasi