espressif/arduino-esp32 #6656: ESP32-C3 won't wake from deep sleep with GPIO0 or GPIO1

pjenks-accudyne, 2022-04-28:
### Board

ESP32-C3-DevKitM-1

### Device Description

ESP32-C3-DevKitM-1

### Hardware Configuration

100k potentiometers connected to GPIO0-GPIO3

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am unable to use GPIO0 or GPIO1 to wake the ESP32-C3 from deep sleep, but I am able to use GPIO2 and GPIO3 to wake it from deep sleep. I'm using GPIO0-GPIO3 to read analog values from force sensors, simulated with potentiometers in this test setup. I'm also using these force sensors to wake the ESP32-C3 from deep sleep. I've been able to read analog and digital inputs on all 4 pins when the board is awake.

I have tested this setup with every combination of including and excluding the following functions:
`gpio_deep_sleep_hold_dis();`
`esp_sleep_config_gpio_isolate();`
`esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);`

The only seemingly important difference I've found between GPIO0-GPIO1 and GPIO2-GPIO3 is that GPIO0-GPIO1 are input disabled after reset, while GPIO2-GPIO3 are input enabled after reset. 

I've also noticed that the reset reason given in the debug messages is only DSLEEP if I wake the ESP32-C3 within about 0.5 seconds. The first boot sequence in my debug messages is from initial power on. The second is from waking the board from deep sleep with GPIO2 after sleeping for 3 seconds. The third is from waking the board from deep sleep with GPIO2 immediately after entering deep sleep. It seems that my bootCount variable does not persist unless the board is awoken immediately.

### Sketch

```cpp
const uint64_t WAKEUP_PIN_BITMASK = 0b1111;     //GPIO0-GPIO3
RTC_DATA_ATTR int bootCount = 0;

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();
  switch(wakeup_reason) {
    case ESP_SLEEP_WAKEUP_GPIO : Serial.println("Wakeup caused by external signal using GPIO"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}

void setup() {
  Serial.begin(115200);
  bootCount++;
  Serial.println();
  Serial.println("Boot number: " + String(bootCount));
  print_wakeup_reason();
  gpio_deep_sleep_hold_dis();
  esp_err_t errPD = esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  Serial.print("esp_sleep_pd_config: ");
  switch(errPD) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  esp_err_t errGPIO = esp_deep_sleep_enable_gpio_wakeup(WAKEUP_PIN_BITMASK, ESP_GPIO_WAKEUP_GPIO_HIGH);
  Serial.print("esp_deep_sleep_enable_gpio_wakeup: ");
  switch(errGPIO) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  Serial.println();
  Serial.println();
  delay(500);
  esp_deep_sleep_start();
}

void loop() {
  //Not used
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
Attempting to boot anyway...
entry 0x403ce000
[    87][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB
Boot number: 1
Wakeup was not caused by deep sleep
esp_sleep_pd_config: ESP_OK
esp_deep_sleep_enable_gpio_wakeup: ESP_OK

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
Attempting to boot anyway...
entry 0x403ce000
[    87][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB:
Boot number: 1
Wakeup was not caused by deep sleep
esp_sleep_pd_config: ESP_OK
esp_deep_sleep_enable_gpio_wakeup: ESP_OK

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x5 (DSLEEP),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
Attempting to boot anyway...
entry 0x403ce000
[    87][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB
Boot number: 2
Wakeup caused by external signal using GPIO
esp_sleep_pd_config: ESP_OK
esp_deep_sleep_enable_gpio_wakeup: ESP_OK
```

### Other Steps to Reproduce

I have tested this setup on two identical ESP32-C3-DevKitM-1 boards.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pjenks-accudyne, 2022-05-03:
I was able to get all of the GPIO pins functioning correctly as wakeup sources for deep sleep by calling the set direction function before entering deep sleep.

```
gpio_set_direction(GPIO_NUM_0, GPIO_MODE_INPUT);
gpio_set_direction(GPIO_NUM_1, GPIO_MODE_INPUT);
gpio_set_direction(GPIO_NUM_2, GPIO_MODE_INPUT);
gpio_set_direction(GPIO_NUM_4, GPIO_MODE_INPUT);
esp_deep_sleep_start();
```

I still have not resolved the incorrect reset cause when exiting deep sleep.

closed 2022-05-03 by pjenks-accudyne

rahmanshaber, 2022-07-13:
@pjenks-accudyne Hi, i was wondering how you external wake up works as Esp32-c3 doesn't have that feature?

pjenks-accudyne, 2022-07-13:
@rahmanshaber The ESP32-C3 does support deep sleep with external wakeup. You can find the API for that feature [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html). The API is slightly different than that for the ESP32. Also only the RTC GPIO pins (GPIO0 - GPIO5) can be used for deep sleep external wakeup. Like I mentioned above, the deep sleep functionality is not complete in terms of reporting the wakeup cause, but that does not affect the utility of the feature.

Here is the basic template I used in my code for deep sleep with GPIO wakeup:

```
#include <driver/gpio.h>
#include "esp_sleep.h"

//Define bitmasks for GPIO pins used to wake from deep sleep
//Wake from deep sleep when these pins are high
const uint64_t WAKEUP_HIGH_PIN_BITMASK = 0b001111;    //GPIO0-GPIO3
//Wake from deep sleep when these pins are low
const uint64_t WAKEUP_LOW_PIN_BITMASK = 0b110000;    //GPIO4-GPIO5

//Call to enter deep sleep mode with GPIO wakeup
void deep_sleep_gpio_init(void){
  //GPIO pins must be set as inputs
  for(int i=0; i<6; i++) {
    gpio_set_direction(gpio_num_t(i), GPIO_MODE_INPUT);
  }

  //hold disable, isolate and power domain config functions may be unnecessary
  gpio_deep_sleep_hold_dis();
  esp_sleep_config_gpio_isolate();
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

  //Wakeup on low and high can be used concurrently for separate pins
  esp_deep_sleep_enable_gpio_wakeup(WAKEUP_HIGH_PIN_BITMASK, ESP_GPIO_WAKEUP_GPIO_HIGH);
  esp_deep_sleep_enable_gpio_wakeup(WAKEUP_LOW_PIN_BITMASK, ESP_GPIO_WAKEUP_GPIO_LOW);

  esp_deep_sleep_start();
}
```

rahmanshaber, 2022-07-14:
@pjenks-accudyne Yes, i tested your code and it works, but why esp didn't listed gpio external wake up as an option in doc here? https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#wakeup-sources

malachib, 2023-01-12:
I was encountering similar issues.  It turns out there's no way (that I know of) to configure an internal pullup resistor for the ESP32C3 GPIO for deep sleep wakeup.  My "wake on high" switch defaulting to GND just misbehaved like a bad citizen.

The [documentation ](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/api-reference/system/sleep_modes.html#configuring-ios) implies you can do an internal pullup in this context, but I've spent many hours trying to make that work with no luck.  External pullup resistor FTW

Chovich, 2023-04-17:
> @rahmanshaber The ESP32-C3 does support deep sleep with external wakeup. You can find the API for that feature [...]

Hey, can you tell me why I cant wake up esp32c3 with this code ?

#include <Arduino.h>
#include "esp_sleep.h"
#include "driver/gpio.h"
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define WAKE_UP_BUTTON_GPIO 3
#define SLEEP_DURATION 10000000 // 10 seconds in microseconds
#define DEBOUNCE_TIME 100

#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET -1

Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

volatile bool buttonPressed = false;

void IRAM_ATTR handleButtonPress() {
  buttonPressed = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Starting...");

  gpio_set_direction(GPIO_NUM_3, GPIO_MODE_INPUT);

  Wire.begin(4, 2);
  if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println("SSD1306 allocation failed");
    for (;;);
  }

  pinMode(WAKE_UP_BUTTON_GPIO, INPUT);

  attachInterrupt(digitalPinToInterrupt(WAKE_UP_BUTTON_GPIO), handleButtonPress, RISING);

  esp_sleep_wakeup_cause_t wakeup_cause = esp_sleep_get_wakeup_cause();
  if (wakeup_cause == ESP_SLEEP_WAKEUP_GPIO) {
    Serial.println("Woken up from deep sleep by GPIO");
  } else if (wakeup_cause == ESP_SLEEP_WAKEUP_TIMER) {
    Serial.println("Woken up from deep sleep by timer");
  } else {
    Serial.println("Power on or reset");
  }
}

void loop() {
  display.clearDisplay();
  display.setTextSize(1);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
  display.println("Hello, World!");
  display.display();

  if (buttonPressed) {
    buttonPressed = false;
    delay(DEBOUNCE_TIME);
    enterDeepSleep(SLEEP_DURATION);
  }
}

void enterDeepSleep(uint64_t sleepDurationUs) {
  // Turn off the display before going to sleep
  display.clearDisplay();
  display.ssd1306_command(SSD1306_DISPLAYOFF);
  display.display();

  Serial.println("Going to deep sleep...");
  gpio_wakeup_enable((gpio_num_t)WAKE_UP_BUTTON_GPIO, GPIO_INTR_HIGH_LEVEL);
  esp_sleep_enable_gpio_wakeup();
  esp_sleep_enable_timer_wakeup(sleepDurationUs);
  esp_deep_sleep_start();
}

pjenks-accudyne, 2023-04-17:
@Chovich I noticed a couple things that may be causing the issue. 

1. You're intermixing light sleep and deep sleep functions. `esp_sleep_enable_gpio_wakeup`  is for light sleep, but at the end you call `esp_deep_sleep_start`. 
2. `gpio_wakeup_enable` is referenced in the definitions of other functions but I don't see an actual definition for that function in the API for the ESP32-C3. That's not to say it doesn't work, but I've never used light sleep with the ESP32-C3 so I don't know if it works.
3. If you are trying to use deep sleep the GPIO pin definition needs to be a bitmask. For GPIO 3 that would look like `#define WAKE_UP_BUTTON_GPIO_BITMASK 0b001000`
4. If you are trying to use deep sleep I don't know if you can have multiple wakeup sources. Try the code below, but that might still cause a problem.

Here is some modified code with the display stuff removed.

If you're trying to use light sleep I assume your code should look like this (never used light sleep so I'm not certain):
```
#include "esp_sleep.h"
#include "driver/gpio.h"

#define WAKE_UP_BUTTON_GPIO 3
#define SLEEP_DURATION 10000000 // 10 seconds in microseconds
#define DEBOUNCE_TIME 100

volatile bool buttonPressed = false;

void IRAM_ATTR handleButtonPress() {
  buttonPressed = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Starting...");

  gpio_set_direction(gpio_num_t(WAKE_UP_BUTTON_GPIO), GPIO_MODE_INPUT);

  pinMode(WAKE_UP_BUTTON_GPIO, INPUT);

  attachInterrupt(digitalPinToInterrupt(WAKE_UP_BUTTON_GPIO), handleButtonPress, RISING);

  esp_sleep_wakeup_cause_t wakeup_cause = esp_sleep_get_wakeup_cause();
  if (wakeup_cause == ESP_SLEEP_WAKEUP_GPIO) {
    Serial.println("Woken up from light sleep by GPIO");
  } else if (wakeup_cause == ESP_SLEEP_WAKEUP_TIMER) {
    Serial.println("Woken up from light sleep by timer");
  } else {
    Serial.println("Power on or reset");
  }
}

void loop() {
  if (buttonPressed) {
    buttonPressed = false;
    delay(DEBOUNCE_TIME);
    enterLightSleep(SLEEP_DURATION);
  }
}

void enterLightSleep(uint64_t sleepDurationUs) {
  Serial.println("Going to light sleep...");
  gpio_wakeup_enable(gpio_num_t(WAKE_UP_BUTTON_GPIO), GPIO_INTR_HIGH_LEVEL);
  esp_sleep_enable_gpio_wakeup();
  esp_sleep_enable_timer_wakeup(sleepDurationUs);
  esp_light_sleep_start();
}
```

If you're trying to use deep sleep it should look like this:
```
#include "esp_sleep.h"
#include "driver/gpio.h"

#define WAKE_UP_BUTTON_GPIO 3
#define WAKE_UP_BUTTON_GPIO_BITMASK 0b001000
#define SLEEP_DURATION 10000000 // 10 seconds in microseconds
#define DEBOUNCE_TIME 100

volatile bool buttonPressed = false;

void IRAM_ATTR handleButtonPress() {
  buttonPressed = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Starting...");

  gpio_set_direction(gpio_num_t(WAKE_UP_BUTTON_GPIO), GPIO_MODE_INPUT);

  pinMode(WAKE_UP_BUTTON_GPIO, INPUT);

  attachInterrupt(digitalPinToInterrupt(WAKE_UP_BUTTON_GPIO), handleButtonPress, RISING);

  esp_sleep_wakeup_cause_t wakeup_cause = esp_sleep_get_wakeup_cause();
  if (wakeup_cause == ESP_SLEEP_WAKEUP_GPIO) {
    Serial.println("Woken up from deep sleep by GPIO");
  } else if (wakeup_cause == ESP_SLEEP_WAKEUP_TIMER) {
    Serial.println("Woken up from deep sleep by timer");
  } else {
    Serial.println("Power on or reset");
  }
}

void loop() {
  if (buttonPressed) {
    buttonPressed = false;
    delay(DEBOUNCE_TIME);
    enterDeepSleep(SLEEP_DURATION);
  }
}

void enterDeepSleep(uint64_t sleepDurationUs) {
  Serial.println("Going to deep sleep...");
  esp_deep_sleep_enable_gpio_wakeup(WAKE_UP_BUTTON_GPIO_BITMASK, ESP_GPIO_WAKEUP_GPIO_HIGH);
  esp_deep_sleep_enable_timer_wakeup(sleepDurationUs);
  esp_deep_sleep_start();
}
```

bricomp, 2023-06-16:
esp_deep_sleep_enable_timer_wakeup(sleepDurationUs); DOES NOT exist for ESP32C3!!

pjenks-accudyne, 2023-06-16:
@bricomp You're right `esp_deep_sleep_enable_timer_wakeup` doesn't exist, but you can get still utilize deep sleep with a timer wakeup on the ESP32C3. You can use `esp_sleep_enable_timer_wakeup(uint64_t time_in_us)` followed by `esp_deep_sleep_start()`. You can also call `esp_deep_sleep(uint64_t time_in_us)` which does the same thing.