espressif/arduino-esp32 #6658: MacAddress.h

mrengineer7777, 2022-04-28:
### Related area

WiFi Library

### Hardware specification

ESP32, no specific hardware

### Is your feature request related to a problem?

I'm finding handling BSSID info to be a pain.  Parsing from string, printing to string, saving, checking if null, etc.  Is there a library for BSSIDs/MAC like IPAddress.cpp/h?

If not, I will create one patterned after IPAddress.cpp/h and submit a PR.

Just making sure I'm not duplicating work, and I'd like to know if this would be useful to others.

### Describe the solution you'd like

Say yes and I'll write a library and submit a PR.  I just don't want to waste my time and yours if a solution already exists.

### Describe alternatives you've considered

Searched Google extensively and didn't find functions to handle BSSID/mac within ESP32/arduino

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2022-04-29:
I've started writing a library because it will make our project cleaner.

dpnebert, 2024-02-15:
I’m not sure if it is what you are looking for, but my PR for getting the MAC addresses was just merged.  #8618

PR #9304 (2024-03-18): Feature: New Class MacAddress similar to IPAddress
## Description of Change
This adds new class `MacAddress` that allows for better handling of MAC and BSSID addresses. 
MacAddress also support  8-byte EUI-64 as noted in `esp_mac.h`.

Changes have been done on top of @mrengineer7777 PR #6667. Thanks David!

## Tests scenarios
Tested by running slightly modified test sketch by @mrengineer7777.
https://gist.github.com/P-R-O-C-H-Y/ab6b2443fc57bc12ec424f89493b9022

## Related links
Closes #6658
closed 2024-03-18 by me-no-dev