espressif/arduino-esp32 #6663: DNSServer memory leak

alex-code, 2022-04-29:
### Board

DFRobot FireBeetle2

### Device Description

None

### Hardware Configuration

None

### Version

v2.0.2

### IDE Name

PlateformIO

### Operating System

Win11

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

A project I'm working on uses `DNSServer` on demand and I noticed a memory leak.

Memory is allocated in the `DNSServer` class constructor but there's no destructor to free it.

https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/src/DNSServer.cpp#L16
https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/src/DNSServer.cpp#L17

### Sketch

```cpp
None
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-04-29:
That's true.  Several libraries have that issue.  However since DNSServer is generally only called once and then remains in memory, it is basically a global allocation.  If it really bothers you, write the destructor and submit a PR.

SuGlider, 2022-04-29:
> That's true. Several libraries have that issue. However since DNSServer is generally only called once and then remains in memory, it is basically a global allocation. If it really bothers you, write the destructor and submit a PR.

I think it would be good to review the whole code and fix these similar issues.
Thanks @alex-code for reporting!

alex-code, 2022-04-29:
> That's true. Several libraries have that issue. However since DNSServer is generally only called once and then remains in memory, it is basically a global allocation. If it really bothers you, write the destructor and submit a PR.

It may typically be used globally but if it allocates memory it should really free it too.

I don't mind doing a PR for allocations that need freeing if it'll be useful.

SuGlider, 2022-05-06:
@alex-code - PR #6707 shall fixes it. 
Please let me know if you can confirm that it solves this issue.

alex-code, 2022-05-07:
> @alex-code - PR #6707 shall fixes it. Please let me know if you can confirm that it solves this issue.

Looks good to me, testing again with the changes the free memory reported before construction is the same after destroying 👍

SuGlider, 2022-05-07:
@alex-code - Thanks for testing and confirming the resolution of this issue.

PR #6707 (2022-05-09): Fixes DNS Server Memory Leak when deleted
## Description of Change
Adds a destructor to DNSServer Class in order free all memory allocated when the object is deleted.

## Tests scenarios
Simple sketch that creates and deletes a DNSServer object.

## Related links
fixes #6663
closed 2022-05-09 by me-no-dev