espressif/arduino-esp32 #6669: `pinMode` disables interrupts as a side effect

quentinmit, 2022-04-30:
### Board

ESP32-DevKitC

### Device Description

DevKitC

### Hardware Configuration

For the demo code below, connect a switch or square wave to pin 19.

### Version

v2.0.2

### IDE Name

PlatformIO

### Operating System

macOS

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The `pinMode` call, in addition to changing pullups/whether a pin is driving the output or not/etc, appears to disable pin interrupts as a side effect. In the ESP-IDF version of the code, `intr_type` is explicitly set to `GPIO_INTR_DISABLE`. In the older register code, the field is accidentally overwritten by the write to `GPIO.pin[x].val` when it's really just trying to set `GPIO.pin[x].pad_driver`.

This is a problem because while `pinMode` is `IRAM_ATTR` and safe to call from interrupt handlers, `attachInterrupt` is not (and it's also rather heavyweight), so it's not possible to re-enable the interrupt if `pinMode` is being used in an ISR.

`pinMode` should just preserve whatever previous state of interrupts existed; if the user has an interrupt configured on a pin and wants to change the pin mode, they probably know what they are doing. If they want to disable the interrupt they can call `detachInterrupt` directly.

### Sketch

```cpp
#include "Arduino.h"

// This sketch should print the number of times pin 19 changes.

uint32_t interrupt_count = 0;

void IRAM_ATTR pin_interrupt(void *arg) {
  interrupt_count++;
  pinMode(19, INPUT);
}

void setup() {
  pinMode(19, INPUT);
  attachInterruptArg(19, pin_interrupt, 0, CHANGE);
}

void loop() {
  ESP_LOGD("loop", "interrupt count %d", interrupt_count);
  delay(500);
}
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-04-30:
Issue confirmed with the latest Arduino Core Version 2.0.3-RC1.

@P-R-O-C-H-Y,  PTAL

TD-er, 2022-05-02:
Hmm I think this also affects PWM.
Set PWM on a pin and call pinMode => No more active PWM on that pin.

P-R-O-C-H-Y, 2022-05-04:
Hi @quentinmit 
When calling `pinMode(pin,mode);` it configures the pin for default config, which is interrupt disabled.
So you need to `call attachInterruptArg(19, pin_interrupt, 0, CHANGE);` again after pinMode to make it work :)

```
#include "Arduino.h"

// This sketch should print the number of times pin 19 changes.

uint32_t interrupt_count = 0;

void IRAM_ATTR pin_interrupt(void *arg) {
  interrupt_count++;
  pinMode(19, INPUT);
  attachInterruptArg(19, pin_interrupt, 0, CHANGE);
}

void setup() {
  pinMode(19, INPUT);
  attachInterruptArg(19, pin_interrupt, 0, CHANGE);
}

void loop() {
  ESP_LOGD("loop", "interrupt count %d", interrupt_count);
  delay(500);
}
```

P-R-O-C-H-Y, 2022-05-04:
Hi @TD-er,

If you setup some PWM peripheral like LEDC on a pin, when you call pinMode on same pin, the configuration will set the pin function to be GPIO and the signal from peripheral will be no longer connected to the pin.

TD-er, 2022-05-04:
> Hi @TD-er,
> 
> If you setup some PWM peripheral like LEDC on a pin, when you call pinMode on same pin, the configuration will set the pin function to be GPIO and the signal from peripheral will be no longer connected to the pin.

Yep, found that too, after updating to IDF4.4
With the older SDK it behaved differently.

P-R-O-C-H-Y, 2022-05-04:
@quentinmit

I have read you decryption again and it make sense, to keep the state interrupt state so when changing pinMode, it won't be disabled. I will add this to roadmap :)

quentinmit, 2022-05-04:
> Hi @quentinmit When calling `pinMode(pin,mode);` it configures the pin for default config, which is interrupt disabled. So you need to `call attachInterruptArg(19, pin_interrupt, 0, CHANGE);` again [...]

Hi @P-R-O-C-H-Y. Your sample code is invalid because `attachInterruptArg` may not be called from an interrupt handler.

P-R-O-C-H-Y, 2022-05-05:
Hi @quentinmit 
If you have a time, can you test changes in linked PR? That should fix your issue.
Let me know, if there is still a problem or fixed :)

PR #6695 (2022-05-09): Changed pinMode() default interrupt type DISABLED to previously set
## Description of Change
pins default configuration has `intr_type =  GPIO_INTR_DISABLE`
With this implementation, it will set the intr_type to previously set intr_type. It will no longer disable interrupt, when pinmode is called multiple times on same pin with interrupt enabled.

There is no option to get pin config from ESP-IDF API, that's why I made this implementation.

## Tests scenarios

```
//To trigger ISR, connect GPIO 19 to GND.

uint32_t interrupt_count = 0;

void IRAM_ATTR pin_interrupt(void *arg) {
  interrupt_cou [... cut]
closed 2022-05-09 by me-no-dev